intel/compute-runtime #123: Build error on recent versions

wangliwei-intel, 2018-12-22:
I failed to build the recent version, it show me the error:
cd /home/wangliwei/Desktop/Neo-Latest/neo/runtime/scheduler && LD_LIBRARY_PATH=/home/wangliwei/Desktop/Neo-Latest/neo/build/bin /home/wangliwei/Desktop/Neo-Latest/neo/build/bin/ocloc -q -file scheduler.cl -device bdw -cl-intel-greater-than-4GB-buffer-required -64 -out_dir /home/wangliwei/Desktop/Neo-Latest/neo/build/bin/scheduler/x64/gen8 -cpp_file -options -I/usr/local/include/igc\ -I/usr/local/include/igc/cif\ -I/usr/local/include/igc/ocl_igc_shared/executable_format\ -I/usr/local/include/igc/ocl_igc_shared/device_enqueue\ -I\ ../gen8\ -cl-kernel-arg-info\ -cl-std=CL2.0
igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen8core.dir/build.make:61: recipe for target 'bin/scheduler/x64/gen8/scheduler_Gen8core.bin' failed
make[2]: *** [bin/scheduler/x64/gen8/scheduler_Gen8core.bin] Error 250
make[2]: Leaving directory '/home/wangliwei/Desktop/Neo-Latest/neo/build'
CMakeFiles/Makefile2:1104: recipe for target 'igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen8core.dir/all' failed
make[1]: *** [igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen8core.dir/all] Error 2
make[1]: Leaving directory '/home/wangliwei/Desktop/Neo-Latest/neo/build'
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

I tried both versions of igc described in the https://github.com/intel/intel-graphics-compiler
But I always got the same error.

Have anyone experienced similar error and overcome it eventually?
If so, give me some hints, thank you very much.

JacekDanecki, 2019-01-11:
It looks like installation issue, can you confirm IGC is installed correctly. Try to run:
```
$ export LD_DEBUG=libs
$ clinfo
```
You should see something similar like:
```
 Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
        76:     find library=libigdfcl.so [0]; searching
        76:      search cache=/etc/ld.so.cache
        76:      search path=/lib/x86_64-linux-gnu:/usr/lib/x86_64-linux-gnu:/lib:/usr/lib              (system search path)
        76:       trying file=/lib/x86_64-linux-gnu/libigdfcl.so
        76:       trying file=/usr/lib/x86_64-linux-gnu/libigdfcl.so
        76:
        76:
        76:     calling init: /usr/lib/x86_64-linux-gnu/libigdfcl.so
        76:
        76:     find library=libigc.so [0]; searching
        76:      search cache=/etc/ld.so.cache
        76:      search path=/lib/x86_64-linux-gnu:/usr/lib/x86_64-linux-gnu:/lib:/usr/lib              (system search path)
        76:       trying file=/lib/x86_64-linux-gnu/libigc.so
        76:       trying file=/usr/lib/x86_64-linux-gnu/libigc.so
        76:
        76:     find library=librt.so.1 [0]; searching
        76:      search cache=/etc/ld.so.cache
        76:       trying file=/lib/x86_64-linux-gnu/librt.so.1
        76:
        76:
        76:     calling init: /lib/x86_64-linux-gnu/librt.so.1
        76:
        76:
        76:     calling init: /usr/lib/x86_64-linux-gnu/libigc.so
        76:
        76:     find library=libopencl_clang.so [0]; searching
        76:      search cache=/etc/ld.so.cache
        76:      search path=/lib/x86_64-linux-gnu:/usr/lib/x86_64-linux-gnu:/lib:/usr/lib              (system search path)
        76:       trying file=/lib/x86_64-linux-gnu/libopencl_clang.so
        76:       trying file=/usr/lib/x86_64-linux-gnu/libopencl_clang.so
        76:
        76:
        76:     calling init: /usr/lib/x86_64-linux-gnu/libopencl_clang.so
        76:
  Preferred work group size multiple              32
  Max sub-groups per work group                   32

```
3 compilers libraries should be initialized:
/usr/lib/x86_64-linux-gnu/libigdfcl.so
/usr/lib/x86_64-linux-gnu/libigc.so
/usr/lib/x86_64-linux-gnu/libopencl_clang.so

JacekDanecki, 2019-01-16:
Closing as it's IGC installation issue.

PTS93, 2019-01-28:
If someone comes across this. Running `sudo ldconfig` was necessary for me to fix this after compiling and installing IGC.

PlutusCN, 2019-02-05:
@JacekDanecki 
(1) Should IGC be installed in /usr/lib/x86_64-linux-gnu ? 
(2) Why compute-runtime can not find IGC from /usr/local/lib ? 

Because I have installed IGC in /usr/local/lib as default.

JacekDanecki, 2019-02-05:
@PlutusCN 
IGC can be installed in different directories, but must be visible for system linker. You can check whether IGC is visible or not by running command:
```
$ ldconfig -p | grep igc.so
        libigc.so (libc6,x86-64) => /usr/local/lib/libigc.so
```
but on other system it can be
```
$ ldconfig -p | grep igc.so
        libigc.so (libc6,x86-64) => /usr/lib/libigc.so
```
or
```
$ ldconfig -p | grep igc.so
        libigc.so (libc6,x86-64) => /lib64/libigc.so
```
On Ubuntu there is configuration file for linker /etc/ld.so.conf.d/libc.conf (from libc-bin package) containing:
```
# libc default configuration
/usr/local/lib
```
so libraries installed in this directory should be visible for linker. If not, you need to run:
```
sudo ldconfig
```
as @PTS93 mentioned earlier, to update linker cache.

On Fedora system there is not such file, so if you install any libraries to /usr/local/lib, you can
add configuration file in /etc/ld.so.conf.d, similar like in Ubuntu, and run
```
sudo ldconfig
```
or set LD_LIBRARY_PATH environment variable, for example:
```
export LD_LIBRARY_PATH=/usr/local/lib
```
What Linux distribution are you using?

PlutusCN, 2019-02-05:
@JacekDanecki 
If I install intel-igc-core_19.02.1330_amd64.deb and intel-igc-opencl_19.02.1330_amd64.deb, then I will compile compute-runtime successfully. But that will fail to compile compute-runtime if I remove intel-igc-core_19.02.1330_amd64.deb and intel-igc-opencl_19.02.1330_amd64.deb. 

BUT, I always compile and install 

GmmLib - https://github.com/intel/gmmlib
Intel Graphics Compiler - https://github.com/intel/intel-graphics-compiler 
libdrm - https://anongit.freedesktop.org/git/mesa/drm.git
libva - https://github.com/intel/libva.git

successfully. I still failed to compile compute-runtime.

why ?

JacekDanecki, 2019-02-05:
@PlutusCN Can you share log from compilation?

PlutusCN, 2019-02-05:
@JacekDanecki 

[cmake.log](https://github.com/intel/compute-runtime/files/2832112/cmake.log)

[make.log](https://github.com/intel/compute-runtime/files/2832113/make.log)

I compiled as the part of [Building](https://github.com/intel/compute-runtime/blob/master/documentation/BUILD_Ubuntu.md) 

Thanks a lot

PTS93, 2019-02-05:
@PlutusCN 
Did you run `sudo make install` for gmmlib? The guide does not mention it but its actually needed to make the compiled lib available.

JacekDanecki, 2019-02-05:
@PlutusCN Can you run in build directory command
```
make -n -f igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen9core.dir/build.make igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen9core.dir/build
```
it will show you how to run ocloc offline compiler, on my system it looks like this:
```
$ make -n -f igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen9core.dir/build.make igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen9core.dir/build
/usr/bin/cmake -E cmake_echo_color --switch= --blue --bold --progress-dir=/media/data/sources/intel/github/build/gcc/CMakeFiles --progress-num= "Generating ../../bin/scheduler/x64/gen9/scheduler_Gen9core.bin"
cd /media/data/sources/intel/github/runtime/scheduler && LD_LIBRARY_PATH=/media/data/sources/intel/github/build/gcc/bin /media/data/sources/intel/github/build/gcc/bin/ocloc -q -file scheduler.cl -device skl -cl-intel-greater-than-4GB-buffer-required -64 -out_dir /media/data/sources/intel/github/build/gcc/bin/scheduler/x64/gen9 -cpp_file -options -I/usr/include/igc\ -I/usr/include/igc/cif\ -I/usr/include/igc/ocl_igc_shared/executable_format\ -I/usr/include/igc/ocl_igc_shared/device_enqueue\ -I\ ../gen9\ -cl-kernel-arg-info\ -cl-std=CL2.0
```
Now you can set LD_DEBUG environment variable:
```
export LD_DEBUG=libs
```
and run ocloc command. In the log you can check whether IGC libraries were found and initialized:
```
29201:     calling init: /lib64/libigdfcl.so
29201:     find library=libigc.so [0]; searching
29201:      search path=/media/data/sources/intel/github/build/gcc/bin             (LD_LIBRARY_PATH)
29201:       trying file=/media/data/sources/intel/github/build/gcc/bin/libigc.so
29201:      search cache=/etc/ld.so.cache
29201:       trying file=/lib64/libigc.so
29201:     calling init: /lib64/libigc.so
29201:
29201:     find library=libopencl_clang.so [0]; searching
29201:      search path=/media/data/sources/intel/github/build/gcc/bin             (LD_LIBRARY_PATH)
29201:       trying file=/media/data/sources/intel/github/build/gcc/bin/libopencl_clang.so
29201:      search cache=/etc/ld.so.cache
29201:       trying file=/lib64/libopencl_clang.so
29201:
29201:
29201:     calling init: /lib64/libopencl_clang.so
```
If libraries were not found, check linker cache:
```
$ ldconfig -p | grep libigc.so
libigc.so (libc6,x86-64) => /lib64/libigc.so
$ ldconfig -p | grep libopencl_clang.so
libopencl_clang.so (libc6,x86-64) => /lib64/libopencl_clang.so
$ldconfig -p | grep libigdfcl.so
libigdfcl.so (libc6,x86-64) => /lib64/libigdfcl.so
```
On your system IGC libraries should be in /usr/local/lib directory. If not, run 
```
sudo ldconfig
```

PlutusCN, 2019-02-14:
@JacekDanecki 

Here is my steps as your words, where is my fault ?  Thank you 

```
dengliqiang@Intel-Dev:~/Dev/source/build_neo$ make -n -f igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen9core.dir/build.make igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen9core.dir/build
/usr/bin/cmake -E cmake_echo_color --switch= --blue --bold --progress-dir=/home/dengliqiang/Dev/source/build_neo/CMakeFiles --progress-num= "Generating ../../bin/scheduler/x64/gen9/scheduler_Gen9core.bin"
cd /home/dengliqiang/Dev/source/neo/runtime/scheduler && LD_LIBRARY_PATH=/home/dengliqiang/Dev/source/build_neo/bin /home/dengliqiang/Dev/source/build_neo/bin/ocloc -q -file scheduler.cl -device skl -cl-intel-greater-than-4GB-buffer-required -64 -out_dir /home/dengliqiang/Dev/source/build_neo/bin/scheduler/x64/gen9 -cpp_file -options -I/usr/local/include/igc\ -I/usr/local/include/igc/cif\ -I/usr/local/include/igc/ocl_igc_shared/executable_format\ -I/usr/local/include/igc/ocl_igc_shared/device_enqueue\ -I\ ../gen9\ -cl-kernel-arg-info\ -cl-std=CL2.0
dengliqiang@Intel-Dev:~/Dev/source/build_neo$ export LD_DEBUG=libs
dengliqiang@Intel-Dev:~/Dev/source/build_neo$ bin/ocloc -?
     30096:	find library=libdl.so.2 [0]; searching
     30096:	 search path=tls/x86_64:tls:x86_64::/usr/local/lib/tls/x86_64:/usr/local/lib/tls:/usr/local/lib/x86_64:/usr/local/lib		(LD_LIBRARY_PATH)
     30096:	  trying file=tls/x86_64/libdl.so.2
     30096:	  trying file=tls/libdl.so.2
     30096:	  trying file=x86_64/libdl.so.2
     30096:	  trying file=libdl.so.2
     30096:	  trying file=/usr/local/lib/tls/x86_64/libdl.so.2
     30096:	  trying file=/usr/local/lib/tls/libdl.so.2
     30096:	  trying file=/usr/local/lib/x86_64/libdl.so.2
     30096:	  trying file=/usr/local/lib/libdl.so.2
     30096:	 search cache=/etc/ld.so.cache
     30096:	  trying file=/lib/x86_64-linux-gnu/libdl.so.2
     30096:	
     30096:	find library=libpthread.so.0 [0]; searching
     30096:	 search path=tls/x86_64:tls:x86_64::/usr/local/lib		(LD_LIBRARY_PATH)
     30096:	  trying file=tls/x86_64/libpthread.so.0
     30096:	  trying file=tls/libpthread.so.0
     30096:	  trying file=x86_64/libpthread.so.0
     30096:	  trying file=libpthread.so.0
     30096:	  trying file=/usr/local/lib/libpthread.so.0
     30096:	 search cache=/etc/ld.so.cache
     30096:	  trying file=/lib/x86_64-linux-gnu/libpthread.so.0
     30096:	
     30096:	find library=libstdc++.so.6 [0]; searching
     30096:	 search path=tls/x86_64:tls:x86_64::/usr/local/lib		(LD_LIBRARY_PATH)
     30096:	  trying file=tls/x86_64/libstdc++.so.6
     30096:	  trying file=tls/libstdc++.so.6
     30096:	  trying file=x86_64/libstdc++.so.6
     30096:	  trying file=libstdc++.so.6
     30096:	  trying file=/usr/local/lib/libstdc++.so.6
     30096:	 search cache=/etc/ld.so.cache
     30096:	  trying file=/usr/lib/x86_64-linux-gnu/libstdc++.so.6
     30096:	
     30096:	find library=libgcc_s.so.1 [0]; searching
     30096:	 search path=tls/x86_64:tls:x86_64::/usr/local/lib		(LD_LIBRARY_PATH)
     30096:	  trying file=tls/x86_64/libgcc_s.so.1
     30096:	  trying file=tls/libgcc_s.so.1
     30096:	  trying file=x86_64/libgcc_s.so.1
     30096:	  trying file=libgcc_s.so.1
     30096:	  trying file=/usr/local/lib/libgcc_s.so.1
     30096:	 search cache=/etc/ld.so.cache
     30096:	  trying file=/lib/x86_64-linux-gnu/libgcc_s.so.1
     30096:	
     30096:	find library=libc.so.6 [0]; searching
     30096:	 search path=tls/x86_64:tls:x86_64::/usr/local/lib		(LD_LIBRARY_PATH)
     30096:	  trying file=tls/x86_64/libc.so.6
     30096:	  trying file=tls/libc.so.6
     30096:	  trying file=x86_64/libc.so.6
     30096:	  trying file=libc.so.6
     30096:	  trying file=/usr/local/lib/libc.so.6
     30096:	 search cache=/etc/ld.so.cache
     30096:	  trying file=/lib/x86_64-linux-gnu/libc.so.6
     30096:	
     30096:	find library=libm.so.6 [0]; searching
     30096:	 search path=tls/x86_64:tls:x86_64::/usr/local/lib		(LD_LIBRARY_PATH)
     30096:	  trying file=tls/x86_64/libm.so.6
     30096:	  trying file=tls/libm.so.6
     30096:	  trying file=x86_64/libm.so.6
     30096:	  trying file=libm.so.6
     30096:	  trying file=/usr/local/lib/libm.so.6
[... 60 lines cut ...]
     30113:	find library=libc.so.6 [0]; searching
     30113:	 search path=tls/x86_64:tls:x86_64::/usr/local/lib		(LD_LIBRARY_PATH)
     30113:	  trying file=tls/x86_64/libc.so.6
     30113:	  trying file=tls/libc.so.6
     30113:	  trying file=x86_64/libc.so.6
     30113:	  trying file=libc.so.6
     30113:	  trying file=/usr/local/lib/libc.so.6
     30113:	 search cache=/etc/ld.so.cache
     30113:	  trying file=/lib/x86_64-linux-gnu/libc.so.6
     30113:	
     30113:	find library=libm.so.6 [0]; searching
     30113:	 search path=tls/x86_64:tls:x86_64::/usr/local/lib		(LD_LIBRARY_PATH)
     30113:	  trying file=tls/x86_64/libm.so.6
     30113:	  trying file=tls/libm.so.6
     30113:	  trying file=x86_64/libm.so.6
     30113:	  trying file=libm.so.6
     30113:	  trying file=/usr/local/lib/libm.so.6
     30113:	 search cache=/etc/ld.so.cache
     30113:	  trying file=/lib/x86_64-linux-gnu/libm.so.6
     30113:	...

```

The detailed building message 
[build_neo_error.txt](https://github.com/intel/compute-runtime/files/2862958/build_neo_error.txt)

PlutusCN, 2019-02-14:
@wangliwei-intel 

Do you fix the compilation problem ? 

If you did it, tell me how you did it, and thanks a lot.

wangliwei-intel, 2019-02-14:
> @wangliwei-intel
> 
> Do you fix the compilation problem ?
> 
> If you did it, tell me how you did it, and thanks a lot.

Hi, Richard,
What model is your machine? If it's older than Broadwell (GEN8), Could you please switch to a newer one and try it again? I don't know why, but just have a try.

B.  R.
LIWEI WANG

PlutusCN, 2019-02-15:
@wangliwei-intel 

I build it on Ubuntu-16.04 LTS x64, the graphics card is Intel® HD Graphics 530 (Skylake GT2) , and the CPU is Intel® Core™ i7-6700 CPU @ 3.40GHz × 8 . 

But the problem does still exist, I do not know how to fix.

feijiang1, 2019-02-18:
Double check with PlutusCN, found the problem he encountered was not same as the problem @wangliwei-intel reported. So it should not be ldconfig related. In PlutusCN provided compiling log, there was:

Expected<T> must be checked before access or destruction.
Expected<T> value was in success state. (Note: Expected<T> values in success mode must still be checked prior to being destroyed).
Aborted (core dumped)
igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen8core.dir/build.make:61: recipe for target 'bin/scheduler/x64/gen8/scheduler_Gen8core.bin' failed
make[2]: *** [bin/scheduler/x64/gen8/scheduler_Gen8core.bin] Error 134
CMakeFiles/Makefile2:858: recipe for target 'igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen8core.dir/all' failed
make[1]: *** [igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen8core.dir/all] Error 2

Although it also failed for scheduler_Gen8core.bin. While  the error is "Expected<T> must be checked before access or destruction" which can be found in https://github.com/llvm-mirror/llvm/blob/master/include/llvm/Support/Error.h. The llvm is built from source code, it seems installed llvm didn't match with current neo code.

Do you have any idea for such problem? If it is still NEO related, should we create another issue to track it separately?

PlutusCN, 2019-02-18:
@JacekDanecki 
```
~/Dev/source/build_neo$ make -n -f igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen9core.dir/build.make igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen9core.dir/build

/usr/bin/cmake -E cmake_echo_color --switch= --blue --bold --progress-dir=/home/dengliqiang/Dev/source/build_neo/CMakeFiles --progress-num= "Generating ../../bin/scheduler/x64/gen9/scheduler_Gen9core.bin"
cd /home/dengliqiang/Dev/source/neo/runtime/scheduler && LD_LIBRARY_PATH=/home/dengliqiang/Dev/source/build_neo/bin /home/dengliqiang/Dev/source/build_neo/bin/ocloc -q -file scheduler.cl -device skl -cl-intel-greater-than-4GB-buffer-required -64 -out_dir /home/dengliqiang/Dev/source/build_neo/bin/scheduler/x64/gen9 -cpp_file -options -I/usr/local/include/igc\ -I/usr/local/include/igc/cif\ -I/usr/local/include/igc/ocl_igc_shared/executable_format\ -I/usr/local/include/igc/ocl_igc_shared/device_enqueue\ -I\ ../gen9\ -cl-kernel-arg-info\ -cl-std=CL2.0

~/Dev/source/build_neo$ cd /home/dengliqiang/Dev/source/neo/runtime/scheduler && LD_LIBRARY_PATH=/home/dengliqiang/Dev/source/build_neo/bin /home/dengliqiang/Dev/source/build_neo/bin/ocloc -q -file scheduler.cl -device skl -cl-intel-greater-than-4GB-buffer-required -64 -out_dir /home/dengliqiang/Dev/source/build_neo/bin/scheduler/x64/gen9 -cpp_file -options -I/usr/local/include/igc\ -I/usr/local/include/igc/cif\ -I/usr/local/include/igc/ocl_igc_shared/executable_format\ -I/usr/local/include/igc/ocl_igc_shared/device_enqueue\ -I\ ../gen9\ -cl-kernel-arg-info\ -cl-std=CL2.0

Expected<T> must be checked before access or destruction.
Expected<T> value was in success state. (Note: Expected<T> values in success mode must still be checked prior to being destroyed).
已放弃 (核心已转储)
```

Does the above error message show anything wrong about:
Expected<T> must be checked before access or destruction.
Expected<T> value was in success state. (Note: Expected<T> values in success mode must still be checked prior to being destroyed).
??? 

The detailed compilation message:
[compile_neo_error.log](https://github.com/intel/compute-runtime/files/2874139/compile_neo_error.log)

BTW: I build it on Ubuntu-16.04 LTS x64, the graphics card is Intel® HD Graphics 530 (Skylake GT2) , and the CPU is Intel® Core™ i7-6700 CPU @ 3.40GHz × 8 .

JacekDanecki, 2019-02-18:
Can you provide what commits ID's were used to compile IGC components (igc, llvm-patches, opencl-clang). Neo currently is using older versions of IGC components, see [releases page](https://github.com/intel/compute-runtime/releases):
- intel/intel-graphics-compiler@31b9624
- intel/llvm-patches@5fc26d8
- intel/opencl-clang@8248120

feijiang1, 2019-02-19:
For IGC compiling, we mainly follow the steps indicated in wiki: https://github.com/intel/intel-graphics-compiler:
intel/intel-graphics-compiler@ 64632f, which is tag igc_release_2019-01-05
intel/llvm-patches@ 703b03
intel/opencl-clang@824812, which is branch ocl-open-40, different according to https://github.com/intel/intel-graphics-compiler/issues/44

additionally:
llvm_source@ c8fccc, which is branch release_40
llvm_patches@ 703b03

Just found IGC README.md has been changed a lot during last two weeks, switching to use branch release_70 for llvm. We will try that now.

PlutusCN, 2019-02-19:
@JacekDanecki 

The following is my commits ID's: 
```
workspace
      |- gmmlib                 https://github.com/intel/gmmlib @ 413896ed8e7ead3dd1c0fea9a4fe7f8326b7b9ff
      |- libdrm                 https://anongit.freedesktop.org/git/mesa/drm 
      |- libva                  https://github.com/intel/libva @ c9bb65b673fb62ba5dace005699901c91e90abab
      |- llvm7.0.0_source https://github.com/llvm-mirror/llvm 
            |- projects/common-clang  https://github.com/intel/opencl-clang
            |- projects/llvm-spirv    https://github.com/KhronosGroup/SPIRV-LLVM-Translator
            |- tools/clang            https://github.com/llvm-mirror/clang
      |- llvm_patches https://github.com/intel/llvm-patches @ 703b033a0d4d2f555cd7fea35c9634aa5786585a 
      |- clang_source           https://github.com/llvm-mirror/clang 
      |- common_clang           https://github.com/intel/opencl-clang
      |- llvm_source            https://github.com/llvm-mirror/llvm
      |- igc                    https://github.com/intel/intel-graphics-compiler @ 312ccf2cdcf44463c0dd3b66c402a7e644e19f22
      |- opencl_headers         https://github.com/KhronosGroup/OpenCL-Headers @ f039db6764d52388658ef15c30b2237bbda49803 
      |- neo                    https://github.com/intel/compute-runtime @ fae6c5fce2701fa0cc82711127ce2d24a19cbc9a 
```

The following is what I have gotten.
```
$ git clone https://github.com/intel/gmmlib.git 
$ git clone https://anongit.freedesktop.org/git/mesa/drm.git libdrm 
$ git clone https://github.com/intel/libva.git 
$ git clone https://github.com/KhronosGroup/OpenCL-Headers opencl_headers
$ git clone -b release_70 https://github.com/llvm-mirror/llvm llvm7.0.0_source
$ git clone -b release_70 https://github.com/llvm-mirror/clang llvm7.0.0_source/tools/clang 
$ git clone -b llvm_release_70 https://github.com/KhronosGroup/SPIRV-LLVM-Translator.git llvm7.0.0_source/projects/llvm-spirv 
$ git clone -b ocl-open-70 https://github.com/intel/opencl-clang.git llvm7.0.0_source/projects/common-clang 
$ git clone https://github.com/intel/llvm-patches llvm_patches 
$ git clone -b release_40 https://github.com/llvm-mirror/clang clang_source
$ git clone -b ocl-open-40 https://github.com/intel/opencl-clang common_clang
$ git clone -b release_40 https://github.com/llvm-mirror/llvm llvm_source 
$ git clone -b igc_release_2018-12-12 https://github.com/intel/intel-graphics-compiler igc && cd igc && git checkout -b igc_release_2018-12-12
```

The following is my compilation steps:
```
cd build_gmmlib && cmake ${WORKSPACE}/gmmlib && make -j8
cd libdrm && ./autogen.sh && make -j8
cd libva && ./autogen.sh && make -j8
cd build_common_clang && cmake -DLLVM_TARGETS_TO_BUILD=X86 ../llvm7.0.0_source && make common_clang -j8
cd build_igc && cmake -DIGC_OPTION__OUTPUT_DIR=../igc-install/Release ../igc/IGC && make -j8 
sudo ldconfig 
cd build_neo && cmake -DCMAKE_BUILD_TYPE=Release ../neo && make -j8
```

JacekDanecki, 2019-02-19:
@feijiang1 I've compiled today the whole stack using below revisions
```
gmmlib
8bee050 (HEAD -> master) Adding Lakefield support

llvm_source
dd3329aeb25 (HEAD -> release_70) cmake: Use 7.1.0 for shared object version instead of 71.0 

llvm_source/tools/clang
4519e2637f (HEAD -> release_70) Merging r345839:

llvm_source/projects/opencl-clang
6257ffe (HEAD -> ocl-open-70) Fix realloc_buffer function

llvm_source/projects/llvm-spirv
8ce6443 (HEAD -> llvm_release_70) Enable publishing of built artifacts via github releases.

llvm_patches
3d4449f (HEAD -> master) Correct llvm_patcher

igc
b0409b8 (HEAD -> master) Add query for CPS

neo
aa587b3b (HEAD -> master) Pass HardwareInfo to AubHelper::checkPTEAddress()
```

JacekDanecki, 2019-02-19:
@PlutusCN IGC was switched to llvm/clang 7, so please use updated [README.md](https://github.com/intel/intel-graphics-compiler/blob/master/README.md) file for IGC build instruction.

PlutusCN, 2019-02-20:
@JacekDanecki 
Did you have this error [Unable to compile OpenCL kernels after installing IGC](https://github.com/intel/intel-graphics-compiler/issues/63) during compilation ?

feijiang1, 2019-02-20:
Tried following commit id, still encountered same problem when compile compute-runtime. 
intel/intel-graphics-compiler@31b9624
intel/llvm-patches@5fc26d8
intel/opencl-clang@8248120
=>
"Expected must be checked before access or destruction.
Expected value was in success state. (Note: Expected values in success mode must still be checked prior to being destroyed).
Aborted (core dumped)"

Then I tried release 19.01.12103, still encountered same problem.
While if I install the intel-igc-opencl_18.50.1270_amd64.deb and intel-igc-core_18.50.1270_amd64.deb, computer-runtime can be compiled successfully. So it can indicate my IGC compiling has some mistake.
While I totally follow the steps in https://github.com/intel/intel-graphics-compiler/tree/31b96245d0ede86fa5a55537f19cd337a3062310, and check corresponding commit id indicated in release. Can you help check if your compiling steps have any difference compared the steps listed in README.md?

JacekDanecki, 2019-02-20:
@feijiang1 Do you have any llvm binaries installed on your system? You've mentioned you're using llvm built from source code. If so please remove/rename llvm/clang binaries before IGC compilation.

feijiang1, 2019-02-21:
@JacekDanecki I tried to remove llvm/clang packages, but still encountered same problem. Does it mean you didn't use llvm from source code when release neo and igc? If yes, then it should be the root cause. Can you help share the llvm binary you are using and the detailed method of compiling IGC from binary instead of source code?

PlutusCN, 2019-02-21:
I found an interesting phenomenon: 

(1) It is failed that compiling neo only based on source code of IGC and others; 
(2) It is also failed that compiling neo only based on .deb files from [19.03.12192](https://github.com/intel/compute-runtime/releases/tag/19.03.12192), the following is wrong message: 
```
~/Dev/install/neo/build$ cmake -DCMAKE_BUILD_TYPE=Release ../neo

-- The C compiler identification is GNU 5.5.0
-- The CXX compiler identification is GNU 5.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Release build configuration
-- Source Level Debugger headers dir: /home/dengliqiang/Dev/install/neo/neo/third_party/source_level_debugger
-- Aub Stream Headers dir: /home/dengliqiang/Dev/install/neo/neo/third_party/aub_stream/headers
-- i915 includes dir: /home/dengliqiang/Dev/install/neo/neo/third_party/uapi
-- Khronos OpenCL headers dir: /home/dengliqiang/Dev/install/neo/neo/third_party/opencl_headers
-- Khronos OpenGL headers dir: /home/dengliqiang/Dev/install/neo/neo/third_party/opengl_headers
-- Third party dir: /home/dengliqiang/Dev/install/neo/third_party
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'igc-opencl'
--   No package 'igc-opencl' found
CMake Error at CMakeLists.txt:241 (message):
  Intel Graphics Compiler not found!

-- Configuring incomplete, errors occurred!

See also "/home/dengliqiang/Dev/install/neo/build/CMakeFiles/CMakeOutput.log".

~/Dev/install/neo/build$ apt-cache search intel-gmmlib*
intel-gmmlib - Intel(R) Graphics Memory Management Library Package

~/Dev/install/neo/build$ apt-cache search intel-igc*
intel-igc-opencl - Intel(R) Graphics Compiler for OpenCL(TM)
intel-igc-core - Intel(R) Graphics Compiler for OpenCL(TM)

~/Dev/install/neo/build$ apt-cache search intel-opencl*
intel-opencl - Intel OpenCL GPU driver

```

neo can be compiled successfully after install IGC from source code and .deb. 

Should I do like this ?  Compiling neo based on IGC source code and .deb files. 

Thanks a lot to @JacekDanecki

JacekDanecki, 2019-02-21:
@feijiang1 I've compiled IGC with llvm sources with commits ID's provided earlier. Currently there is bug in IGC when you try to build IGC components on system with installed llvm/clang 7 binaries. The fix should be soon with some improvements regarding building IGC in different configurations.
@PlutusCN You can build Neo when IGC/gmmlib binaries/headers/pkgconfig files are installed in the system. To resolve second issue (No package 'igc-opencl' found) you can:
- create and install all packages for gmmlib and IGC (including -devel packages). You can use command
```
make -j `nproc` package
```
In Neo releases we don't provide (at this moment) -devel packages for IGC and gmmlib. 
After package installation make sure, linker cache can find igc/gmmlib components (ldconfig -p), and pkg-config can find igc/gmmlib packages:
```
$ export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig:/usr/local/lib64/pkgconfig
$ pkg-config --list-all | grep -E "gmm|igc"
igc-opencl                     igc-opencl - Intel(R) Graphics Compiler for OpenCL(TM)
igdgmm                         igdgmm - Intel(R) Graphics Memory Management Library
```

I've created [wiki page](https://github.com/JacekDanecki/compute-runtime/wiki/How-to-build-Neo-components) with commands I've used to build all components needed by Neo. I hope it can help.

feijiang1, 2019-02-22:
@JacekDanecki Big thanks for your support. According to your new wiki, I can build IGC and NEO successfully. The problem I encountered seems caused by opencl-clang built separately previously, although I uninstalled llvm/clang package by "dpkg -P", but previous opencl-clan is install by "make install", I should use "xargs rm < install_manifest.txt" to uninstall all of them. I am newbie for cmake and opencl, sorry for the inconvenience brought to you and thanks you again for patient replies.

PlutusCN, 2019-02-24:
@JacekDanecki You are so nice, and I am so appreciated. According to your new wiki,I also can build IGC and NEO successfully. But the system environment must be fine to build. After I removed binary/include files of llvm/clang, I got the mistake: [error: unknown type name 'intel_sub_group_avc_mce_payload_t'
intel_sub_group_avc_mce_payload_t **attribute**((overloadable))](https://github.com/intel/intel-graphics-compiler/issues/64), and still do not know how to fix. So I changed to another laptop for building, that is OK. Please tell me if you know how to solve this mistake.  Thank you can not any more.

PlutusCN, 2019-02-25:
@JacekDanecki Thank you very much for your help. And according to your new wiki, I also build IGC and NEO successfully.

JacekDanecki, 2019-02-25:
@PlutusCN I've seen you'd observed issue:
```
error: unknown type name 'intel_sub_group_avc_mce_payload_t'
intel_sub_group_avc_mce_payload_t attribute((overloadable)),
```
It can be related to problems with patching. You should see logs similar to below after you run cmake for IGC build:
```
Switched to a new branch 'ocl-open-70'
T       test/Driver/Inputs/CUDA-symlinks/usr/bin/ptxas
T       test/Driver/Inputs/basic_cross_linux_tree/usr/bin/i386-unknown-linux-gnu-ld
T       test/Driver/Inputs/basic_cross_linux_tree/usr/bin/x86_64-unknown-linux-gnu-ld
T       test/Driver/Inputs/basic_cross_linux_tree/usr/i386-unknown-linux-gnu/bin/ld
T       test/Driver/Inputs/basic_cross_linux_tree/usr/x86_64-unknown-linux-gnu/bin/ld
T       test/Driver/Inputs/multilib_32bit_linux_tree/usr/bin/as
T       test/Driver/Inputs/multilib_32bit_linux_tree/usr/bin/ld
T       test/Driver/Inputs/multilib_32bit_linux_tree/usr/i386-unknown-linux/bin/as
T       test/Driver/Inputs/multilib_32bit_linux_tree/usr/i386-unknown-linux/bin/ld
T       test/Driver/Inputs/multilib_64bit_linux_tree/usr/bin/as
T       test/Driver/Inputs/multilib_64bit_linux_tree/usr/bin/ld
T       test/Driver/Inputs/multilib_64bit_linux_tree/usr/x86_64-unknown-linux/bin/as
T       test/Driver/Inputs/multilib_64bit_linux_tree/usr/x86_64-unknown-linux/bin/ld
Applying: Add cl_intel_planar_yuv extension
Applying: Unify ZeroToOCL* cast types
Applying: Add support of cl_intel_device_side_avc_motion_estimation extension
Applying: Relax diagnostics on OpenCL access qualifiers
Applying: Fix invalid address space generation for clk_event_t
-- /root/build/llvm/src/projects/llvm-spirv:
Switched to a new branch 'ocl-open-70'
Applying: Do not run LowerSPIRBlocks pass if there are no SPIR 2.0 blocks in the module.
Applying: Fix disabling of fp-contract
Applying: Preserve debug location attached to a builtin call.
Applying: Add no signed/unsigned wrap decoration support
-- Looking for sys/resource.h
-- Looking for sys/resource.h - found
-- Clang version: 7.1.0

```

JacekDanecki, 2019-02-25:
Closing as Neo can be compiled using [wiki page](https://github.com/JacekDanecki/compute-runtime/wiki/How-to-build-Neo-components)

PlutusCN, 2019-03-02:
@JacekDanecki 
I have no: 
```
Applying: Add cl_intel_planar_yuv extension
Applying: Unify ZeroToOCL* cast types
Applying: Add support of cl_intel_device_side_avc_motion_estimation extension
Applying: Relax diagnostics on OpenCL access qualifiers
Applying: Fix invalid address space generation for clk_event_t

Applying: Do not run LowerSPIRBlocks pass if there are no SPIR 2.0 blocks in the module.
Applying: Fix disabling of fp-contract
Applying: Preserve debug location attached to a builtin call.
Applying: Add no signed/unsigned wrap decoration support
```

What that mean ?

JacekDanecki, 2019-03-04:
This means that patches from directory common_clang/patches were not applied.
There is set 9 patches in  [common_clang repository](https://github.com/intel/opencl-clang/tree/ocl-open-70/patches):
* 5 for [clang](https://github.com/llvm-mirror/clang.git)
* 4 for [spirv](https://github.com/KhronosGroup/SPIRV-LLVM-Translator.git)

The latest changes in [IGC](https://github.com/intel/intel-graphics-compiler/commit/606f78808a0e88a47a1d3811db55ae44f8d28d9c) allow to build IGC with system llvm/clang without patches.

Can you share output from cmake command executed to configure IGC?

PlutusCN, 2019-03-05:
@JacekDanecki 
The following is my compiling environment: 
```
git clone https://github.com/intel/llvm-patches llvm_patches @3d4449ff6f8ae8b8f1f4258c36a57f77f9ca8491 
git clone -b release_70 https://github.com/llvm-mirror/llvm llvm_source @dd3329aeb25d87d4ac6429c0af220f92e1ba5f26 
git clone -b release_70 https://github.com/llvm-mirror/clang llvm_source/tools/clang @4519e2637fcc4bf6e3049a0a80e6a5e7b97667cb 
git clone -b llvm_release_70 https://github.com/KhronosGroup/SPIRV-LLVM-Translator.git llvm_source/projects/llvm-spirv @8ce6443ec1020183eafaeb3410c7d1edc2355dc3 
git clone -b ocl-open-70 https://github.com/intel/opencl-clang.git llvm_source/projects/opencl-clang @35f893bb4acb3c79bc6bfed18d23bc2714cd84ab 
git clone https://github.com/intel/intel-graphics-compiler igc @606f78808a0e88a47a1d3811db55ae44f8d28d9c 
```

And the log for building: 
[make_igc_error.log](https://github.com/intel/compute-runtime/files/2929176/make_igc_error.log)

BTW: 
I chose 
opencl-clang @6257ffe137a2c8df95a3f3b39fa477aa8ed15837 
igc @b0409b893508e55d9548a1d6658788e7cd9963bd
before now, and the result of compilation was also like the above log file.

JacekDanecki, 2019-03-05:
The latest changes on IGC side allow to build IGC with system llvm/clang 7 without applying patches. 
Such configuration requires additional parameter for IGC cmake:
```
-DVME_TYPES_DEFINED=FALSE
```
@PlutusCN In log you've provided I can see you have llvm/clang 7.1.0 installed on your system in /usr/local directory.
Have you applied any patched on this llvm? How did you build and install this version?
If you build llvm with opencl-clang project inside llvm sources, patches would be applied already during llvm build
```
-- Constructing LLVMBuild project information
-- Linker detection: GNU ld
-- Targeting AArch64
-- Targeting AMDGPU
-- Targeting ARM
-- Targeting BPF
-- Targeting Hexagon
-- Targeting Lanai
-- Targeting Mips
-- Targeting MSP430
-- Targeting NVPTX
-- Targeting PowerPC
-- Targeting Sparc
-- Targeting SystemZ
-- Targeting X86
-- Targeting XCore
-- Linker detection: GNU ld
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- /root/open-source/llvm_source/tools/clang:
Switched to a new branch 'ocl-open-70'
Applying: Add cl_intel_planar_yuv extension
Applying: Unify ZeroToOCL* cast types
Applying: Add support of cl_intel_device_side_avc_motion_estimation extension
Applying: Relax diagnostics on OpenCL access qualifiers
Applying: Fix invalid address space generation for clk_event_t
-- /root/open-source/llvm_source/projects/llvm-spirv:
Switched to a new branch 'ocl-open-70'
Applying: Do not run LowerSPIRBlocks pass if there are no SPIR 2.0 blocks in the module.
Applying: Fix disabling of fp-contract
Applying: Preserve debug location attached to a builtin call.
Applying: Add no signed/unsigned wrap decoration support
```

If you built in this workspace already, patches could be applied earlier.
```
-- Linker detection: GNU ld
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- /root/open-source/llvm_source/tools/clang:
Already on 'ocl-open-70'
-- /root/open-source/llvm_source/projects/llvm-spirv:
Already on 'ocl-open-70
```

To use llvm 7.1.0, you can modify CMakeLists.txt as was mentioned in [this issue](https://github.com/intel/intel-graphics-compiler/issues/46#issuecomment-460900543).

But if you have llvm_patches directory created, you need to add below flag to force system llvm usage.
```
-DIGC_OPTION__FORCE_SYSTEM_LLVM =1
```

I've created another [wiki page](https://github.com/JacekDanecki/compute-runtime/wiki/How-to-build-Neo-components-(system-llvm-clang-7)) with instruction how to build Neo components using installed llvm 7.0.1

JacekDanecki, 2019-03-29:
Closing as new instructions were provided. Feel free to reopen or create new issue if you find any new problems.

closed 2019-03-29 by JacekDanecki

PlutusCN, 2019-04-17:
A ton of thanks to @JacekDanecki , I can compile the latest NEO successfully. And 
[build_intel_neo_v2.sh.txt](https://github.com/intel/compute-runtime/files/3087581/build_intel_neo_v2.sh.txt)
is my process, I wrote them into the script file.

And thanks to @JacekDanecki again.