intel/compute-runtime #125: Update invalidates cached kernels (darktable artifacts and crashes)

kxv, 2019-01-18:
OS: Linux (Manjaro, linux 4.20)
HW: Kaby Lake R / i7-8550U
[clinfo.txt](https://github.com/intel/compute-runtime/files/2772369/clinfo.txt)
Intel OpenCL NEO versions: 19.01 (one after update to it, the behavior occurred)

The RAW Photo Editing Software darktable uses cached kernels to their OpenCL image processing.
A update of the compute runtime seems to invalidate those and leads to sudden artifacts and crashes in darktable.
A manual deletion of the cached kernels and subsequent regeneration seems to lead to stable behavior.

Please see https://redmine.darktable.org/issues/12541 for the corresponding report at darktable.

alalek, 2019-01-18:
> manual deletion of the cached kernels

Is it darktable own cache or is it about builtin "cl_cache" directory?

---

Usually upgrading of OpenCL implementation upgrades used OpenCL compiler (with fixing bugs, races or new optimizations passes), so kernel binaries may be updated too.
Assumption, that kernel binary cache should stay the same with runtime upgrade is not valid.

Updates dependencies => invalidate related caches.

---

<details>

<summary>OpenCL cache in OpenCV</summary>

OpenCV uses new cache directory for new runtime version:
`<platform_name>--<device_name>--<device_version>`
```
$ ls -1 ~/.cache/opencv/4.0-dev/opencl_cache/
Intel_R__Corporation--Intel_R__Gen9_HD_Graphics_NEO--18_33_11309
Intel_R__Corporation--Intel_R__Gen9_HD_Graphics_NEO--18_46_0
Intel_R__Corporation--Intel_R__Gen9_HD_Graphics_NEO--18_48_11934
```

(this is snapshot from dev machine: by default, we automatically clean other "non-current" caches and keep only latest cache directory - to avoid wasting of disk space on users machine)

</details>

kxv, 2019-01-19:
Thank you for the clarification and additional information.
The aforementioned darktable cache is located under  .cache/darktable/cached_kernels_for_IntelRGen9HDGraphicsNEO/ for my system. This path does not seem to be runtime-version dependent.

kxv, 2019-01-23:
I have proposed a similar cache dir naming for darktable in https://github.com/darktable-org/darktable/pull/2033 .

JacekDanecki, 2019-09-02:
As I can see issue was fixed (under Linux) in darktable, so closing issue here. Please reopen it, or create new one if you find new issue.

closed 2019-09-02 by JacekDanecki