intel/compute-runtime #128: Is development for Ivy Bridge architecture going to start soon?

ivanmlerner, 2019-02-09:
Hello, according to intel's page, this driver stack will substitute both closed source drivers and beignet, when is support for Ivy Bridge going to get in?

bfliflet, 2019-02-09:
There are no plans that I am aware of to internally back port this library for Gen7 class (ie. Ivybridge and Haswell) hardware.  There is a fair amount of runtime work but the bigger effort would be related to the compiler itself which was completely rearchitected in the Gen8/9 timeframe.  That said, there's nothing prohibiting the underlying compute-runtime SW architecture from supporting it through community provided submissions.

JacekDanecki, 2019-02-25:
Closing as answer was provided.

closed 2019-02-25 by JacekDanecki