intel/compute-runtime #130: Fine Grain SVM buffers

FePwR, 2019-02-11:
Hi!

I need to use SVM buffers. I'm trying to patch my kernel but I don't figure it out how. I don't know if this is the correct place to ask for help. I appreciate so much your help.

Thanks.

You can always add required patch manually: https://patchwork.kernel.org/patch/10638787/

Additional changes in OCL driver are also required: https://github.com/intel/compute-runtime/pull/53

_Originally posted by @BartoszDunajski in https://github.com/intel/compute-runtime/issues/105#issuecomment-436240899_

BartoszDunajski, 2019-02-12:
Hello,
To use fine grain SVM you have to apply patch on OCL driver (https://github.com/intel/compute-runtime/pull/53) and mentioned Linux Kernel patch.
Please refer to https://github.com/intel/compute-runtime/issues/105. Issue is already well described there.

FePwR, 2019-02-12:
Ok! I'll keep trying.
Thank you, @BartoszDunajski .

closed 2019-02-12 by FePwR