intel/compute-runtime #131: Please update libva version used in release build

ruiling, 2019-02-15:
Currently the release build still use very old libva version, this is not friendly to users who want to use buffer sharing between va-api and opencl. please consider updating to libva latest release version regularly. Thanks!

AdamCetnerowski, 2019-02-21:
Thank you for the suggestion. Request is in our queue. We will update this issue, once it's done.

ArturHarasimiuk, 2019-03-29:
@ruiling could you please elaborate more on what problem are you observing? va_api_major_version was not changed between proposed revisions thus it doesn't look that update on our side is required.

ruiling, 2019-04-02:
@ArturHarasimiuk I double check my issue (I meet segfault when using Neo with media-driver for buffer sharing). And I find that my issue was caused by gmmlib version instead of libva version. Neo release binary packages works with specific gmmlib version.  But in order to use media-driver,  I have to build gmmlib from source code again. And then I unintentionally overwrite gmmlib with an incompatible version. Then segfault occurs.
Back to the libva version, I think Neo binary package should work with libva 2.x, right? But I have not seen any material mention about this. I think it's better at the release page, add one note saying "although we use specific libva to build the package, it should work with other libva 2.x version".

ArturHarasimiuk, 2019-04-11:
Thank you for comments. We will consider doing this and adding more information.

closed 2019-04-11 by ArturHarasimiuk