intel/compute-runtime #132: Driver reset on 1804 and 6700hq

isgursoy, 2019-02-15:
Machine is http://www.vorke.com/project/vorke-v6/

OS is Linux 4.15.0-45-generic #48-Ubuntu SMP Tue Jan 29 16:28:13 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

I tried all releases, result is same.

dmesg:
[377999.628419] [drm] GPU HANG: ecode 9:0:0x85ddfffb, in luxmark.bin [20464], reason: Hang on rcs0, action: reset
[377999.628425] i915 0000:00:02.0: Resetting rcs0 after gpu hang

```
$ sudo cat /sys/class/drm/card0/error 
GPU HANG: ecode 9:0:0x85ddfffb, in luxmark.bin [20464], reason: Hang on rcs0, action: reset
Kernel: 4.15.0-45-generic
Time: 1550268925 s 731763 us
Boottime: 377998 s 780104 us
Uptime: 377996 s 217560 us
Active process (on ring render): luxmark.bin [20464], score 0
Reset count: 0
Suspend count: 0
Platform: SKYLAKE
PCI ID: 0x191b
PCI Revision: 0x06
PCI Subsystem: 1991:5594
IOMMU enabled?: 0
DMC loaded: yes
DMC fw version: 1.26
GT awake: yes
RPM wakelock: yes
PM suspended: no
EIR: 0x00000000
IER: 0x08000000
GTIER[0]: 0x01010101
GTIER[1]: 0x01010101
GTIER[2]: 0x00000070
GTIER[3]: 0x00000101
PGTBL_ER: 0x00000000
FORCEWAKE: 0xffff0001
DERRMR: 0x2077efef
CCID: 0x00000000
Missed interrupts: 0x00000000
  fence[0] = 00000000
  fence[1] = 00000000
  fence[2] = 00000000
  fence[3] = 00000000
  fence[4] = 00000000
  fence[5] = 00000000
  fence[6] = 00000000
  fence[7] = 00000000
  fence[8] = 00000000
  fence[9] = 00000000
  fence[10] = 00000000
  fence[11] = 00000000
  fence[12] = 00000000
  fence[13] = 00000000
  fence[14] = 00000000
  fence[15] = 00000000
  fence[16] = 00000000
  fence[17] = 00000000
  fence[18] = 00000000
  fence[19] = 00000000
  fence[20] = 00000000
  fence[21] = 00000000
  fence[22] = 00000000
  fence[23] = 00000000
  fence[24] = 00000000
  fence[25] = 00000000
  fence[26] = 00000000
  fence[27] = 00000000
  fence[28] = 00000000
[... 344 lines cut ...]
i915.enable_guc_loading=0
i915.enable_guc_submission=0
i915.guc_log_level=-1
i915.guc_firmware_path=(null)
i915.huc_firmware_path=(null)
i915.mmio_debug=0
i915.edp_vswing=0
i915.reset=2
i915.inject_load_failure=0
i915.alpha_support=no
i915.enable_cmd_parser=yes
i915.enable_hangcheck=yes
i915.fastboot=no
i915.prefault_disable=no
i915.load_detect_test=no
i915.force_reset_modeset_test=no
i915.error_capture=yes
i915.disable_display=no
i915.verbose_state_checks=yes
i915.nuclear_pageflip=no
i915.enable_dp_mst=yes
i915.enable_dpcd_backlight=no
i915.enable_gvt=no

```

JacekDanecki, 2019-02-18:
What version of luxmark are you using? Have you tried to disable GPU watchdog to check whether it's real gpu hang or gpu workload is to big? To disable gpu watchdog you can run as root below command
```
echo N > /sys/module/i915/parameters/enable_hangcheck
```

isgursoy, 2019-02-21:
- Luxmark is v3.1. Everything is fine on identical hardware and Ubuntu 16.04 with that Luxmark. Sadly I can not check clean 18.04. Only 18.04 that I can run on this machine is mine.

- I tried disabling watchdog, gpu is really hanging.

JacekDanecki, 2019-02-22:
What scene are you using in luxmark?
What version of OpenCV are you using? Can you provide steps to reproduce gpu hang in OpenCV?
Are you using Xorg/Wayland? What desktop environment/window manager?

isgursoy, 2019-02-22:
Hi,
- All scenes are doing the same.
- OpenCV is 4.0.1. I will attach an OpenCV specific example case soon but hang occurs in all OpenCL use attempts.
- Xorg. Gnome. Ubuntu 18.04 default desktop environment is running.

JacekDanecki, 2019-02-25:
Can you compare BIOS settings on these 2 machines, especially power management settings and GPU configurations? Are there the same kernel parameters set in kernel command line (grub)? Do you have the same packages for power management installed on Ubuntu 16.04 and 18.04?
Can you compare on both machines during luxmark test (before hang) as root in file /sys/kernel/debug/dri/0/i915_frequency_info information about GPU frequency (if Intel GPU is configured as 0 drm device)?
```
Lowest (RPN) frequency: 350MHz
Nominal (RP1) frequency: 350MHz
Max non-overclocked (RP0) frequency: 1100MHz
Max overclocked frequency: 1100MHz
Current freq: 1100 MHz
Actual freq: 1100 MHz
Idle freq: 350 MHz
Min freq: 350 MHz
Boost freq: 1100 MHz
Max freq: 1100 MHz
```
Have you tried to run [clpeak](https://github.com/krrishnarraj/clpeak)?

As I can see there is also Nvidia card in this machine? Have you configured it also? Do you have nvidia or nouveau kernel modules installed/loaded? Have you enabled opencl for this card?
What is the output from commands:
```
sudo lspci -v
ls -l /dev/dri/by-path
clinfo
```

MichalMrozek, 2019-03-29:
Closing due to inaction, feel free to reopen.

closed 2019-03-29 by MichalMrozek

isgursoy, 2019-03-31:
I had not time to check yet. I will be back here with outputs soon. Thanks.