intel/compute-runtime #133: Can't build runtime/built_ins/kernels

nandojve, 2019-02-18:
I've been trying to build NEO at Debian 9 to be installed at /opt/osgeo without success.
All dependencies have been build from source without problem from recent versions (02/18/2019).

```
Generating ../../../bin/built_ins/x64/gen10/aux_translation_Gen10core.spv, ../../../bin/built_ins/x64/gen10/aux_translation_Gen10core.bin, ../../../bin/built_ins/x64/gen10/aux_translation_Gen10core.cpp, ../../../bin/built_ins/x64/gen10/aux_translation_Gen10core.gen
```

First **ocloc** failed without print any error. I detected that **libigdfcl.so** ins't find by **ocloc** app. I added to **ld.config** the /opt/osgeo path with:
```
create /etc/ld.so.conf.d/libigdfcl.conf file
cat /etc/ld.so.conf.d/libigdfcl.conf
# lib* default configuration
/opt/osgeo/lib

update cache
sudo ldconfig -p | grep libigdfcl.so
```

Now, library it's ok but I'm facing the same problem at same point with this trace:

```
gfbudke@g1513-fox7:~/osgeo/opencl/intel/neo/build$ make
[  0%] Built target gmock-gtest
[  0%] Built target elflib
[  2%] Built target ocloc
[  2%] Built target igdrcl_lib_release_sharings_enable
[  2%] Built target biksim
[ 19%] Built target igdrcl_lib_release
[ 19%] Built target copy_compiler_files
[ 19%] Generating ../../../bin/built_ins/x64/gen10/aux_translation_Gen10core.spv, ../../../bin/built_ins/x64/gen10/aux_translation_Gen10core.bin, ../../../bin/built_ins/x64/gen10/aux_translation_Gen10core.cpp, ../../../bin/built_ins/x64/gen10/aux_translation_Gen10core.gen
[0]: /home/gfbudke/osgeo/opencl/intel/neo/build/bin/ocloc(_ZN16SafetyGuardLinux9sigActionEiP9siginfo_tPv+0x36) [0x565523726fe6]
[1]: /lib/x86_64-linux-gnu/libpthread.so.0(+0x110e0) [0x7f4e6506d0e0]
[2]: /lib64/ld-linux-x86-64.so.2(+0x15571) [0x7f4e65492571]
[3]: /lib64/ld-linux-x86-64.so.2(+0xf644) [0x7f4e6548c644]
[4]: /lib/x86_64-linux-gnu/libdl.so.2(+0x1531) [0x7f4e6527a531]
[5]: /lib/x86_64-linux-gnu/libdl.so.2(dlclose+0x1f) [0x7f4e65279fdf]
[6]: /opt/osgeo/lib/libigdfcl.so(+0x2bb27) [0x7f4e64123b27]
[7]: /opt/osgeo/lib/libigdfcl.so(+0x37aa0) [0x7f4e6412faa0]
[8]: /home/gfbudke/osgeo/opencl/intel/neo/build/bin/ocloc(_ZN5OCLRT15OfflineCompiler15buildSourceCodeEv+0xf9) [0x56552371c209]
[9]: /home/gfbudke/osgeo/opencl/intel/neo/build/bin/ocloc(_ZN5OCLRT15OfflineCompiler5buildEv+0xe) [0x565523721b3e]
[10]: /home/gfbudke/osgeo/opencl/intel/neo/build/bin/ocloc(_ZN16SafetyGuardLinux4callIiN5OCLRT15OfflineCompilerEMS2_FivEEET_PT0_T1_S5_+0x7c) [0x5655237270bc]
[11]: /home/gfbudke/osgeo/opencl/intel/neo/build/bin/ocloc(_Z20buildWithSafetyGuardPN5OCLRT15OfflineCompilerE+0x7b) [0x565523726f8b]
[12]: /home/gfbudke/osgeo/opencl/intel/neo/build/bin/ocloc(main+0x265) [0x56552370ca45]
[13]: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f4e644402e1]
[14]: /home/gfbudke/osgeo/opencl/intel/neo/build/bin/ocloc(_start+0x2a) [0x56552370cc2a]
Aborted
igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen10core.dir/build.make:72: recipe for target 'bin/built_ins/x64/gen10/aux_translation_Gen10core.spv' failed
make[2]: *** [bin/built_ins/x64/gen10/aux_translation_Gen10core.spv] Error 134
CMakeFiles/Makefile2:927: recipe for target 'igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen10core.dir/all' failed
make[1]: *** [igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen10core.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

Using **make VERBOSE=1** I found the command line and executed **ocloc** from **gdb** to check stack trace and received this:

```
gfbudke@g1513-fox7:~/osgeo/opencl/intel/neo/build$ cd /home/gfbudke/osgeo/opencl/intel/neo/runtime/built_ins/kernels
gfbudke@g1513-fox7:~/osgeo/opencl/intel/neo/runtime/built_ins/kernels$ LD_LIBRARY_PATH=/home/gfbudke/osgeo/opencl/intel/neo/build/bin gdb --args /home/gfbudke/osgeo/opencl/intel/neo/build/bin/ocloc -file aux_translation.igdrcl_built_in -device cnl -64 -out_dir /home/gfbudke/osgeo/opencl/intel/neo/build/bin/built_ins/x64/gen10 -cpp_file -options -cl-kernel-arg-info
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
[... 88 lines cut ...]
  Manually-specified variables were not used by the project:

    ARCH
    BUILD_TYPE

-- Build files have been written to: /home/gfbudke/osgeo/opencl/intel/neo/build
```

The igc-opencl is version 0.1.0 because is compiled from source and install at /opt/osgeo.

```
gfbudke@g1513-fox7:~/osgeo/opencl/intel/neo/runtime/built_ins/kernels$ cat /opt/osgeo/lib/pkgconfig/igc-opencl.pc 
prefix=/opt/osgeo
exec_prefix=${prefix}
includedir=${prefix}/include/igc
libdir=${exec_prefix}/lib

Name: igc-opencl
Description: Intel(R) Graphics Compiler for OpenCL(TM)
Version: 0.1.0
Cflags: -I${includedir} -I${includedir}/cif -I${includedir}/ocl_igc_shared/executable_format -I${includedir}/ocl_igc_shared/device_enqueue
Libs:
```

Any comment will be appreciated.

JacekDanecki, 2019-02-19:
Can you set environment variable:
```
export LD_DEBUG=libs
```
and check whether libcommon_clang.so or libopencl_clang.so library was initialized correctly. The name of this library depends on version of IGC you are using.

nandojve, 2019-02-20:
Thanks for the tip, problem solved!

After enabled *export LD_DEBUG=libs* I detected *libcommon_clang.so* isn't found. I added a rule to *ld.so.conf.d* but same behaviour. So, I inspected cache:

```
root@g1513-fox7:/etc# ldconfig -p | grep /opt
	libva.so.2 (libc6,x86-64) => /opt/osgeo/lib/libva.so.2
	libva.so (libc6,x86-64) => /opt/osgeo/lib/libva.so
	libva-x11.so.2 (libc6,x86-64) => /opt/osgeo/lib/libva-x11.so.2
	libva-x11.so (libc6,x86-64) => /opt/osgeo/lib/libva-x11.so
	libva-glx.so.2 (libc6,x86-64) => /opt/osgeo/lib/libva-glx.so.2
	libva-glx.so (libc6,x86-64) => /opt/osgeo/lib/libva-glx.so
	libva-drm.so.2 (libc6,x86-64) => /opt/osgeo/lib/libva-drm.so.2
	libva-drm.so (libc6,x86-64) => /opt/osgeo/lib/libva-drm.so
	libkms.so.1 (libc6,x86-64) => /opt/osgeo/lib/libkms.so.1
	libkms.so (libc6,x86-64) => /opt/osgeo/lib/libkms.so
	libigdgmm.so.7 (libc6,x86-64) => /opt/osgeo/lib/libigdgmm.so.7
	libigdgmm.so.5 (libc6,x86-64) => /opt/osgeo/lib/libigdgmm.so.5
	libigdgmm.so (libc6,x86-64) => /opt/osgeo/lib/libigdgmm.so
	libigdfcl.so (libc6,x86-64) => /opt/osgeo/lib/libigdfcl.so
	libigc.so (libc6,x86-64) => /opt/osgeo/lib/libigc.so
	libiga64.so (libc6,x86-64) => /opt/osgeo/lib/libiga64.so
	libdrm_radeon.so.1 (libc6,x86-64) => /opt/osgeo/lib/libdrm_radeon.so.1
	libdrm_radeon.so (libc6,x86-64) => /opt/osgeo/lib/libdrm_radeon.so
	libdrm_nouveau.so.2 (libc6,x86-64) => /opt/osgeo/lib/libdrm_nouveau.so.2
	libdrm_nouveau.so (libc6,x86-64) => /opt/osgeo/lib/libdrm_nouveau.so
	libdrm_intel.so.1 (libc6,x86-64) => /opt/osgeo/lib/libdrm_intel.so.1
	libdrm_intel.so (libc6,x86-64) => /opt/osgeo/lib/libdrm_intel.so
	libdrm_amdgpu.so.1 (libc6,x86-64) => /opt/osgeo/lib/libdrm_amdgpu.so.1
	libdrm_amdgpu.so (libc6,x86-64) => /opt/osgeo/lib/libdrm_amdgpu.so
	libdrm.so.2 (libc6,x86-64) => /opt/osgeo/lib/libdrm.so.2
	libdrm.so (libc6,x86-64) => /opt/osgeo/lib/libdrm.so
	libcommon_clang.so.7.1 (libc6,x86-64) => /opt/osgeo/lib/libcommon_clang.so.7.1
	libOpenCL.so.1 (libc6,x86-64) => /opt/osgeo/lib/libOpenCL.so.1
	libOpenCL.so (libc6,x86-64) => /opt/osgeo/lib/libOpenCL.so
```
As you can see only *libcommon_clang.so.7.1* was found but not *libcommon_clang.so*. So, after inspect the libraries I discover this:

```
-rw-r--r-- 1 root root 73373504 Feb 18 20:17 libcommon_clang.so
lrwxrwxrwx 1 root root       18 Feb 13 23:45 libcommon_clang.so.7.1 -> libcommon_clang.so

```
The library is not properly denominated when installed. Well, after rename properly system start to build properly.

```
lrwxrwxrwx 1 root root       22 Feb 19 21:13 libcommon_clang.so -> libcommon_clang.so.7.1
lrwxrwxrwx 1 root root       22 Feb 19 21:13 libcommon_clang.so.7 -> libcommon_clang.so.7.1
-rw-r--r-- 1 root root 73373504 Feb 18 20:17 libcommon_clang.so.7.1

```
Finally 4 files installed
```
Install the project...
-- Install configuration: "Release"
-- Installing: /opt/osgeo/lib/libigdrcl.so
-- Installing: /etc/ld.so.conf.d/libintelopencl.conf
-- Installing: /etc/OpenCL/vendors/intel.icd
-- Installing: /opt/osgeo/bin/ocloc

```

JacekDanecki, 2019-02-20:
Closing this issue, as the problem was with compiler component, and issue was created in IGC project.

closed 2019-02-20 by JacekDanecki