intel/compute-runtime #134: Build error on NEO patch version

FePwR, 2019-02-19:
Hello!
I continue trying to install the patch version https://github.com/intel/compute-runtime/issues/105#issuecomment-436240899

- [ ] Patch and install the kernel: https://patchwork.kernel.org/patch/10638787/

- [ ] Install gmmlib: [intel-gmmlib-18.3](https://github.com/intel/gmmlib/tree/intel-gmmlib-18.3
)

- [ ] Install IGC: [This commit](https://github.com/intel/compute-runtime/blob/3a16705386b011aacac4687caf9c30d6439e91eb/documentation/BUILD_Ubuntu.md)

- [ ] Install libdrm and libva: [link](https://github.com/intel/compute-runtime/blob/master/documentation/cl_intel_va_api_media_sharing.md)

- [ ] Neo with patch https://github.com/intel/compute-runtime/pull/53 , https://github.com/intel/compute-runtime/pull/53/commits/3a16705386b011aacac4687caf9c30d6439e91eb :  [This commit](https://github.com/intel/compute-runtime/tree/3a16705386b011aacac4687caf9c30d6439e91eb)

I get the next built error:

[built-error.txt](https://github.com/intel/compute-runtime/files/2880031/built-error.txt)

I tried almost everything. I read all the related issues but I can't find a solution. I really need SVM.

Thank you in advance!

JacekDanecki, 2019-02-19:
As I understand you're trying to build old Neo + svm patch with the newest IGC and old gmmlib. It can be problematic. Have you tried to rebase svm patch on Neo HEAD?

FePwR, 2019-02-19:
I made a mistake in the comment. I'm using [igc_release_2018-11-15](https://github.com/intel/intel-graphics-compiler/tree/igc_release_2018-11-15), with SKIP_SPIRV_TOOLS_INSTALL:BOOL=ON.

I don't know how to rebase svm patch on Neo HEAD, but I'll search about it.

Thanks, @JacekDanecki

JacekDanecki, 2019-02-19:
You can start with [18.25.10965 release](https://github.com/intel/compute-runtime/releases/tag/18.25.10965) as a baseline. So checkout all components using commits ID's provided in release notes, and then apply pull request on Neo repo:
```
$ git fetch origin +pull/53/head:svm
From https://github.com/intel/compute-runtime
 * [new ref]           refs/pull/53/head -> svm

$ git checkout svm
Switched to branch 'svm'

$ git rebase 18.25.10965
First, rewinding head to replay your work on top of it...
Applying: Control data port coherency at context level
Applying: Add I915_CONTEXT_PARAM_COHERENCY uapi define

```

FePwR, 2019-02-19:
That works!! I know what I was doing wrong. Thank you very much, @JacekDanecki .
A last question before close the issue.
I still cannot use SVM. I'm using Ubuntu 18.04.2. I suppose the built driver is too old, but I know how to build it now. 
Will you recommend me another release to prove with?

BartoszDunajski, 2019-02-20:
@FePwR, could you give us more details about your SVM issue after applying appropriate patches?

FePwR, 2019-02-20:
Any problem so far with Neo driver build. The problem is that _clinfo_ doesn't show me fine grain SVM buffer capabilities on integrated graphics card.
Maybe I made a mistake patching the kernel, don't know. The only kernel that I could [patch](https://patchwork.kernel.org/patch/10638787/) without errors was v5.0. @BartoszDunajski , Do you think that it's the only one.
Maybe I should close this issue because I don't have any problems building Neo.
Thanks!

BartoszDunajski, 2019-02-20:
Looks like patch is active, but Device info doesn't report CL_DEVICE_SVM_FINE_GRAIN_BUFFER.
You can find this logic in device.cpp: Device::initializeCaps(). You can try to force it. 
And make sure that new ioctl from OCL PR are called successfully.

FePwR, 2019-02-21:
I'm trying, but I get a core dumped when I use clSVMAlloc over the GPU. I forced CL_DEVICE_SVM_FINE_GRAIN_BUFFER in device_caps.cpp: Device::initializeCaps(). Now, I can see SVM capabilities with _clinfo_ .
How can I make that ioctl is called successfully?
Thank you for your help, @BartoszDunajski .

BartoszDunajski, 2019-02-21:
What is the reason of core dump? You have to use debugger to get more data.
For ioctls, also use debugger (or printf). Just make sure that error codes are ok.

BartoszDunajski, 2019-02-21:
1. clSVMAlloc returns valid pointer (not nullptr)? Failure hits on any specific offset?
2. Just find out in OCL patch where new ioctls are used and check error codes

FePwR, 2019-02-21:
Exactly! clSVMAlloc returns a null pointer. I forgot to check it.

FePwR, 2019-02-22:
Maybe I discover what is my problem. I think the kernel hasn't a good configuration. Could anyone show to me a configuration kernel file where SVM works fine?? ($ cat /boot/config-$(uname -r) > config.txt )
Thank you very much!

FePwR, 2019-02-27:
I continue without  fine-grained buffer capabilities. No errors in installation so far, but I have this when I compile igc:
[CMakeError.log](https://github.com/intel/compute-runtime/files/2909760/CMakeError.log)
Could any of these errors interfere?
Thanks!

BartoszDunajski, 2019-02-27:
It looks like IGC specific issue.
Please refer to https://github.com/intel/intel-graphics-compiler for support.

Does it mean that we can close this issue?

FePwR, 2019-02-27:
Yes, I close it.
Thank you for all the help and support.

closed 2019-02-27 by FePwR