intel/compute-runtime #135: OpenCL: Segmentation Fault and no idea anymore

gitdev-f11, 2019-02-20:
Installed Arch Linux on a Dell XPS 9360 with Intel HD Graphics 620.
Via Pacaur i've installed `compute-runtime-bin` and running `clinfo` ends everytime in a segmentation fault.
Now, after try and error for one week, i've no ideas what to do.

Here some infos

    $ clinfo
    Number of platforms                               1
    Platform Name                                   Intel(R) OpenCL HD Graphics
    Platform Vendor                                 Intel(R) Corporation
    Platform Version                                OpenCL 2.1 
    Platform Profile                                FULL_PROFILE
    Platform Extensions                             
    cl_khr_3d_image_writes cl_khr_byte_addressable_store       
    cl_khr_fp16 cl_khr_depth_images cl_khr_global_int32_base_atomics
    cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_image2d_from_buffer
    cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_subgroups  
    cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator  
    cl_intel_media_block_io cl_intel_driver_diagnostics cl_intel_device_side_avc_motion_estimation  
    cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_khr_fp64  
    cl_khr_subgroups cl_khr_il_program cl_intel_spirv_device_side_avc_motion_estimation  
    cl_intel_spirv_media_block_io cl_intel_spirv_subgroups cl_khr_mipmap_image 
    cl_khr_mipmap_image_writes cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_motion_estimation  
    cl_intel_advanced_motion_estimation cl_intel_va_api_media_sharing 
    Platform Host timer resolution                  1ns
    Platform Extensions function suffix             INTEL

    Platform Name                                   Intel(R) OpenCL HD Graphics
    Number of devices                                 1
    Device Name                                     Intel(R) Gen9 HD Graphics NEO
    Device Vendor                                   Intel(R) Corporation
    Device Vendor ID                                0x8086
    Device Version                                  OpenCL 2.1 NEO 
    Driver Version                                  19.05.12254
    Device OpenCL C Version                         OpenCL C 2.0 
    Device Type                                     GPU
    Device Profile                                  FULL_PROFILE
    Device Available                                Yes
    Compiler Available                              Yes
    Linker Available                                Yes
    Max compute units                               24
    Max clock frequency                             1150MHz
    Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
    Supported affinity domains                    (n/a)
    Max work item dimensions                        3
    Max work item sizes                             256x256x256
    Max work group size                             256
    Speicherzugriffsfehler (Speicherabzug geschrieben)
    ^^^^^^^^^^^^^^^^^^^^^^^ german for segmentation fault

    $ ldconfig -p | grep -i opencl
	libopencl_clang.so (libc6,x86-64) => /usr/lib/libopencl_clang.so
	libOpenCL.so.1 (libc6,x86-64) => /usr/lib/libOpenCL.so.1
	libOpenCL.so (libc6,x86-64) => /usr/lib/libOpenCL.so
	
    $ ldd /usr/bin/clinfo
	linux-vdso.so.1 (0x00007fffb1fa7000)
	libOpenCL.so.1 => /usr/lib/libOpenCL.so.1 (0x00007fb0e9920000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007fb0e991b000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007fb0e9757000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007fb0e9b82000)

     # the single *.icd-file
     $ cat /etc/OpenCL/vendors/intel-gpu.icd
	/usr/lib/libigdrcl.so

And the output of `strace /usr/bin/clinfo` as file: 
[strace.txt](https://github.com/intel/compute-runtime/files/2885549/strace.txt)

A little bit help to solve my problem would be really nice ;)

Helmut.

JacekDanecki, 2019-02-20:
Here is a problem:
```
openat(AT_FDCWD, "/usr/lib/libtinfo.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
```

libopencl_clang.so library from compute-runtime-bin package require libtinfo.so.5 library which can be installed from: https://aur.archlinux.org/packages/ncurses5-compat-libs

gitdev-f11, 2019-02-20:
Thx for the really fast answer !
I've already checked the required packages, and now reinstalled.
ncurses5-compat-libs: Version 6.1-2

But:  

     $ ll /usr/lib/libtinfo*
     lrwxrwxrwx 1 root root 16 11. Jan 09:45 /usr/lib/libtinfo.so.6 -> libncursesw.so.6
     -rw-r--r-- 1 root root 24 11. Jan 09:45 /usr/lib/libtinfo.so

Missinig is `libtinfo.so.5`

gitdev-f11, 2019-02-20:
# in /usr/lib
    $ sudo ln -sf libncursesw.so.6 libtinfo.so.5

solved my problems ;)

closed 2019-02-20 by gitdev-f11