intel/compute-runtime #136: Hard-lock running opencl samples - E3-1245v5

Cp-Graham, 2019-02-27:
**Setup:**
HW
- model : 94
- model name : Intel(R) Xeon(R) CPU E3-1245 v5 @ 3.50GHz
- stepping : 3
- GPU: HD Graphics P530

OS Name Ubuntu 
OS Version 16.04
Other 
- Linux version 4.7.0.intel.r4.0
- gcc version 5.4.0 20160609 
- Ubuntu 5.4.0-6ubuntu1~16.04.4

Microcode (HT disabled)
- [    2.763729] microcode: CPU0 sig=0x506e3, pf=0x2, revision=0xc6
- [    2.763811] microcode: CPU1 sig=0x506e3, pf=0x2, revision=0xc6
- [    2.763892] microcode: CPU2 sig=0x506e3, pf=0x2, revision=0xc6
- [    2.763895] microcode: CPU3 sig=0x506e3, pf=0x2, revision=0xc6
- [    2.764043] microcode: Microcode Update Driver: v2.01 

Driver Version
-   CL_DEVICE_NAME - Intel(R) Gen9 HD Graphics NEO
-   CL_DEVICE_VENDOR - Intel(R) Corporation
-   CL_DRIVER_VERSION - 19.07.12410
-   CL_DEVICE_TYPE - CL_DEVICE_TYPE_GPU

**Issue:**
We are encountering an issue while running NEO drivers on a E3-1245v5 CPU.
Running the attached OpenCL sample (intel_gpu_struct_bug) repeatedly will cause the device to eventually hardlock (seems to be random timing) 
Reboot required to recover and no output obtainable/no dump generated

**Repro steps:**
- untar the attached file
- in bash run "watch -n0.1 ./intel_gpu_struct_bug"
- wait for the CPU to hardlock/crash

**Notes:**
- We disabled hyper-threading to check if it works around the issue. Sadly it doesn't 
- We might be able to provide a server type/setup config if required.
- This issue doesn't seem to happen on non NEO drivers
- Server firmware is fully up to date
- The test case provided is just an easy way to reproduce the issue we are seeing in our normal testing

Thanks
Graham
[intel_gpu_struct_bug.tar.gz](https://github.com/intel/compute-runtime/files/2910599/intel_gpu_struct_bug.tar.gz)

MichalMrozek, 2019-02-28:
Looks like you are running custom Kernel.
* Linux version 4.7.0.intel.r4.0

Neo driver doesn't work well with non plain Open Source Linux Kernels.
Can you try with plain Kernel without any Intel patches?
Can you try with some more recent Kernels?

Cp-Graham, 2019-03-04:
Hi @MichalMrozek,

Apologies for not getting back to you sooner it does indeed look like the issue is resolved in the latest standard kernel on 16.04 and worked out of the box on 18.04!

Thanks again for your help!
Cheers
Graham

closed 2019-03-04 by Cp-Graham