intel/compute-runtime #138: Does the CPU runtime support: model name      : Pentium(R) Dual-Core  CPU E5400 @ 2.70GHz

danielsokolowski, 2019-03-02:
The reason I ask is that the relase notes state that SSE3 is supported, but after installing the packages on Debian (followed Ubuntu install)  `clinfo` does not return any platforms. Checking with `strace` confirms the libraries are being loaded. 

The old 5.0 version (intel-opencl-r5.0 (SRB5.0) Linux driver) of the CPU runtime `clinfo` does report a platform but fails when I test some test programs, after digging it's because SSE4.1 is required (https://software.intel.com/en-us/forums/opencl/topic/506830)

AdamCetnerowski, 2019-03-08:
Neo provides runtime for GPU-only. Other devices, including CPU, are supported in our Developer Zone: https://software.intel.com/en-us/forums/opencl

MichalMrozek, 2019-03-15:
Looks like question is answered, closing.

closed 2019-03-15 by MichalMrozek