intel/compute-runtime #139: Build Issue: Could not load libigdfcl.so

jebos, 2019-03-04:


JacekDanecki, 2019-03-04:
Have you executed
```
sudo ldconfig
```
after installing IGC?
You can find some tips in [this issue](https://github.com/intel/compute-runtime/issues/123)

jebos, 2019-03-04:
Well, this can be closed, Just did a clean new attempt and it worked out this time to compile, not sure what caused the issue on first run.

closed 2019-03-04 by jebos

eero-t, 2020-02-18:
> Well, this can be closed, Just did a clean new attempt and it worked out this time to compile, not sure what caused the issue on first run.

If you did first one build which failed, and then a second one from start, some other component probably did "ldconfig" call, which resulted IGC libraries being in dynamic linker cache.  IGC should do ldconfig on install like other libraries do (preferably so that it's allowed to fail, if install isn't done as root, as then user is supposed to have set LD_LIBRARY_PATH to point to install path instead).

JacekDanecki, 2020-02-18:
ldconfig has been added in IGC [commit](https://github.com/intel/intel-graphics-compiler/commit/fcba69c1f9a6fd1efb7ce8a54cabe2acafe20d6a#diff-eb42e3a6a676df5441ef16072e9fb3a8). If you still observe issue with IGC installation, please fill new bug in [IGC repository](https://github.com/intel/intel-graphics-compiler/issues)