intel/compute-runtime #140: Possible NEO patch bug

FelipeMLopez, 2019-03-05:
Hello!
I've been testing these changes and I think there is a problem with #53 patch.
Like I said in #134, I had problems to achieve fine-grained buffer support.
I managed to install the driver following the @JacekDanecki instructions:
```
$ git fetch origin +pull/53/head:svm
$ git checkout svm
$ git rebase 18.49.11968
```
But I continued without fine-grained buffer support. Then, following the @BartoszDunajski instructions in #134, I was reviewing the sources and I made these changes:

- runtime/os_interface/linux/hw_info_config.cpp
```
    // outHwInfo->capabilityTable.ftrSupportsCoherency = false;
    outHwInfo->capabilityTable.ftrSupportsCoherency = true;
```
And I fixed some test problems with ftrSupportsCoherency:

- unit_tests/os_interface/linux/hw_info_config_linux_tests.cpp
```

// TEST_F(HwInfoConfigTestLinuxDummy, givenEnabledPlatformCoherencyWhenConfiguringHwInfoThenIgnoreAndSetAsDisabled) {
//     drm->StoredDeviceID = 21;
//     int ret = hwConfig.configureHwInfo(pInHwInfo, &outHwInfo, osInterface);
//     EXPECT_EQ(0, ret);
//     EXPECT_FALSE(outHwInfo.capabilityTable.ftrSupportsCoherency);
// }
//
// TEST_F(HwInfoConfigTestLinuxDummy, givenDisabledPlatformCoherencyWhenConfiguringHwInfoThenSetValidCapability) {
//     drm->StoredDeviceID = 20;
//     int ret = hwConfig.configureHwInfo(pInHwInfo, &outHwInfo, osInterface);
//     EXPECT_EQ(0, ret);
//     EXPECT_FALSE(outHwInfo.capabilityTable.ftrSupportsCoherency);
// }

TEST_F(HwInfoConfigTestLinuxDummy, givenEnabledPlatformCoherencyWhenConfiguringHwInfoThenSetValidCapability) {
    drm->StoredDeviceID = 21;
    int ret = hwConfig.configureHwInfo(pInHwInfo, &outHwInfo, osInterface);
    EXPECT_EQ(0, ret);
    EXPECT_TRUE(outHwInfo.capabilityTable.ftrSupportsCoherency);
}

TEST_F(HwInfoConfigTestLinuxDummy, givenDisabledPlatformCoherencyWhenConfiguringHwInfoThenIgnoreAndSetAsEnabled) {
    drm->StoredDeviceID = 20;
    int ret = hwConfig.configureHwInfo(pInHwInfo, &outHwInfo, osInterface);
    EXPECT_EQ(0, ret);
    EXPECT_TRUE(outHwInfo.capabilityTable.ftrSupportsCoherency);
}
```
I have fine-grained buffer support now, and this change is cleanly building and passing ULTs.
Could be this a driver bug?
I need to be sure before starting a pull request .
Thank you!

FelipeMLopez, 2019-03-05:
I also managed to apply and to test the changes in 19.08 version, but it's necessary an additional change in #53:

```
 if (this->drm->peekDataPortCoherencyPatchActive() && requiresCoherency != this->drm->peekIsContextDataPortCoherent()) {
        this->drm->setContextDataPortCoherent(requiresCoherency);
    }
```
Instead of:

```
    if (drm->peekDataPortCoherencyPatchActive() && requiresCoherency != drm->peekIsContextDataPortCoherent()) {
        drm->setContextDataPortCoherent(requiresCoherency);
    }
```
I hope that helps.
Thanks.

BartoszDunajski, 2019-03-08:
@FelipeMLopez, missing ftrSupportsCoherency flag setup is implementation gap. It will be fixed during PR merge - it required rebase anyway. 

Is your local fix enough to test fine grain SVM?

FelipeMLopez, 2019-03-08:
Yes, it is. I can use fine grain SVM without problem.

BartoszDunajski, 2019-03-08:
Great :). Thank you for showing interest. Feel free to contribute.

We are going to rebase current PR to top of the branch. Please consider postponing your fix, as it may cause many merge conflicts after rebase.

BartoszDunajski, 2019-03-08:
Can we close this issue?

FelipeMLopez, 2019-03-08:
Of course.
I close it.
Thank you.

closed 2019-03-08 by FelipeMLopez