intel/compute-runtime #141: parallel make doesn't work

tripzero, 2019-03-08:
Doing a make -j5, I get the following error on clearlinux:

 make -j5
Scanning dependencies of target elflib
Scanning dependencies of target gmock-gtest
Scanning dependencies of target biksim
Scanning dependencies of target igdrcl_lib_release_sharings_enable
Scanning dependencies of target copy_compiler_files
[  0%] Building CXX object elf/CMakeFiles/elflib.dir/reader.cpp.o
[  0%] Built target copy_compiler_files
Scanning dependencies of target builtins_sources
[  0%] Building CXX object third_party/gtest/CMakeFiles/gmock-gtest.dir/gmock-gtest-all.cc.o
[  0%] Building CXX object igdrcl_lib_release/built_ins/registry/CMakeFiles/builtins_sources.dir/register_copy_kernels_source.cpp.o
[  0%] Building CXX object igdrcl_lib_release/sharings/CMakeFiles/igdrcl_lib_release_sharings_enable.dir/d3d/enable_d3d.cpp.o
[  0%] Building CXX object igdrcl_lib_release/builtin_kernels_simulation/CMakeFiles/biksim.dir/opencl_c.cpp.o
[  0%] Building CXX object igdrcl_lib_release/sharings/CMakeFiles/igdrcl_lib_release_sharings_enable.dir/va/enable_va.cpp.o
[  0%] Building CXX object elf/CMakeFiles/elflib.dir/writer.cpp.o
[  0%] Linking CXX static library ../lib/libelflib.a
[  0%] Building CXX object igdrcl_lib_release/builtin_kernels_simulation/CMakeFiles/biksim.dir/scheduler_simulation.cpp.o
[  0%] Built target elflib
Scanning dependencies of target igdrcl_lib_mockable_sharings_enable
[  0%] Building CXX object igdrcl_lib_mockable/sharings/CMakeFiles/igdrcl_lib_mockable_sharings_enable.dir/d3d/enable_d3d.cpp.o
[  0%] Built target igdrcl_lib_release_sharings_enable
Scanning dependencies of target igdrcl_mocks
[  0%] Building CXX object igdrcl_lib_mockable/sharings/CMakeFiles/igdrcl_lib_mockable_sharings_enable.dir/va/enable_va.cpp.o
[  0%] Building CXX object igdrcl_lib_release/built_ins/registry/CMakeFiles/builtins_sources.dir/register_ext_vme_source.cpp.o
[  0%] Building CXX object unit_tests/mocks/CMakeFiles/igdrcl_mocks.dir/mock_async_event_handler.cpp.o
cc1plus: error: too many filenames given.  Type cc1plus --help for usage

alalek, 2019-03-09:
Try `make VERBOSE=1` to get details.

JacekDanecki, 2019-03-11:
@tripzero What Neo commit ID are you trying to build?

JacekDanecki, 2019-03-13:
I've reproduced this issue on clearlinux. I don't see such problem on another Linux distributions we test.

JacekDanecki, 2019-03-13:
As a workaround you can use ninja tool:
```
cmake -G Ninja ../
ninja -j `nproc`
```
but it'll fail because of issue on [igc side](https://github.com/intel/intel-graphics-compiler/issues/75)

JacekDanecki, 2019-03-15:
You can use below patch to enable Neo compilation, using make or ninja
```
diff --git a/unit_tests/CMakeLists.txt b/unit_tests/CMakeLists.txt
index 393267ec..6bcfe3fa 100644
--- a/unit_tests/CMakeLists.txt
+++ b/unit_tests/CMakeLists.txt
@@ -9,15 +9,6 @@ project(igdrcl_tests)
 set(TEST_PROJECTS_FOLDER "opencl runtime")
 set(PLATFORM_SPECIFIC_TARGETS_FOLDER "${TEST_PROJECTS_FOLDER}/test platforms")
 
-# disable optimizations for ults
-if(UNIX)
-  string(REPLACE "-O2" "" CMAKE_CXX_FLAGS ${CMAKE_CXX_FLAGS})
-  #disable fortify source as this requires optimization to be on
-  string(REPLACE "-D_FORTIFY_SOURCE=2" "" CMAKE_CXX_FLAGS ${CMAKE_CXX_FLAGS})
-  set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O0")
-  set(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O0")
-endif()
-
 if(WIN32)
   string(REPLACE "/O2" "" CMAKE_CXX_FLAGS ${CMAKE_CXX_FLAGS})
   string(REPLACE "/O2" "/Od" CMAKE_CXX_FLAGS_RELEASE ${CMAKE_CXX_FLAGS_RELEASE})
```
But I've found another issue during ULT tests
```
cd /root/compute-runtime/unit_tests/test_files && LD_LIBRARY_PATH=/root/compute-runtime/build/bin /root/compute-runtime/build/bin/ocloc -q -file CopyBuffer_simd8.cl -device glk -64 -out_dir /root/compute-runtime/build/bin/Gen9lp/test_files/x64/ -output -cl-kernel-debug-enable -internal_options -cl-kernel-debug-enable -options -g
[0]: /root/compute-runtime/build/bin/ocloc() [0x42400b]
[1]: /usr/lib64/libpthread.so.0(+0x14610) [0x7f92a231d610]
[2]: /usr/lib64/libLLVM-7.so(_ZN4llvm13StringMapImpl15LookupBucketForENS_9StringRefE+0xc0) [0x7f929ec96f50]
[3]: /usr/lib64/libLLVM-7.so(+0xafa889) [0x7f929edd4889]
[4]: /usr/lib64/libLLVM-7.so(_ZN4llvm8MDString3getERNS_11LLVMContextENS_9StringRefE+0x11) [0x7f929edd4851]
[5]: /usr/lib64/libLLVM-7.so(+0xa7c066) [0x7f929ed56066]
[6]: /usr/lib64/libLLVM-7.so(_ZN4llvm9DIBuilder10createFileENS_9StringRefES1_NS_8OptionalINS_6DIFile12ChecksumInfoIS1_EEEENS2_IS1_EE+0x42) [0x7f929ed52302]
[7]: /usr/local/lib/libigc.so(+0x197a32) [0x7f929c899a32]
[8]: /usr/local/lib/libigc.so(+0x1980d5) [0x7f929c89a0d5]
[9]: /usr/local/lib/libigc.so(+0x1baf7f) [0x7f929c8bcf7f]
[10]: /usr/local/lib/libigc.so(+0x1bbca8) [0x7f929c8bdca8]
[11]: /usr/local/lib/libigc.so(+0x11a1c6) [0x7f929c81c1c6]
[12]: /usr/local/lib/libigc.so(+0x11dc96) [0x7f929c81fc96]
[13]: /usr/local/lib/libigc.so(+0x1c683b) [0x7f929c8c883b]
[14]: /usr/local/lib/libigc.so(+0x1c97a7) [0x7f929c8cb7a7]
[15]: /root/compute-runtime/build/bin/ocloc(_ZN5OCLRT15OfflineCompiler15buildSourceCodeEv+0x6b9) [0x41a9f9]
[16]: /root/compute-runtime/build/bin/ocloc(_ZN5OCLRT15OfflineCompiler5buildEv+0xe) [0x41eb6e]
[17]: /root/compute-runtime/build/bin/ocloc() [0x4240b4]
[18]: /root/compute-runtime/build/bin/ocloc(_Z20buildWithSafetyGuardPN5OCLRT15OfflineCompilerE+0x74) [0x423fb4]
[19]: /root/compute-runtime/build/bin/ocloc(main+0xc5) [0x40c555]
[20]: /usr/lib64/haswell/libc.so.6(__libc_start_main+0xf3) [0x7f92a1e3ff13]
[21]: /root/compute-runtime/build/bin/ocloc(_start+0x2e) [0x40c87e]
Aborted (core dumped)
```
To workaround this you can build Neo without ULT
```
cmake ..
make -j `nproc` igdrcl_dll
```
Btw, I was not able to reproduce https://github.com/intel/intel-graphics-compiler/issues/75, when I've recompiled IGC on clearlinux 28280.

tripzero, 2019-03-15:
Disable ULT should probably be added as a cmake option.

tripzero, 2019-03-15:
@JacekDanecki you will need https://github.com/intel/opencl-clang/pull/49 to reproduce https://github.com/intel/intel-graphics-compiler/issues/75 and get past your current crash in ULT.

JacekDanecki, 2019-03-15:
@To disable ULT use this flag:
```
cmake .. -DSKIP_UNIT_TESTS=1
```

commit c373baf0 (2020-01-09): Fix unit test compilation under Clear linux

Fixes #141

Change-Id: I1d52046a8951ae15a39f76e67c7d0d2b537d946b
closed 2020-01-09 by VPG-SWE-Github