intel/compute-runtime #149: Where to download OpenCL™ Runtime 15.1?

danielsokolowski, 2019-03-24:
Continuation to this issue: https://github.com/intel/compute-runtime/issues/138

Where can I get 'OpenCL™ Runtime 15.1' package? The official [releases](https://github.com/intel/compute-runtime/releases) only go down to 18.X, but the [release docs](https://software.intel.com/sites/default/files/managed/c9/24/opencl_runtime-15.1-release_notes.pdf) state I should be able to download for that url. 

[image]

PiotrRozenfeld, 2019-03-24:
See here: https://software.intel.com/en-us/articles/legacy-opencl-drivers#latest_CPU_runtime

PiotrRozenfeld, 2019-03-29:
Response provided - closing. 
If you have further questions for OpenCL CPU, the [forums](https://software.intel.com/en-us/forums/opencl/) might be a better place to get support.

closed 2019-03-29 by MichalMrozek