intel/compute-runtime #155: Unit tests failed while building NEO on CentOS 7

akharche, 2019-04-05:
We are building NEO for MediaSDK release 19.1. Build failed due to issue in the Unit test, see the log below. We found similar [issue](https://github.com/intel/compute-runtime/issues/137). The [PR](https://github.com/intel/compute-runtime/pull/53) with fix was not merged. This issue is showstopper problem for the release, it should be fixed ASAP!

Versions:
NEO - 19.12.12612
gmmlib - 18.4.1
IGC - igc_release_2019-03-12

Build commands:
```sh
export PKG_CONFIG_PATH=${PKG_CONFIG_PATH}:/localdisk/gmmlib/usr/local/lib64/pkgconfig/  
source /opt/rh/devtoolset-6/enable && cmake3 -DCMAKE_INSTALL_PREFIX=/usr/local -DIGC_DIR=/localdisk/igc/usr/local/ /build/build_dir/repos/opencl_runtime  
source /opt/rh/devtoolset-6/enable && make -j`nproc`  
```

```sh
*** Error in `/build/build_dir/build/bin/igdrcl_aub_tests': double free or corruption (!prev): 0x0000000018d836a0 ***
======= Backtrace: =========
/usr/lib64/libc.so.6(+0x81489)[0x7effd2e90489]
/build/build_dir/build/bin/igdrcl_aub_tests[0x75fdee]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZdlPv+0x18)[0x75f44c]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZN9__gnu_cxx13new_allocatorISt13_Rb_tree_nodeISt4pairIKmPFPN3CIF4ICIFEmmEEEE10deallocateEPSA_m+0x20)[0x7a0482]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZNSt16allocator_traitsISaISt13_Rb_tree_nodeISt4pairIKmPFPN3CIF4ICIFEmmEEEEE10deallocateERSA_PS9_m+0x2b)[0x79ffdc]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZNSt8_Rb_treeImSt4pairIKmPFPN3CIF4ICIFEmmEESt10_Select1stIS7_ESt4lessImESaIS7_EE11_M_put_nodeEPSt13_Rb_tree_nodeIS7_E+0x33)[0x79f709]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZNSt8_Rb_treeImSt4pairIKmPFPN3CIF4ICIFEmmEESt10_Select1stIS7_ESt4lessImESaIS7_EE12_M_drop_nodeEPSt13_Rb_tree_nodeIS7_E+0x36)[0x79eff8]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZNSt8_Rb_treeImSt4pairIKmPFPN3CIF4ICIFEmmEESt10_Select1stIS7_ESt4lessImESaIS7_EE12_M_erase_auxESt23_Rb_tree_const_iteratorIS7_E+0x3e)[0x79fd64]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZNSt8_Rb_treeImSt4pairIKmPFPN3CIF4ICIFEmmEESt10_Select1stIS7_ESt4lessImESaIS7_EE5eraseB5cxx11ESt17_Rb_tree_iteratorIS7_E+0x59)[0x79f37f]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZNSt3mapImPFPN3CIF4ICIFEmmESt4lessImESaISt4pairIKmS4_EEE5eraseB5cxx11ESt17_Rb_tree_iteratorIS9_E+0x23)[0x79ecf1]
/build_dir/build/bin/igdrcl_aub_tests(_ZN5OCLRT11MockCIFMain23removeGlobalCreatorFuncINS_19MockFclOclDeviceCtxEEEvv+0x7a)[0x79dcc2]
/build_dir/build/bin/igdrcl_aub_tests(_ZN5OCLRT23MockCompilerEnableGuard7DisableEv+0x45)[0x79a9a3]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZN5OCLRT23MockCompilerEnableGuardD1Ev+0x18)[0x79a80c]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZN15TestEnvironmentD1Ev+0x30)[0x77fcc0]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZN15TestEnvironmentD0Ev+0x18)[0x77fd4e]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZN7testing8internal12UnitTestImplD1Ev+0x88)[0x951ed8]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZN7testing8internal12UnitTestImplD0Ev+0x9)[0x9521c9]
/build/build_dir/build/bin/igdrcl_aub_tests(_ZN7testing8UnitTestD1Ev+0x1e)[0x9521fe]
/usr/lib64/libc.so.6(+0x39b69)[0x7effd2e48b69]
/usr/lib64/libc.so.6(+0x39bb7)[0x7effd2e48bb7]
/usr/lib64/libc.so.6(__libc_start_main+0xfc)[0x7effd2e313dc]
/build/build_dir/build/bin/igdrcl_aub_tests[0x67bb35]
======= Memory map: ========
00400000-00a95000 r-xp 00000000 08:01 4066275                            /localdisk/tmp/akharche/build/build_dir/build/bin/igdrcl_aub_tests
00c95000-00ca1000 r--p 00695000 08:01 4066275                            /localdisk/tmp/akharche/build/build_dir/build/bin/igdrcl_aub_tests
00ca1000-00ce7000 rw-p 006a1000 08:01 4066275                            /localdisk/tmp/akharche/build/build_dir/build/bin/igdrcl_aub_tests
00ce7000-16cf0000 rw-p 00000000 00:00 0
18ca6000-18da4000 rw-p 00000000 00:00 0                                  [heap]
7effcc000000-7effcc021000 rw-p 00000000 00:00 0
7effcc021000-7effd0000000 ---p 00000000 00:00 0
7effd260e000-7effd260f000 ---p 00000000 00:00 0
7effd260f000-7effd2e0f000 rw-p 00000000 00:00 0
7effd2e0f000-7effd2fd1000 r-xp 00000000 08:01 2359898                    /usr/lib64/libc-2.17.so
7effd2fd1000-7effd31d1000 ---p 001c2000 08:01 2359898                    /usr/lib64/libc-2.17.so
7effd31d1000-7effd31d5000 r--p 001c2000 08:01 2359898                    /usr/lib64/libc-2.17.so
7effd31d5000-7effd31d7000 rw-p 001c6000 08:01 2359898                    /usr/lib64/libc-2.17.so
7effd31d7000-7effd31dc000 rw-p 00000000 00:00 0
7effd31dc000-7effd31f1000 r-xp 00000000 08:01 2377364                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7effd31f1000-7effd33f0000 ---p 00015000 08:01 2377364                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7effd33f0000-7effd33f1000 r--p 00014000 08:01 2377364                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7effd33f1000-7effd33f2000 rw-p 00015000 08:01 2377364                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7effd33f2000-7effd34f3000 r-xp 00000000 08:01 2365222                    /usr/lib64/libm-2.17.so
7effd34f3000-7effd36f2000 ---p 00101000 08:01 2365222                    /usr/lib64/libm-2.17.so
7effd36f2000-7effd36f3000 r--p 00100000 08:01 2365222                    /usr/lib64/libm-2.17.so
7effd36f3000-7effd36f4000 rw-p 00101000 08:01 2365222                    /usr/lib64/libm-2.17.so
7effd36f4000-7effd37dd000 r-xp 00000000 08:01 2360207                    /usr/lib64/libstdc++.so.6.0.19
7effd37dd000-7effd39dc000 ---p 000e9000 08:01 2360207                    /usr/lib64/libstdc++.so.6.0.19
7effd39dc000-7effd39e4000 r--p 000e8000 08:01 2360207                    /usr/lib64/libstdc++.so.6.0.19
7effd39e4000-7effd39e6000 rw-p 000f0000 08:01 2360207                    /usr/lib64/libstdc++.so.6.0.19
7effd39e6000-7effd39fb000 rw-p 00000000 00:00 0
7effd39fb000-7effd3a12000 r-xp 00000000 08:01 2359924                    /usr/lib64/libpthread-2.17.so
7effd3a12000-7effd3c11000 ---p 00017000 08:01 2359924                    /usr/lib64/libpthread-2.17.so
7effd3c11000-7effd3c12000 r--p 00016000 08:01 2359924                    /usr/lib64/libpthread-2.17.so
7effd3c12000-7effd3c13000 rw-p 00017000 08:01 2359924                    /usr/lib64/libpthread-2.17.so
7effd3c13000-7effd3c17000 rw-p 00000000 00:00 0
7effd3c17000-7effd3c19000 r-xp 00000000 08:01 2365221                    /usr/lib64/libdl-2.17.so
7effd3c19000-7effd3e19000 ---p 00002000 08:01 2365221                    /usr/lib64/libdl-2.17.so
7effd3e19000-7effd3e1a000 r--p 00002000 08:01 2365221                    /usr/lib64/libdl-2.17.so
7effd3e1a000-7effd3e1b000 rw-p 00003000 08:01 2365221                    /usr/lib64/libdl-2.17.so
7effd3e1b000-7effd3e3d000 r-xp 00000000 08:01 2359887                    /usr/lib64/ld-2.17.so[2019-04-05 10:28:33,744] OpenCL: build ERROR: Scanning dependencies of target copy_compiler_files
Scanning dependencies of target test_dynamic_lib
Scanning dependencies of target elflib
Scanning dependencies of target gmock-gtest

Scanning dependencies of target run_ocloc_tests
deleting offline compiler files and directories...

Error! Couldn't copy kernel_KernelHeap.bin
Error while processing kernel!

[ 70%] Built target run_kbl_aub_tests
Scanning dependencies of target run_skl_aub_tests

unknown file: Failure
C++ exception with description "std::exception" thrown in auxiliary test code (environments or event listeners).
SIGABRT on: AUBFillBuffer.givenFillBufferWhenSeveralSubmissionsWithoutPollForCompletionBetweenThenTestConcurrentCS
[ 70%] Building CXX object unit_tests/CMakeFiles/igdrcl_tests.dir/aub_mem_dump/lrca_helper_tests.cpp.o
/bin/sh: line 1: 22801 Aborted                 /build/build_dir/build/bin/igdrcl_aub_tests --product skl --slices 1 --subslices 3 --eu_per_ss 8 --gtest_repeat=1 --disable_default_listener
unit_tests/aub_tests/CMakeFiles/run_skl_aub_tests.dir/build.make:61: recipe for target 'run_skl_aub_tests' failed
make[2]: *** [run_skl_aub_tests] Error 134
CMakeFiles/Makefile2:4564: recipe for target 'unit_tests/aub_tests/CMakeFiles/run_skl_aub_tests.dir/all' failed
make[1]: *** [unit_tests/aub_tests/CMakeFiles/run_skl_aub_tests.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

```

AdamCetnerowski, 2019-04-05:
Please provide full build log

akharche, 2019-04-05:
Cmake and make logs:
[OpenCL_cmake.log](https://github.com/intel/compute-runtime/files/3047794/OpenCL_cmake.log)
[OpenCL_build.log](https://github.com/intel/compute-runtime/files/3047796/OpenCL_build.log)

ArturHarasimiuk, 2019-04-05:
This version of IGC was not tested in connection with Neo. Please use revision from our [release](https://github.com/intel/compute-runtime/releases/tag/19.12.12612): intel/intel-graphics-compiler@429b1d5

risingape, 2019-05-17:
Same unit test fails on Opensuse Tumbleweed using IGC @664021b and GmmLib @8fe838.

Please let me know if any further details are needed.

Cheers,

Guido

ArturHarasimiuk, 2019-05-17:
the best way to build on Centos is:
* build intel/gmmlib and intel/intel-graphics-compiler projects with cmake parameter `-DCMAKE_INSTALL_PREFIX=/usr`
* install intel-gmmlib-devel and intel-igc-opencl-devel packages and theirs dependencies
* run ldconfig
* build intel/compute-runtime

MichalMrozek, 2019-07-26:
I assume problem is resolved as there is no further activity here.
In case problem still reproduces please reopen.

closed 2019-07-26 by MichalMrozek