intel/compute-runtime #156: "LLVM ERROR: unknown argument number for an object" when building 19.13.12717

dev-zero, 2019-04-07:
When trying to build 19.13.12717, compilation fails here:

```
cmake -C /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Gentoo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/gentoo_toolchain.cmake  /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717
loading initial cache file /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/gentoo_common_config.cmake
CMake Warning (dev) at gentoo_common_config.cmake:8 (SET):
  implicitly converting 'BOOLEAN' to 'STRING' type.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/lib64/ccache/bin/x86_64-pc-linux-gnu-gcc
-- Check for working C compiler: /usr/lib64/ccache/bin/x86_64-pc-linux-gnu-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/lib64/ccache/bin/x86_64-pc-linux-gnu-g++
-- Check for working CXX compiler: /usr/lib64/ccache/bin/x86_64-pc-linux-gnu-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Gentoo build configuration
-- Source Level Debugger headers dir: /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/third_party/source_level_debugger
-- Aub Stream Headers dir: /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/third_party/aub_stream/headers
-- i915 includes dir: /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/third_party/uapi
-- Khronos OpenCL headers dir: /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/third_party/opencl_headers
-- Khronos OpenGL headers dir: /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/third_party/opengl_headers
-- Third party dir: /var/tmp/portage/dev-libs/neo-19.13.12717/work/third_party
-- Found PkgConfig: x86_64-pc-linux-gnu-pkg-config (found version "0.29.2") 
-- Checking for module 'igc-opencl'
--   Found igc-opencl, version 1.0.0
-- IGC include dirs: /usr/include/igc;/usr/include/igc/cif;/usr/include/igc/ocl_igc_shared/executable_format;/usr/include/igc/ocl_igc_shared/device_enqueue
-- Checking for module 'igdgmm'
--   Found igdgmm, version 9.0.0
-- GmmLib include dirs: /usr/include/igdgmm;/usr/include/igdgmm/GmmLib;/usr/include/igdgmm/GmmLib/inc;/usr/include/igdgmm/inc;/usr/include/igdgmm/inc/common;/usr/include/igdgmm/util
-- Checking for module 'libva>=1.0.0'
--   Found libva, version 1.4.0
-- Looking for vaGetLibFunc in va
-- Looking for vaGetLibFunc in va - found
-- Using libva 
-- AUB_STREAM_DIR = 
-- Engine node dir: /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/third_party/aub_stream/headers
-- All supported platforms:  CNL SKL KBL GLK CFL BXT BDW
-- All tested platforms:  CNL SKL KBL GLK CFL BXT BDW
-- Default supported platform: SKL
-- Default tested platform: SKL
-- Default tested family name: SKLFamily
-- Found ccache: /usr/bin/ccache
CMake Warning (dev) at runtime/dll/CMakeLists.txt:57 (target_sources):
  Policy CMP0076 is not set: target_sources() command converts relative paths
  to absolute.  Run "cmake --help-policy CMP0076" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

  A private source from a directory other than that of target "igdrcl_dll"
  has a relative path.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- GTest repeat count set to 1
-- GTest shuffle set to --gtest_shuffle;--gtest_random_seed=0
-- GTest exception options set to --gtest_catch_exceptions=1
-- GTest filter for mt tests:-AsyncGPUoperations.MapBufferAfterWriteBuffer
-- <<< Gentoo configuration >>>
Build type      Gentoo
Install path    /usr
Compiler flags:
C               -march=broadwell -mrtm -mabm -O2 -ggdb -pipe
C++             -march=broadwell -mrtm -mabm -O2 -ggdb -pipe -ftemplate-depth=1024 -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed
Module          -Wl,-O1 -Wl,--as-needed
Shared          -Wl,-O1 -Wl,--as-needed

-- Configuring done
-- Generating done
-- Build files have been written to: /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build
[...]
[1133/1405] cd /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/unit_tests/test_files && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/ocloc -q -file CopyBuffer_simd8.cl -device glk -64 -out_dir /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/Gen9lp/test_files/x64/ -output -cl-kernel-debug-enable -internal_options -cl-kernel-debug-enable -options -g
[1134/1405] cd /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/unit_tests/test_files && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/ocloc -q -file simple_block_kernel.cl -device bdw -64 -out_dir /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/Gen8core/test_files/x64/ -options -cl-std=CL2.0 -options_name
FAILED: bin/Gen8core/test_files/x64/simple_block_kernel_Gen8core.spv-cl-std=CL2.0 bin/Gen8core/test_files/x64/simple_block_kernel_Gen8core.bin-cl-std=CL2.0 bin/Gen8core/test_files/x64/simple_block_kernel_Gen8core.gen-cl-std=CL2.0 
cd /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/unit_tests/test_files && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/ocloc -q -file simple_block_kernel.cl -device bdw -64 -out_dir /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/Gen8core/test_files/x64/ -options -cl-std=CL2.0 -options_name
LLVM ERROR: unknown argument number for an object
[1135/1405] cd /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/unit_tests/test_files && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/ocloc -q -file simple_nonuniform.cl -device bdw -64 -out_dir /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/Gen8core/test_files/x64/ -options -cl-std=CL2.0 -options_name
[1136/1405] cd /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/unit_tests/test_files && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/ocloc -q -file 7030307152995455603.cl -device bdw -64 -out_dir /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/Gen8core/test_files/x64/
[1137/1405] cd /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/unit_tests/test_files && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/ocloc -q -file media_kernels_backend.cl -device bdw -64 -out_dir /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/Gen8core/test_files/x64/
ninja: build stopped: subcommand failed.
[...]
```

with:

* llvm/clang 8.0.0 (unpatched/original)
* SPIRV-LLVM-Translator 8.0.0-1
* opencl-clang 8.0.0-1
* IGC 1.0.0 (with a [small patch](https://github.com/intel/intel-graphics-compiler/issues/83#issuecomment-480562279) to make it compile)
* gmmlib-19.1.1
* libva-2.4.0
* libdrm-2.4.97

JacekDanecki, 2019-04-08:
@dev-zero Can you confirm that patches from [here](https://github.com/intel/opencl-clang/tree/ocl-open-80/patches/clang) were applied to clang 8.0.0, you are using, and also patch from [here](https://github.com/intel/opencl-clang/tree/ocl-open-80/patches/spirv) was applied on [SPIRV-LLVM-Translator](https://github.com/KhronosGroup/SPIRV-LLVM-Translator.git)?
I've observed such issue when trying to build Neo using unpatched clang 8.

dev-zero, 2019-04-09:
@JacekDanecki I didn't when I first opened the issue, but now I have:

* clang-8.0.0 with the mentioned patches
* SPIRV-LLVM-Translator 8.0.0-1 without the patch explicitly applied, but when I try to apply it, `patch`tells me it discovered a reversed patch

After that I rebuilt opencl-clang and IGC.

JacekDanecki, 2019-04-09:
@dev-zero What about Neo? Do you still observe issue with ocloc, during Neo build, after clang was patched?

dev-zero, 2019-04-09:
Yes, I still see the same error after patching clang:

```
[...]
FAILED: bin/Gen8core/test_files/x64/simple_block_kernel_Gen8core.spv-cl-std=CL2.0 bin/Gen8core/test_files/x64/simple_block_kernel_Gen8core.bin-cl-std=CL2.0 bin/Gen8core/test_files/x64/simple_block_kernel_Gen8core.gen-cl-std=CL2.0 
cd /var/tmp/portage/dev-libs/neo-19.13.12717/work/compute-runtime-19.13.12717/unit_tests/test_files && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/ocloc -q -file simple_block_kernel.cl -device bdw -64 -out_dir /var/tmp/portage/dev-libs/neo-19.13.12717/work/neo-19.13.12717_build/bin/Gen8core/test_files/x64/ -options -cl-std=CL2.0 -options_name
LLVM ERROR: unknown argument number for an object
[...]
```

I also cleared the `ccache` cache for neo before rebuilding it.
Is there a way I can increase the verbosity to figure out which object was invalid to possibly trace it back to one of the components?

dev-zero, 2019-04-09:
Ok, thanks. So, you were able to reproduce it?

JacekDanecki, 2019-04-10:
I've reproduced this issue on Ubuntu 19.04 with system clang:
```
$ clang++-8 --version
clang version 8.0.0-3 (tags/RELEASE_800/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
$ apt-cache policy clang-8
clang-8:
  Installed: 1:8-3
  Candidate: 1:8-3
  Version table:
 *** 1:8-3 500
        500 http://archive.ubuntu.com/ubuntu disco/universe amd64 Packages
        100 /var/lib/dpkg/status
$ apt-cache show clang-8
Package: clang-8
Architecture: amd64
Version: 1:8-3
Priority: optional
Section: universe/devel
Source: llvm-toolchain-8
Origin: Ubuntu
```
I've not verified whether this clang contains patches from opencl-clang. Will check this later.
I've not tried to build it on Gentoo yet.

JacekDanecki, 2019-04-10:
clang 8 from Ubuntu 19.04 contains 3 patches from opencl-clang repository:
```
OpenCL-Change-type-of-block-pointer-for-OpenCL.patch
OpenCL-Simplify-LLVM-IR-generated-for-OpenCL-blocks.patch
OpenCL-Fix-assertion-due-to-blocks.patch
```
In my first test I've compiled SPIRV-LLVM-Translator myself without patch:
```
Update-LowerOpenCL-pass-to-handle-new-blocks-represn.patch
```
and then problem reproduced:
```
LLVM ERROR: unknown argument number for an object
```
during Neo compilation.
When I installed libllvmspirvlib8 and libllvmspirvlib-dev which contain above patch, Neo compiled without issue. More details about these packages in Ubuntu you can find [here](https://packages.ubuntu.com/source/disco/spirv-llvm-translator)

dev-zero, 2019-04-10:
I'm using SPIRV-LLVM-Translator from here: https://github.com/KhronosGroup/SPIRV-LLVM-Translator/releases/tag/v8.0.0-1

Trying to apply the mentioned patch results in:

```
localhost /var/tmp/portage/dev-libs/llvm-spirv-8.0.0_p1/work/SPIRV-LLVM-Translator-8.0.0-1 # patch -p1 --dry-run < /var/tmp/portage/dev-libs/llvm-spirv-8.0.0_p1/files/0001-Update-LowerOpenCL-pass-to-handle-new-blocks-represn.patch
checking file lib/SPIRV/SPIRVLowerOCLBlocks.cpp
Hunk #1 FAILED at 1 (different line endings).
1 out of 1 hunk FAILED
checking file test/global_block.ll
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
2 out of 2 hunks ignored
checking file test/literal-struct.ll
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
4 out of 4 hunks ignored
checking file test/transcoding/block_w_struct_return.ll
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
6 out of 6 hunks ignored
checking file test/transcoding/enqueue_kernel.ll
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
9 out of 9 hunks ignored
```

dev-zero, 2019-04-10:
ah, wait, just saw that the first part of that patch would actually apply if the line endings inside would be correct :-/

JacekDanecki, 2019-04-10:
Patch provided in opencl-clang repository contains CRLF line terminators so can't be applied on release 8.0.0-1 or code from git, you need to change line endings.

Original patch:
```
$ file 0001-Update-LowerOpenCL-pass-to-handle-new-blocks-represn.patch 
0001-Update-LowerOpenCL-pass-to-handle-new-blocks-represn.patch: unified diff output, ASCII text, with very long lines, with CRLF, LF line terminators
$ patch -p1 < 0001-Update-LowerOpenCL-pass-to-handle-new-blocks-represn.patch 
patching file lib/SPIRV/SPIRVLowerOCLBlocks.cpp
Hunk #1 FAILED at 1 (different line endings).
1 out of 1 hunk FAILED -- saving rejects to file lib/SPIRV/SPIRVLowerOCLBlocks.cpp.rej
patching file test/global_block.ll
patching file test/literal-struct.ll
patching file test/transcoding/block_w_struct_return.ll
patching file test/transcoding/enqueue_kernel.ll
```

Cleaned up patch:
```
$ file 0001-Update-LowerOpenCL-pass-to-handle-new-blocks-represn.patch
0001-Update-LowerOpenCL-pass-to-handle-new-blocks-represn.patch: unified diff output, ASCII text, with very long lines
$ patch -p1 < 0001-Update-LowerOpenCL-pass-to-handle-new-blocks-represn.patch 
patching file lib/SPIRV/SPIRVLowerOCLBlocks.cpp
patching file test/global_block.ll
patching file test/literal-struct.ll
patching file test/transcoding/block_w_struct_return.ll
patching file test/transcoding/enqueue_kernel.ll

```

dev-zero, 2019-04-10:
Yes, sorry, didn't see that the first part of the patch would apply after fixing it with `dos2unix`/`fromdos`. 

For the rest (patches for the `.ll` files) I still get an error that they are already applied and I would have to provide `-R` to reverse them when trying to apply them to v8.0.0-1

dev-zero, 2019-04-10:
ok, now I got it: it was only the first part of the patch which was botched. So the second part was already applied and when I tried to reapply it after fixing it it complained that some part was already applied.

dev-zero, 2019-04-10:
Sorry for the confusion. After applying the fixed patch I was able to build NEO and its testsuite seems to work as well. Thanks a lot!

JacekDanecki, 2019-04-10:
Ok, great, closing issue.

closed 2019-04-10 by JacekDanecki