intel/compute-runtime #157: source missing: unit_tests/test_files/copybuffer.elf

tjaalton, 2019-04-10:
Hi, this file is missing source, any chance it could be built along with the rest?

ArturHarasimiuk, 2019-04-10:
Hi,
file is here: https://github.com/intel/compute-runtime/blob/master/unit_tests/test_files/copybuffer.elf

Could you share more details on error you observe? cmake output, build log, etc...

alalek, 2019-04-10:
Perhaps he requesting "source code" for this binary blob. I believe, source of mentioned file is `copybuffer.cl` from the same directory.

BTW, It is just a "test data" (probably some kind of reference data, if its contents are used).

tjaalton, 2019-04-11:
Right, I'm packaging compute-runtime for Debian/Ubuntu, and I'm not sure if having this as-is is a blocker for entry or not. I know it's used only for the tests, so the worst case scenario is to not run those tests if this file has to be removed from the packaging.

ArturHarasimiuk, 2019-04-11:
@tjaalton current content of ~canonical-x/+archive/ubuntu/x-staging looks good, intel-opencl .deb package contains all required files. You can safely skip everything what is under the unit_tests because this content is used only during the driver build.

ArturHarasimiuk, 2019-04-23:
File removed by https://github.com/intel/compute-runtime/commit/dbe4aec7f565280f98a43370672e3a97e70dd5cb that's why I will close this issue. Feel free to open new issue if you find something new.

closed 2019-04-23 by ArturHarasimiuk