intel/compute-runtime #160: Suitable hardware (CPU, Mainboard)

justxi, 2019-04-13:
Hello,

I own a notebook with Intel Core i7 6700HQ (six generation) with "Intel HD-530" graphics. But this does not seem to be supported by Intel Compute-Runtime.

I would like to test and use Intel Compute-Runtime, and buy some hardware. To be sure that it will work, my question is: Is a "Intel Core i5 8400 6x 2.80GHz So.1151" with a "Asus Prime B360-Plus Intel B360 So.1151 Dual Channel DDR4 ATX" suitable hardware to test Intels Compute-Runtime?

PiotrRozenfeld, 2019-04-13:
The hardware you have (i7-6700HQ/HD Graphics 530) is on the list of supported devices. 
Could you provide more details on why Compute Runtime does not work for you? What OS and driver version do you currently use?

justxi, 2019-04-13:
Sorry, this was a little bit too short and misunderstandable. I have a notebook with Intel Core i7 6700HW and nVidia Geforce 1060.  I thought that all Core i processors have integrated graphics (https://ark.intel.com/content/www/de/de/ark/products/88967/intel-core-i7-6700hq-processor-6m-cache-up-to-3-50-ghz.html). I use Linux so I tested the hardware with "lshw" and "inxi", but no Intel graphics device is listed, only the "nVidia GTX 1060". The question is, is it correct that I do not find an Intel graphics device on my notebook?

alalek, 2019-04-13:
@justxi Probably you need to play with BIOS settings about "Integrated graphics". Linux's `lspci` should show both "VGA" devices if all is OK.

justxi, 2019-04-13:
@alalek I didn´t find a Bios option. It is an Acer Predator G5-793-7108. According to a post in the Acer community it is not possible to switch to the iGPU. I found, that the iGPU is not used on another page. Interesstingly, sites that sold this notebook give the information that there is an integrated iGPU. But GPU-Z and GPU-Z list also only the nVidia GPU.  But this is not a problem, I would like to spend some money to buy a simple system, if I can be sure that it will work.

alalek, 2019-04-13:
@justxi Like [this](https://community.acer.com/en/discussion/547277/no-option-to-run-both-integrated-graphics-and-discrete-graphics-in-bios) or [this](https://superuser.com/questions/93194/how-do-i-enable-the-on-board-graphics-card)?

If option is disabled in BIOS, then you can't see iGPU in Operating System (Linux's `lspci` or Windows "Devices").
No needs to switch display "output", we need to turn it ON.

justxi, 2019-04-13:
@alalek I mean this post https://community.acer.com/en/discussion/474950/predator-g5-793-gtx-1060-enable-internal-gpu. It seems that it is possible to unlock extended option with special keyboard shortcuts (which not work for all notebook) or flash a new BIOS. But I will not do it, I will not brick my device ;). Many thanks for your effort.

phush0, 2019-04-18:
You can not enable iGPU on this model, because GTX 1060 card is directly linked to LCD, but you can use Nvidia card for OpenCL too, also you can install CUDA runtime on it. Just search hot to enable OpenCL on Nvidia card, not Intel one.

justxi, 2019-04-18:
Yes thats clear. I wanted to know if an core i5 is a suitable processor. But in the meantime I found the compatibility list and I think it should work.

PiotrRozenfeld, 2019-05-17:
Are there any additional questions? Otherwise please close this issue.

justxi, 2019-05-18:
No more questions, thank you.

closed 2019-05-18 by justxi