intel/compute-runtime #161: Abort was called at 99 line in file: ../neo/runtime/built_ins/built_ins.cpp Aborted (core dumped)

KhazAkar, 2019-04-23:
Can't run even clinfo after updating compute-runtime to the newest version, after trying it to run shows this:
`Abort was called at 99 line in file:
../neo/runtime/built_ins/built_ins.cpp
Aborted (core dumped)`
I'm using OpenSUSE Tumbleweed with Skylake CPU.

JacekDanecki, 2019-04-23:
Can you provide output from clinfo after setting below environment variable:
```
export LD_DEBUG=libs
```

KhazAkar, 2019-04-24:
`LD_DEBUG=libs clinfo >> clinfo.txt
     26157:	find library=libOpenCL.so.1 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/usr/lib64/libOpenCL.so.1
     26157:	
     26157:	find library=libdl.so.2 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/lib64/libdl.so.2
     26157:	
     26157:	find library=libc.so.6 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/lib64/libc.so.6
     26157:	
     26157:	
     26157:	calling init: /lib64/libc.so.6
     26157:	
     26157:	
     26157:	calling init: /lib64/libdl.so.2
     26157:	
     26157:	
     26157:	calling init: /usr/lib64/libOpenCL.so.1
     26157:	
     26157:	
     26157:	initialize program: clinfo
     26157:	
     26157:	
     26157:	transferring control: clinfo
     26157:	
     26157:	find library=libnvidia-opencl.so.1 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/usr/lib64/libnvidia-opencl.so.1
     26157:	
     26157:	find library=libm.so.6 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/lib64/libm.so.6
     26157:	
     26157:	find library=libpthread.so.0 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/lib64/libpthread.so.0
     26157:	
     26157:	find library=librt.so.1 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/lib64/librt.so.1
     26157:	
     26157:	find library=libnvidia-fatbinaryloader.so.418.56 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/usr/lib64/libnvidia-fatbinaryloader.so.418.56
     26157:	
     26157:	
     26157:	calling init: /lib64/libpthread.so.0
     26157:	
     26157:	
     26157:	calling init: /usr/lib64/libnvidia-fatbinaryloader.so.418.56
     26157:	
     26157:	
     26157:	calling init: /lib64/librt.so.1
     26157:	
     26157:	
     26157:	calling init: /lib64/libm.so.6
     26157:	
     26157:	
     26157:	calling init: /usr/lib64/libnvidia-opencl.so.1
     26157:	
     26157:	find library=libstdc++.so.6 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/usr/lib64/libstdc++.so.6
     26157:	
     26157:	find library=libgcc_s.so.1 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/lib64/libgcc_s.so.1
     26157:	
     26157:	
     26157:	calling init: /lib64/libgcc_s.so.1
     26157:	
     26157:	
     26157:	calling init: /usr/lib64/libstdc++.so.6
     26157:	
     26157:	
     26157:	calling init: /usr/local/lib/intel-opencl/libigdrcl.so
     26157:	
     26157:	/usr/lib64/libnvidia-opencl.so.1: error: symbol lookup error: undefined symbol: clIcdGetPlatformIDsKHR (fatal)
     26157:	clinfo: error: symbol lookup error: undefined symbol: pthread_setaffinity_np (fatal)
     26157:	clinfo: error: symbol lookup error: undefined symbol: pthread_getaffinity_np (fatal)
     26157:	find library=libigfxdbgxchg64.so [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	 search path=/lib64/tls/haswell/x86_64:/lib64/tls/haswell:/lib64/tls/x86_64:/lib64/tls:/lib64/haswell/x86_64:/lib64/haswell:/lib64/x86_64:/lib64:/usr/lib64/tls/haswell/x86_64:/usr/lib64/tls/haswell:/usr/lib64/tls/x86_64:/usr/lib64/tls:/usr/lib64/haswell/x86_64:/usr/lib64/haswell:/usr/lib64/x86_64:/usr/lib64		(system search path)
     26157:	  trying file=/lib64/tls/haswell/x86_64/libigfxdbgxchg64.so
     26157:	  trying file=/lib64/tls/haswell/libigfxdbgxchg64.so
     26157:	  trying file=/lib64/tls/x86_64/libigfxdbgxchg64.so
     26157:	  trying file=/lib64/tls/libigfxdbgxchg64.so
     26157:	  trying file=/lib64/haswell/x86_64/libigfxdbgxchg64.so
     26157:	  trying file=/lib64/haswell/libigfxdbgxchg64.so
     26157:	  trying file=/lib64/x86_64/libigfxdbgxchg64.so
     26157:	  trying file=/lib64/libigfxdbgxchg64.so
     26157:	  trying file=/usr/lib64/tls/haswell/x86_64/libigfxdbgxchg64.so
     26157:	  trying file=/usr/lib64/tls/haswell/libigfxdbgxchg64.so
     26157:	  trying file=/usr/lib64/tls/x86_64/libigfxdbgxchg64.so
     26157:	  trying file=/usr/lib64/tls/libigfxdbgxchg64.so
     26157:	  trying file=/usr/lib64/haswell/x86_64/libigfxdbgxchg64.so
     26157:	  trying file=/usr/lib64/haswell/libigfxdbgxchg64.so
     26157:	  trying file=/usr/lib64/x86_64/libigfxdbgxchg64.so
     26157:	  trying file=/usr/lib64/libigfxdbgxchg64.so
     26157:	
     26157:	find library=libigdgmm.so.9 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/usr/local/lib/libigdgmm.so.9
     26157:	
     26157:	
     26157:	calling init: /usr/local/lib/libigdgmm.so.9
     26157:	
     26157:	find library=libva.so.2 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	  trying file=/usr/lib64/libva.so.2
     26157:	
     26157:	
     26157:	calling init: /usr/lib64/libva.so.2
     26157:	
     26157:	
     26157:	calling fini: /usr/lib64/libva.so.2 [0]
     26157:	
     26157:	find library=libigdfcl.so.1 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	 search path=/lib64:/usr/lib64		(system search path)
     26157:	  trying file=/lib64/libigdfcl.so.1
     26157:	  trying file=/usr/lib64/libigdfcl.so.1
     26157:	
     26157:	find library=libigc.so.1 [0]; searching
     26157:	 search cache=/etc/ld.so.cache
     26157:	 search path=/lib64:/usr/lib64		(system search path)
     26157:	  trying file=/lib64/libigc.so.1
     26157:	  trying file=/usr/lib64/libigc.so.1
     26157:	
Aborted (core dumped)
`
Output from LD_DEBUG=libs clinfo command.

JacekDanecki, 2019-04-24:
Where have you installed igc libraries? They were not found in locations:
```
/lib64/libigdfcl.so.1 
/usr/lib64/libigdfcl.so.1 
/lib64/libigc.so.1 
/usr/lib64/libigc.so.1 
```
Can you add directory with IGC libraries to linker cache using file in /etc/ld.so.conf.d directory, and running
```
sudo ldconfig
```
You can also set environment variable to directory containing IGC libraries, for example
```
export LD_LIBRARY_PATH=/usr/local/lib:/usr/local/lib
```

KhazAkar, 2019-04-24:
They are in /usr/local/lib directory, adding LD_LIBRARY_PATH helped to run it. It won't affect running other apps?

JacekDanecki, 2019-04-24:
It depends on where you've added LD_LIBRARY_PATH. You can set it locally in a terminal, so it'll have impact on applications executed only in this terminal. If you add this to ${HOME}/.bashrc or ${HOME}/.bash_profile (assuming you are using bash shell), it'll have impact on application executed by current user. You can also add this to some file in /etc directory. In such case it'll have impact on the whole system.
If you want to use Neo (which requires working IGC libraries) with OpenCL applications I'd recommend adding /usr/local/lib directory to /etc/ld.so.conf.d/igc.conf file with below content:
```
/usr/local/lib
```
and run:
```
sudo ldconfig
```
If you have more libraries installed in /usr/local/lib you can configure it differently.

JacekDanecki, 2019-05-23:
As LD_LIBRARY_PATH helped to run clinfo, I'm closing this issue. Feel free to reopen it or create new issue if you find any problems.

AnonGuy, 2019-06-05:
I'm having the same issues with `clinfo` segfaulting with the same error. I've only just set up the compute runtime on this device, so help would be appreciated.

Output of `strace clinfo`: https://hastebin.com/uhalegohay
Output of `LD_DEBUG=libs clinfo`: https://hastebin.com/iwuvijufud

**What I've tried already:**
* [All troubleshooting steps in the Linux guide](https://github.com/bashbaug/OpenCLPapers/blob/markdown/OpenCLOnLinux.md)

I'm running this on an [Intel i7-7500U (Kaby Lake)](https://ark.intel.com/content/www/us/en/ark/products/95451/intel-core-i7-7500u-processor-4m-cache-up-to-3-50-ghz.html) on Arch Linux, with the relevant [OpenCL packages from the wiki](https://wiki.archlinux.org/index.php/GPGPU) installed.

JacekDanecki, 2019-06-05:
@AnonGuy We've validated Neo release 19.21.13045 with IGC 1.0.6. As I can see in Arch there is newer IGC now (1.0.7). I've reproduced your issue on my setup:
```
arch:/# clinfo
Abort was called at 99 line in file:
/build/intel-compute-runtime/src/compute-runtime-19.21.13045/runtime/built_ins/built_ins.cpp
Aborted (core dumped)
arch:/# pacman -Ss intel-compute-runtime
community/intel-compute-runtime 19.21.13045-1 [installed]
arch:/# pacman -Ss intel-graphics-compiler
community/intel-graphics-compiler 1:1.0.7-1 [installed]
    Intel Graphics Compiler for OpenCL
```

JacekDanecki, 2019-06-05:
It looks like there were some changes on IGC side, which requires modification on Neo side. Let me work with IGC team on this.
@dbermond Can we downgrade IGC in Arch until we have fix on Neo side?

JacekDanecki, 2019-06-05:
@AnonGuy It's enough if we rebuild intel-compute-runtime package with IGC 1.0.7. clinfo and clpeak works fine.

AnonGuy, 2019-06-05:
I see, thanks for the heads up!

AnonGuy, 2019-06-05:
I can confirm that `clinfo` now works without a hitch with the latest update!

dbermond, 2019-06-05:
I've rebuilt intel-compute-runtime on Arch Linux. It's now on the repository.

JacekDanecki, 2019-06-06:
Closing as the issue was resolved.

closed 2019-06-06 by JacekDanecki

hizukiayaka, 2020-05-09:
similar error in ubuntu 18.04
http://paste.ubuntu.com/p/KWyxTG2763/

JacekDanecki, 2020-05-11:
What are Neo and IGC versions installed?