intel/compute-runtime #163: Compilation error on IBIF_PreRelease_Impl.cl with Docker

speryt, 2019-04-25:
When building NEO and it's components according to guide from [here](https://github.com/JacekDanecki/compute-runtime/wiki/How-to-build-Neo-components) I am constantly facing error when building IGC
```
[ 81%] BiF: "IBiF_PreRelease_int.bc": Compiling OpenCL source: "IBIF_PreRelease_Impl.cl"
In file included from <built-in>:2:
/root/build/Release/bif/./opencl_cth.h:6908:1: error: unknown type name 'intel_sub_group_avc_mce_payload_t'
intel_sub_group_avc_mce_payload_t __attribute__((overloadable))
^
/root/build/Release/bif/./opencl_cth.h:6908:35: error: expected identifier or '('
intel_sub_group_avc_mce_payload_t __attribute__((overloadable))
                                  ^
/root/build/Release/bif/./opencl_cth.h:6934:1: error: unknown type name 'intel_sub_group_avc_mce_payload_t'
intel_sub_group_avc_mce_payload_t __attribute__((overloadable))
^

...
```
I've tested it both with master and with suggested commit from the guide, but error is still there.

JacekDanecki, 2019-04-26:
When I use opencl-clang with [commit id](https://github.com/intel/opencl-clang/commit/d29e1fa373f942165465f1d8b9708baf5a02e5e2) IGC compilation finished successfully:
```
CPack: - package: /root/build/intel-0.1.0-Linux-igc-core.deb generated.
CPack: - package: /root/build/intel-0.1.0-Linux-igc-media.deb generated.
CPack: - package: /root/build/intel-0.1.0-Linux-igc-opencl.deb generated.
CPack: - package: /root/build/intel-0.1.0-Linux-igc-opencl-devel.deb generated.
# ls *.deb
intel-0.1.0-Linux-igc-core.deb  intel-0.1.0-Linux-igc-media.deb  intel-0.1.0-Linux-igc-opencl-devel.deb  intel-0.1.0-Linux-igc-opencl.deb
``` 
There is similar issue reported in [IGC project](https://github.com/intel/intel-graphics-compiler/issues/87)

JacekDanecki, 2019-05-23:
@speryt Do you observe compilation issue with opencl-clang commit ID I've mentioned?

speryt, 2019-05-27:
Sorry for a little delay.
Just to clarify - do you mean opencl-clang ID as mentioned in Neo components release (https://github.com/intel/opencl-clang/commit/6257ffe137a2c8df95a3f3b39fa477aa8ed15837)? Or ocl-open-70 as mentioned in the guide?

I tried with the first option and from yet unknown reason it is switching on cmake to ocl-open-70 by itself.

JacekDanecki, 2019-05-27:
I meant opencl-clang commit ID, because I was using commands:
```
cd opencl-clang; git checkout d29e1fa373f942165465f1d8b9708baf5a02e5e2
```
on updated [wiki page](https://github.com/JacekDanecki/compute-runtime/wiki/How-to-build-Neo-components). 
If you want to use newer compiler components (igc, opencl-clang, llvm-spirv-translator), please use commits id's provided on [releases page](https://github.com/intel/compute-runtime/releases).

opencl-clang during configuration (cmake command) apply [patches](https://github.com/intel/opencl-clang/tree/ocl-open-70/patches/clang) on clang and other [patches](https://github.com/intel/opencl-clang/tree/ocl-open-70/patches/spirv) on llvm-spirv-translator, after switching to ocl-open-70 branch. See [this function](https://github.com/intel/opencl-clang/blob/ocl-open-70/cmake/modules/CMakeFunctions.cmake#L47)

speryt, 2019-05-29:
Ok, it seems to be working now. Thanks!

closed 2019-05-29 by speryt