intel/compute-runtime #165: Duplicated .so file in Ubuntu 16.04 pre-compiled binaries

umaplehurst, 2019-05-07:
Hi there,

An optimization can be made to the Ubuntu binary packages which will reduce the amount of disk space that is used up; namely, `libopencl-clang.so` should be a symlink to `libopencl-clang.so.7` as is already the case for `libigdfcl.so`:

```
$ dpkg -x intel-igc-opencl_1.0.3-1844_amd64.deb out/
$ ls -l out/usr/local/lib/
total 171064
-rw-r--r-- 1 user user 59496880 Apr 19 10:47 clang
lrwxrwxrwx 1 user user       14 Apr 19 10:51 libigdfcl.so -> libigdfcl.so.1
lrwxrwxrwx 1 user user       18 Apr 19 10:51 libigdfcl.so.1 -> libigdfcl.so.1.0.3
-rw-r--r-- 1 user user  1239888 May  7 11:14 libigdfcl.so.1.0.3
-rw-r--r-- 1 user user 56793368 May  7 11:14 libopencl-clang.so
-rw-r--r-- 1 user user 56793368 May  7 11:14 libopencl-clang.so.7
-rw-r--r-- 1 user user   835955 Apr 16 23:12 opencl-c.h
``` 

Thanks!

JacekDanecki, 2019-05-08:
It'll be fixed soon on IGC side. When we get newer version of IGC, it'll be in Neo release.

JacekDanecki, 2019-06-14:
It was fixed on IGC side, and new IGC build was taken to Neo release [19.23.13131](https://github.com/intel/compute-runtime/releases/tag/19.23.13131).
Closing issue as fixed.

closed 2019-06-14 by JacekDanecki