intel/compute-runtime #166: Need cl_khr_gl_sharing

zhoub, 2019-05-09:
Hi

I found that this driver doesn't support *cl_khr_gl_sharing*, is there any plan for this ? Thanks !

PiotrRozenfeld, 2019-05-09:
At this point this extension is supported only on Windows. There are currently no plans to implement this extension on Linux.

zhoub, 2019-05-10:
Hi

This is actually important as Windows platform. So many video applications require the OpenCL and OpenGL interop, and now our work is based on Intel Up board. Please consider this. Thanks !

zhoub, 2019-05-21:
```
Number of platforms                               3
  Platform Name                                   Intel(R) OpenCL HD Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 1.2 
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_depth_images cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_image2d_from_buffer cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_media_block_io cl_intel_driver_diagnostics cl_intel_device_side_avc_motion_estimation cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_khr_fp64 cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_motion_estimation cl_intel_advanced_motion_estimation cl_intel_va_api_media_sharing 
  Platform Extensions function suffix             INTEL

  Platform Name                                   Intel Gen OCL Driver
  Platform Vendor                                 Intel
  Platform Version                                OpenCL 1.2 beignet 1.3 (git-5aba95a)
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_spir cl_khr_icd cl_intel_accelerator cl_intel_subgroups cl_intel_subgroups_short cl_khr_gl_sharing
  Platform Extensions function suffix             Intel

  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 18.0.5
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Intel(R) OpenCL HD Graphics
Number of devices                                 1
  Device Name                                     Intel(R) Gen9 HD Graphics NEO
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 1.2 NEO 
  Driver Version                                  19.14.12751
  Device OpenCL C Version                         OpenCL C 1.2 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Max compute units                               18
  Max clock frequency                             750MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
  Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
  Preferred work group size multiple              32
  Preferred / native vector sizes                 
    char                                                16 / 16      
    short                                                8 / 8       
    int                                                  4 / 4       
    long                                                 1 / 1       
    half                                                 8 / 8        (cl_khr_fp16)
    float                                                1 / 1       
    double                                               1 / 1        (cl_khr_fp64)
  Half-precision Floating-point support           (cl_khr_fp16)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  No
  Single-precision Floating-point support         (core)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  Yes
  Double-precision Floating-point support         (cl_khr_fp64)
    Denormals                                     Yes
[... 141 lines cut ...]
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [INTEL]
  clCreateContext(NULL, ...) [default]            Success [INTEL]
  clCreateContext(NULL, ...) [other]              Success [Intel]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 Intel(R) OpenCL HD Graphics
[... cut ...]
  ICD loader Version                              2.2.8
  ICD loader Profile                              OpenCL 1.2
	NOTE:	your OpenCL library declares to support OpenCL 1.2,
		but it seems to support up to OpenCL 2.1 too.
```

In contrast, very funny.

The device *Intel(R) Gen9 HD Graphics NEO* has all video-related extensions but no OpenGL sharing, another device *Intel(R) HD Graphics Broxton 0* has all graphics extensions but no VAAPI sharing. It means either 1 or 2, but no possible to use OpenGL to visualize the thing from 1, really sad.

juliusmh, 2019-07-20:
Id reaally like to see the CL GL context sharing aswell :+1:  .

dasco597, 2019-07-30:
same here, I'm really disappointed that I can't port my application to linux with intel cpus because of this. cl_khr_gl_sharing seems like such a key extension to have I'm surprised you're not supporting it.

dasco597, 2019-07-30:
I also tried using the beignet implementation and while it does support the cl_khr_gl_sharing extension, one of the functions I required, cl_mem_new_gl_buffer was empty but for a line of code:
`FATAL ("Not implemented")`
tears all round.

PiotrRozenfeld, 2019-08-02:
Your arguments are convincing. 
We will fit this effort into our development schedule for the remainder of this year.

dasco597, 2019-08-02:
wow that's great to hear! your hard work is very much appreciated. best of luck to your team!

ormf, 2019-10-17:
Hi. Yesterday my app seized working as it used cl_khr_gl_sharing with the beignet driver on linux and that seems to be disfunctional with my most recent update now. For my compositions it is absolutely crucial to work, so I'm quite devastated, as it is more or less impossible to implement my system on other Platforms/OSes (see: https://www.youtube.com/watch?v=2rWha1HTfFE&t=360s). I'd be extremely grateful if you'd implemented it and am absolutely willing to support funding if you let me know how!

ormf, 2019-12-22:
@PiotrRozenfeld
> We will fit this effort into our development schedule for the remainder of this year.

Are there any news about whether work has been done on this and whether/when it will be implemented?

AdamCetnerowski, 2019-12-23:
This is being actively worked on. We don't have a clear trend date, but Q1 is very likely.

ormf, 2020-01-16:
> This is being actively worked on. We don't have a clear trend date, but Q1 is very likely.

@AdamCetnerowski: Thanks, very much appreciated!

pioto1225, 2020-03-17:
Has there been any progress on this issue?

smistad, 2020-03-17:
I would also very much like to see this feature implemented

AdamCetnerowski, 2020-03-19:
Some foundational work has been done to enable the extension. At the this time the Q1 trend is no longer valid. We intend to provide a partial implementation early Q2 for evaluation. We’re looking forward to your feedback when it’s available.

ormf, 2020-05-20:
> We’re looking forward to your feedback when it’s available.

Most definitely! Right now I'm holding back any upgrades on my system as I have to keep llvm 8.0.1 for the beignet drivers.

AdamCetnerowski, 2020-07-10:
We have made progress towards this feature, but it was slower than initially expected. We’ll update this issue when we have something that is user-testable.”

Apahdos, 2020-09-13:
Hi. Any news on when this will be released? Could use it for my thesis. Otherwise I need to find a workaround ^^

PiotrRozenfeld, 2020-09-14:
While we did lay some groundwork for cl_khr_gl_sharing with MESA on Linux, we also had to increase our initial effort estimates, as  additional work to be done in MESA and Compute before we can proceed with implementation.
This does not mean we abandon the effort, but we can no longer accommodate that in the near future. 

We'll post updates when the work will resume.

ormf, 2020-09-14:
@PiotrRozenfeld 
thanks a lot for the info! Although this is very bad news for me, I now know that I'll either have to get beignet compiled with llvm 10, or resort to installing a dual boot until this is resolved. I sincerely hope this is not the final blow to this in compute-runtime.

Randrianasulu, 2020-10-04:
Probably not most useful comment (I'm not a developer, just user whole likes to test something new) but I was looking at AMD's ROCM sources and found there 

https://github.com/ROCm-Developer-Tools/ROCclr/blob/master/device/rocm/mesa_glinterop.h
```
/* Mesa OpenGL inter-driver interoperability interface designed for but not
 * limited to OpenCL.
```

Of course this is only smallest part of whole work .

mathucub, 2021-01-19:
@PiotrRozenfeld any ETA of having a cl_khr_gl_sharing solution going? This extension missing causes a major problem since most of our linux machines are using Intel integrated GPUs.

chamois94, 2021-05-20:
> While we did lay some groundwork for cl_khr_gl_sharing with MESA on Linux, we also had to increase our initial effort estimates, as additional work to be done in MESA and Compute before we can [...]

Any news on that ? This extension id really a must have for Manu applications.

smlehbleh, 2021-06-09:
Also curious how this is going? We are building and deploying Intel NUC based video acquisition/processing solutions and the lack of this extension is the only reason we're still on Beignet drivers

PiotrRozenfeld, 2021-06-28:
The work on this extension has stalled due to other work that our team is facing, unfortunately.
 
@smlehbleh - could you share a sample workload that you are looking to enable? What is your platform of interest? You are indicating this already works on Beignet - implying no additional work in MESA is needed.

smlehbleh, 2021-06-28:
Hi Piotr,
Our use case is an Intel NUC powered cinema camera/recording device. The RAW image processing (debayering and colour transformation etc...) is performed in OpenCL kernels and the user interface is rendered with OpenGL ES. We use Beignet OpenCL drivers which support the cl_khr_gl_sharing extension to share the processed RAW image from an OpenCL buffer with OpenGL ES to draw it to the monitor screen.
We are using Ubuntu 18.04 with an Intel NUC8v7PNB and (soon) an NUC11TNBv7. The cl_khr_gl_sharing extension implementation in Beignet has been present for quite a while (Maybe a couple of years...).
For an experiment we tried the NEO CL driver and replaced our 'cl_khr_gl_sharing' GL/CL interop code with a manual additional copy from a CL buffer into a GL texture - the total CPU usage appeared to go from about 10% to 15-20% and a noticeable frame delay was introduced. The buffer was a UHD RGBA32 image (~33mb).

smlehbleh, 2021-08-20:
Hi @PiotrRozenfeld , I was just wondering if Beignet's existing cl_khr_gl_sharing support/implementation could be ported or copied over into this driver - or if it could generally reduce the amount of work as you mentioned?

JacekDanecki, 2021-09-01:
As I can see in Beignet documentation cl_khr_gl_sharing is partially supported, and allow to create memory objects from OpenGL buffers or 2D textures.
If I understand your use case correctly you're using cl/gl sharing differently, and you share OpenCL buffers with OpenGL, so memory for buffers is allocated by OpenCL not by OpenGL. Could you clarify?

smlehbleh, 2021-09-01:
Hi Jacek,
We are using the Beignet implementation of cl_khr_gl_sharing in the standard intended use case - we create an OpenGL texture and then create an OpenCL buffer from it using 'clCreateFromGLTexture'.
I realise in the previous comment it seems like we're directly sharing an OpenCL buffer with OpenGL, but our currently active implementation is an OpenCL buffer created from an OpenGL texture (clCreateFromGLTexture). 
This allows us to draw the GL texture to the screen with OpenGL after an OpenCL kernel writes to the underlying buffer without any additional 'copy'.

ormf, 2021-09-01:
Hi Jacek,

Am Mittwoch, den 01. September 2021 um 05:26:00 Uhr (-0700) schrieb
Jacek Danecki:
> As I can see in Beignet documentation cl_khr_gl_sharing is partially
> supported, and allow to create memory objects from OpenGL buffers or
> 2D textures.
>
> If I understand your use case correctly you're [...]

 it's unclear whether you address me in your mail.

The way I used it with the Beignet driver was to first allocate the
buffers in OpenGL (using "gen-buffer") and then "recreating" them in
OpenCL (within the opengl context with the allocated OpenGL buffer
mapped) by calling the OpenCL routine "create-buffer" with the
USE-HOST-PTR flag.

Is that what you were asking?

Best,
Orm

mathucub, 2021-09-01:
I'm going to chime in, even though it has taken so long to get any traction that I've change jobs so no longer care.
GL and CL are supposed to be able to interop with calls like clCreateFromGLTexture and stay in the GPU address space.

This allows for work to be done in CL, but presented in GL in realtime -- which any copy to host techniques will quash. This has been as baseline feature in nVidia and AMD's implementation for years.

The Intel stack on Linux does not match what the Intel stack on Windows can do, that is the baseline issue of what is wrong. 
They have to be in parity otherwise software is trapped on Windows since it can't be ported.

JacekDanecki, 2021-09-20:
In Beignet code there is example: https://github.com/intel/beignet/blob/master/examples/gl_buffer_sharing/gl_buffer_sharing.cpp, and it looks similar to your usages. After small changes in Beignet code I was able to run this example on Ubuntu 18.04 and it worked correctly. Looking at sharing implementation in Beignet I can see it uses https://www.khronos.org/registry/EGL/extensions/MESA/EGL_MESA_image_dma_buf_export.txt, so I suppose it should be enough on mesa side to implement basic sharing in Neo.

JacekDanecki, 2021-09-22:
I've created new repository [cl-gl-tests](https://github.com/JacekDanecki/cl-gl-tests) where I've pushed modified example from Beignet. I'll use this repo to test cl/gl implementation in Neo, so if you see any potential issues with this test in comparison to your usage scenario, please let me know. In first step I'll prepare some changes in Neo to run only tests from this repo.
When I've any working Neo code I'll push it to my fork: https://github.com/JacekDanecki/compute-runtime on clgl branch.

JacekDanecki, 2021-09-29:
With commit https://github.com/JacekDanecki/compute-runtime/commit/b95bc31fd7547cf5da7560bea822dca159c9e07d test gl_buffer_sharing from: https://github.com/JacekDanecki/cl-gl-tests is working now. 
This is still very experimental code, under development, there are some hard coded values, and some parts of code are stubbed. I've only executed gl_buffer_sharing test so far.

If you have any tests, you can share, I can focus on enabling them.

JacekDanecki, 2021-10-06:
@smlehbleh @ormf any feedback on experimental code/test I've provided?

smlehbleh, 2022-01-27:
Hi @JacekDanecki, sorry was pulled onto other things, but I'm back on this now. I'm not too familiar with building the necessary .deb packages from the compute-runtime repo. Are you able to enable CI on your forked repo to make some .deb packages show up in the releases pane? Then I'll give them a go with some gl_cl interop samples I've got.
Cheers, Russell

smlehbleh, 2022-01-31:
Ubuntu 20.04.3 on an 11th Gen Intel NUC

JacekDanecki, 2022-02-02:
I'll rebase my current code to the latest Neo release and push to my fork, so you can use igc and gmmlib packages from release.

smlehbleh, 2022-02-03:
Sounds great, thanks, let me know when to grab the packages

Abdob, 2022-02-07:
"Right now, Intel does not support surface sharing on Linux* for example. Customer requests can drive changes to this decision."
Last Updated: 12/15/2014

By Adam T Lake, Robert M Ioffe

https://www.intel.com/content/www/us/en/developer/articles/technical/sharing-surfaces-between-opencl-and-opengl-43-on-intel-processor-graphics-using-implicit.html

Is this still the case today?

smlehbleh, 2022-02-07:
@Abdob, this is my understanding/experience:

It depends on the driver. Beignet OCL driver (1.3.2-6) supports the gl/cl sharing extension. The NEO Compute runtime OCL driver does not. However, @JacekDanecki has an experimental/wip branch with the feature using I think an approach similar to the Beignet driver. 

Unfortunately the Beignet driver doesn't appear to support 11th Gen Intel GPUs (Xe) so for the latest hardware, cl/gl sharing support is not in any official driver release.

Abdob, 2022-02-08:
Hi @smlehbleh 

My intel GPU is:
glxinfo | grep Mesa
client glx vendor string: Mesa Project and SGI
    Device: Mesa Intel(R) UHD Graphics 630 (CML GT2) (0x9bc5)
OpenGL renderer string: Mesa Intel(R) UHD Graphics 630 (CML GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.0.3
OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.0.3
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.0.3

I install beignet-opencl-icd and clinfo and I get this:

wave@wave:~$ clinfo
Number of platforms                               1
  Platform Name                                   Intel Gen OCL Driver
  Platform Vendor                                 Intel
  Platform Version                                OpenCL 2.0 beignet 1.3
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_spir cl_khr_icd cl_intel_accelerator cl_intel_subgroups cl_intel_subgroups_short
  Platform Extensions function suffix             Intel
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)

  Platform Name                                   Intel Gen OCL Driver
Number of devices                                 0

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  No platform
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   No platform
  clCreateContext(NULL, ...) [default]            No platform
  clCreateContext(NULL, ...) [other]              No platform
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  No devices found in platform

Is cl_khr_gl_sharing suppose to be listed as a device extension?

Thanks,
Abdo

JacekDanecki, 2022-02-08:
@Abdob  cl_khr_gl_sharing is listed on my setup with Beignet
In case of CML support in Beignet see: https://bugs.launchpad.net/ubuntu/+source/beignet/+bug/1905340

glxinfo grep Mesa
```
client glx vendor string: Mesa Project and SGI
    Device: Mesa Intel(R) Iris(R) Pro Graphics 580 (SKL GT4) (0x193b)
OpenGL renderer string: Mesa Intel(R) Iris(R) Pro Graphics 580 (SKL GT4)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.0.3
OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.0.3
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.0.3
```
clinfo | grep cl_khr_gl
```
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_spir cl_khr_icd cl_intel_accelerator cl_intel_subgroups cl_intel_subgroups_short cl_intel_required_subgroup_size cl_intel_media_block_io cl_intel_planar_yuv cl_khr_gl_sharing
  Device Extensions                               cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_spir cl_khr_icd cl_intel_accelerator cl_intel_subgroups cl_intel_subgroups_short cl_intel_required_subgroup_size cl_intel_media_block_io cl_intel_planar_yuv cl_khr_gl_sharing cl_khr_fp16 cl_intel_device_side_avc_motion_estimation
```
clinfo
```
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_spir cl_khr_icd cl_intel_accelerator cl_intel_subgroups cl_intel_subgroups_short cl_intel_required_subgroup_size cl_intel_media_block_io cl_intel_planar_yuv cl_khr_gl_sharing
  Device Extensions                               cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_spir cl_khr_icd cl_intel_accelerator cl_intel_subgroups cl_intel_subgroups_short cl_intel_required_subgroup_size cl_intel_media_block_io cl_intel_planar_yuv cl_khr_gl_sharing cl_khr_fp16 cl_intel_device_side_avc_motion_estimation
root@gubuntuPC:/home/jdanecki# cat clinfo.txt
Number of platforms                               1
  Platform Name                                   Intel Gen OCL Driver
  Platform Vendor                                 Intel
  Platform Version                                OpenCL 2.0 beignet 1.4 (git-419c0417)
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_spir cl_khr_icd cl_intel_accelerator cl_intel_subgroups cl_intel_subgroups_short cl_intel_required_subgroup_size cl_intel_media_block_io cl_intel_planar_yuv cl_khr_gl_sharing
  Platform Extensions function suffix             Intel

  Platform Name                                   Intel Gen OCL Driver
Number of devices                                 1
  Device Name                                     Intel(R) HD Graphics Skylake Halo GT4
  Device Vendor                                   Intel
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.0 beignet 1.4 (git-419c0417)
  Driver Version                                  1.4
  Device OpenCL C Version                         OpenCL C 2.0 beignet 1.4 (git-419c0417)
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               72
  Max clock frequency                             1000MHz
  Device Partition                                (core)
    Max number of sub-devices                     1
    Supported partition types                     None, None, None
    Supported affinity domains                    (n/a)
  Max work item dimensions                        3
  Max work item sizes                             512x512x512
  Max work group size                             256
  Preferred work group size multiple              16
  Sub-group sizes (Intel)                         8, 16
  Preferred / native vector sizes                 
    char                                                16 / 8       
    short                                                8 / 8       
    int                                                  4 / 4       
    long                                                 2 / 2       
    half                                                 0 / 8        (cl_khr_fp16)
    float                                                4 / 4       
[... cut ...]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 Intel Gen OCL Driver
    Device Name                                   Intel(R) HD Graphics Skylake Halo GT4

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1
```

Abdob, 2022-02-08:
Hi @JacekDanecki 

Thank you for the follow up. 

According to the link you provided beignet 1.3.2-8 fixed the issue. I installed 1.4 from source and it still doesn't show. We decided to move forward with another API and will no longer attempt to get the OpenCL interoperability to work.

wave@wave:~/Documents/ocl/beignet/build$ clinfo
Number of platforms                               1
  Platform Name                                   Intel Gen OCL Driver
  Platform Vendor                                 Intel
  Platform Version                                OpenCL 2.0 beignet 1.4 (git-419c0417)
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_spir cl_khr_icd cl_intel_accelerator cl_intel_subgroups cl_intel_subgroups_short cl_intel_required_subgroup_size cl_intel_media_block_io cl_intel_planar_yuv
  Platform Extensions function suffix             Intel
cl_get_gt_device(): error, unknown device: 9bc5

  Platform Name                                   Intel Gen OCL Driver
Number of devices                                 0

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  No platform
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   No platform
  clCreateContext(NULL, ...) [default]            No platform
  clCreateContext(NULL, ...) [other]              No platform
cl_get_gt_device(): error, unknown device: 9bc5
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
cl_get_gt_device(): error, unknown device: 9bc5
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
cl_get_gt_device(): error, unknown device: 9bc5
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  No devices found in platform

eero-t, 2022-02-08:
> According to the link you provided beignet 1.3.2-8 fixed the issue.

@Abdob That link was to Ubuntu bug tracker, and that fix was into a _Ubuntu_ (and Debian) 1.3.2-8 package of beignet, not to beignet upstream project.

> I installed 1.4 from source and it still doesn't show.
...
> Platform Version OpenCL 2.0 beignet 1.4 (git-419c0417)
> cl_get_gt_device(): error, unknown device: 9bc5

If you built it from upstream, and it's this one (which I found by googling), quite obviously it's missing CML support:
https://github.com/intel/beignet/commits/master/src/cl_device_data.h

As the PR for that has not been merged: https://github.com/intel/beignet/pull/20

You should have built e.g. the Debian version (with Debian tools, so that CML support patch gets applied on top of upstream sources): https://salsa.debian.org/opencl-team/beignet

smlehbleh, 2022-02-12:
Hi  @JacekDanecki, I was wondering if you were able to build and generate the NEO compute runtime .deb files for Ubuntu 20.04from your fork with the experimental/wip cl/gl sharing?

JacekDanecki, 2022-02-14:
@smlehbleh I've pushed current experimental implementation rebased on release https://github.com/intel/compute-runtime/releases/tag/22.06.22433 to clgl-fork branch in repository: https://github.com/JacekDanecki/compute-runtime/tree/clgl-fork
You can  use gmmlib and igc packages from Neo release.

Neo can be compiled with commands:
```
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Debug -DDISABLE_WDDM_LINUX=1 -DBUILD_WITH_L0=1 -DNEO_DISABLE_LD_GOLD=1 ..
make  igdrcl_dll
```
Once compiled you can create file neo.icd in /etc/OpenCL/vendors directory containing full path to libigdrcl.so library.

JacekDanecki, 2022-03-03:
I'm leaving compute team, so will not work on cl/gl sharing anymore.

Abdob, 2022-04-12:
@smlehbleh 

I'm able to build  https://salsa.debian.org/opencl-team/beignet with an ubuntu 16 docker container, I see that it has the cl_khr_gl_sharing extension.

Thanks,
Abdo

AwesomeCronk, 2022-04-20:
Tried Beignet, it doesn't support my CPU. -_-
Also tried building @JacekDanecki 's fork, but even though I installed IGC in order to run normal NEO (which works), the CMake script failed to find it.

smlehbleh, 2022-04-20:
Hi @JacekDanecki - is it possible someone else will pick up the work? It seems like not a massive amount of dev work is needed to get this highly desirable feature over the line.

JacekDanecki, 2022-04-21:
@smlehbleh Please wait for an answer from the team, as I'm not a member of compute team anymore, so can't answer your question.
@AwesomeCronk It'll be helpful if you provide log from cmake.

AwesomeCronk, 2022-04-21:
I'll try and get one this afternoon @JacekDanecki . Should I put it here?

JacekDanecki, 2022-04-21:
@AwesomeCronk copy it here, or add the link to the log.

AwesomeCronk, 2022-04-21:
@JacekDanecki I believe this is the right log. (`build/CMakeFiles/CMakeOutput.log`)
[CMakeOutput.log](https://github.com/intel/compute-runtime/files/8530732/CMakeOutput.log)

PoggPoggBeanMan, 2022-04-22:
Would love to see this as well.

seadra, 2022-05-13:
The lack of this extension has long prevented DaVinci Resolve and DaVinci Resolve Studio from functioning on Linux (the editor starts, but no visuals from the videos appear).

kallaballa, 2022-10-06:
> @smlehbleh I've pushed current experimental implementation rebased on release https://github.com/intel/compute-runtime/releases/tag/22.06.22433 to clgl-fork branch in repository: [...]

Based on your work I've got a full example of OpenGL -> OpenCL -> VAAPI working on my Tigerlake (i7-1160G7) CPU . Stay tuned for full instructions and an example based on OpenCV.

https://github.com/kallaballa/compute-runtime

kallaballa, 2022-10-06:
I had to fork the 4.x branch of OpenCV and still work on cleaning my fork up, but once I pushed that (and the instructions) you'll be able to do something like the following on a Tigerlake CPU (and maybe others) running Linux:

```c++
int main(int argc, char **argv) {
	using namespace kb;

	if (!va::open_display())
		throw std::runtime_error("Failed to open VA display for CL-VA interoperability");

	va::check_if_YUV420_available();

	cv::va_intel::ocl::initializeContextFromVA(va::display, true);
	cv::ocl::OpenCLExecutionContext vaContext = cv::ocl::OpenCLExecutionContext::getCurrentRef().cloneWithNewQueue();

	VideoWriter video("out.mkv", CAP_FFMPEG, VideoWriter::fourcc('V', 'P', '9', '0'), FPS, Size(WIDTH, HEIGHT), {
			VIDEOWRITER_PROP_HW_DEVICE, 0,
			VIDEOWRITER_PROP_HW_ACCELERATION, VIDEO_ACCELERATION_VAAPI,
			VIDEOWRITER_PROP_HW_ACCELERATION_USE_OPENCL, 1
	});

	egl::init_egl();
	gl::init_gl();
	cv::ocl::OpenCLExecutionContext glContext = cv::ocl::OpenCLExecutionContext::getCurrentRef().cloneWithNewQueue();

	UMat frameBuffer(HEIGHT, WIDTH, CV_8UC4, Scalar::all(0));
	UMat grey(HEIGHT, WIDTH, CV_8UC1, Scalar::all(0));
	UMat bgr(HEIGHT, WIDTH, CV_8UC3, Scalar::all(0));

	while (true) {
		//Draw a large triangle on blue background using OpenGL
		glContext.bind();
		gl::begin_gl();
		glClearColor(1.0f, 0.0f, 0.0f, 1.0f);
		glClear(GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT);
		glBegin(GL_POLYGON);
		glColor3f(1, 0, 0);
		glVertex3f(-0.6, -0.75, 0.5);
		glColor3f(0, 1, 0);
		glVertex3f(0.6, -0.75, 0);
		glColor3f(0, 0, 1);
		glVertex3f(0, 0.75, 0);
		glEnd();
		glFlush();
		gl::end_gl();
		gl::fetch_frame_buffer(frameBuffer); //hand over the data (GPU 2 GPU) to OpenCV/OpenCL

		//do colorspace conversions using OpenCL in the background
		cvtColor(frameBuffer, grey, COLOR_BGRA2GRAY);
		cvtColor(grey, bgr, COLOR_BGRA2BGR);

		vaContext.bind();
		//encode the frame using VAAPI on the GPU.
		video.write(bgr);
	}

	return 0;
}
```

kallaballa, 2022-10-06:
this branch contains the changes it required to get OpenCV to play nicely: https://github.com/kallaballa/opencv/tree/GCV
this branch contains the changes to compute-runtime it needed: https://github.com/kallaballa/compute-runtime/tree/clgl-fork
and this repo contains the full demo source code: https://github.com/kallaballa/GCV

Proper instructions follow on the weekend :)

[33 more replies]

closed 2023-03-17 by Compute-Runtime-Automation