intel/compute-runtime #167: Regression with the last NEO drivers (OpenCL extensions)

Nyrio, 2019-05-09:
This is the sequel of https://github.com/CNugteren/CLBlast/issues/355
@JacekDanecki suggested following up here

To reproduce the issue, you can use the samples from the CLBlast repository (disclaimer: it's not what I'm using but I've been able to reproduce the issue with them).

* install the latest NEO drivers (my version is `19.16.12873`)
* clone [CLBlast](https://github.com/CNugteren/CLBlast.git)
* edit `samples/sgemm.cpp` with the platform and device number corresponding to Intel GPU, e.g I had to change the platform id to 1
* build with the samples activated:
  ```
  mkdir build
  cmake -GNinja -DSAMPLES=ON ..
  ninja
  ```
* run `clblast_sample_sgemm`

What I get is (multiple times):
```
In file included from <built-in>:2:
CTHeader.h:6913:7: error: use of type 'intel_sub_group_avc_mce_payload_t' requires cl_intel_device_side_avc_motion_estimation extension to be enabled
      intel_sub_group_avc_mce_payload_t payload );
```

More info about what we tried in the original thread.

JacekDanecki, 2019-05-09:
I've reproduction on my side. Let me look at this.
```
CLBlast/build$ ./clblast_sample_dgemv_c
OpenCL compiler error/warning:
In file included from <built-in>:2:
CTHeader.h:6918:7: error: use of type 'intel_sub_group_avc_mce_payload_t' requires cl_intel_device_side_avc_motion_estimation extension to be enabled
      intel_sub_group_avc_mce_payload_t payload );
      ^
```

JacekDanecki, 2019-05-10:
@Nyrio I've dumped opencl kernel generated by CLBlast and found there is not pragma included to enable cl_intel_subgroups extension, so I've executed clblast_sample_dgemv_c under gdb and found in function
 clblast::CompileFromSource check for precision:
```
   // For Intel GPUs with subgroup support, use subgroup shuffling.
  if (device.IsGPU() && device.HasExtension(kKhronosIntelSubgroups) &&
      (precision == Precision::kSingle || precision == Precision::kHalf)) {
     header_string += "#define USE_SUBGROUP_SHUFFLING 1\n";
     header_string += "#define SUBGROUP_SHUFFLING_INTEL 1\n";
   }
```
under Neo precision is set to: 
```
(gdb) p precision
$2 = clblast::Precision::kDouble
```
so defines for subgroups are not added to header_string.
When I remove this check with below patch
```
 git diff
diff --git a/src/utilities/compile.cpp b/src/utilities/compile.cpp
index 00cb90c..e63ea5e 100644
--- a/src/utilities/compile.cpp
+++ b/src/utilities/compile.cpp
@@ -59,8 +59,8 @@ std::shared_ptr<Program> CompileFromSource(
   }
 
   // For Intel GPUs with subgroup support, use subgroup shuffling.
-  if (device.IsGPU() && device.HasExtension(kKhronosIntelSubgroups) &&
-      (precision == Precision::kSingle || precision == Precision::kHalf)) {
+  if (device.IsGPU() && device.HasExtension(kKhronosIntelSubgroups) ) 
+  {
     header_string += "#define USE_SUBGROUP_SHUFFLING 1\n";
     header_string += "#define SUBGROUP_SHUFFLING_INTEL 1\n";
   }
```
defines are added to dumped kernel:
```
diff -u kernel.cl kernel1.cl 
--- kernel.cl   2019-05-10 13:53:46.082682559 +0200
+++ kernel1.cl  2019-05-10 13:53:35.555437090 +0200
@@ -1,5 +1,7 @@
 #define PRECISION 64
 #define ROUTINE_GEMV
+#define USE_SUBGROUP_SHUFFLING 1
+#define SUBGROUP_SHUFFLING_INTEL 1
```
but there  is still #pragma missing for subgroups in generated kernel. Can you verify CLBlast code?

Nyrio, 2019-05-10:
@JacekDanecki The PR that @CNugteren made for this issue (https://github.com/CNugteren/CLBlast/pull/357) adds the pragma, can you checkout that branch?

Also, I have tested this branch with single precision, it reached the `#pragma` statement and displayed the additional error message:
```
1:2179:28: warning: unsupported OpenCL extension 'cl_intel_subgroups' - ignoring
  #pragma OPENCL EXTENSION cl_intel_subgroups: enable
```

Or, in the older version of the PR:
```
1:2179:28: warning: unsupported OpenCL extension 'cl_intel_device_side_avc_motion_estimation' - ignoring
  #pragma OPENCL EXTENSION cl_intel_device_side_avc_motion_estimation: enable
```

JacekDanecki, 2019-05-10:
I was using PR from @CNugteren, but with different sample. When I use clblast_sample_sgemm I can reproduce your issue with:
```
1:2179:28: warning: unsupported OpenCL extension 'cl_intel_subgroups' - ignoring
  #pragma OPENCL EXTENSION cl_intel_subgroups: enable
                                                  ^
```
As I can see in parameters provided to  clBuildProgram
```
ThreadID: 140503422388032 Function Enter: clBuildProgram
------------------------------
	ThreadID: 	140503422388032
	clProgram: 	0x2756a80
	numDevices: 	1
	cl_device_id: 	0x7ffc941f4678
	options: 	 -cl-std=CL1.1
	funcNotify: 	0
	userData: 	0
```
it was called with CL 1.1, but in [cl_intel_subgroups spec](https://www.khronos.org/registry/OpenCL/extensions/intel/cl_intel_subgroups.html) OpenCL 1.2 is required.
So I've tested generated opencl kernel with ocloc, with -cl-std=CL1.1
```
1:2179:28: warning: unsupported OpenCL extension 'cl_intel_subgroups' - ignoring
  #pragma OPENCL EXTENSION cl_intel_subgroups: enable
                           ^

Build failed with error code: -11
```

with "-cl-std=CL1.2"
```
 ./ocloc -file kernel3.cl -device skl -options "-cl-std=CL1.2"
Build succeeded.
```
When I apply below patch:
```
git diff
diff --git a/src/clpp11.hpp b/src/clpp11.hpp
index 786f7db..a52e298 100644
--- a/src/clpp11.hpp
+++ b/src/clpp11.hpp
@@ -485,7 +485,7 @@ class Program {
 
   // Compiles the device program and checks whether or not there are any warnings/errors
   void Build(const Device &device, std::vector<std::string> &options) {
-    options.push_back("-cl-std=CL1.1");
+    options.push_back("-cl-std=CL1.2");
     auto options_string = std::accumulate(options.begin(), options.end(), std::string{" "});
     const cl_device_id dev = device();
     CheckError(clBuildProgram(program_, 1, &dev, options_string.c_str(), nullptr, nullptr));
```

sample clblast_sample_sgemm works
```
./clblast_sample_sgemm
Completed SGEMM in 545.114 ms with status 0
```

Nyrio, 2019-05-10:
I posted an update on the CLBlast issue then.

JacekDanecki, 2019-05-10:
I think I can close issue here. If you find new problem, just let us know.

closed 2019-05-10 by JacekDanecki

CNugteren, 2019-05-11:
Thanks both for looking into this. I suspect this will now work :+1: 

Still one question though, to me it seems as the original error message @Nyrio was getting is wrong, which was:
```
OpenCL compiler error/warning:
In file included from <built-in>:2:
CTHeader.h:6913:7: error: use of type 'intel_sub_group_avc_mce_payload_t' requires cl_intel_device_side_avc_motion_estimation extension to be enabled
      intel_sub_group_avc_mce_payload_t payload );
```
This appears when using `get_sub_group_local_id()` and `intel_sub_group_shuffle()` functions. Now we know that the solution is to enable `cl_intel_subgroups` (the right extension) and use OpenCL 1.2, but that's not what the error message suggests.