intel/compute-runtime #168: compute-runtime project only for 4th or newer gen processors?

maverick6664, 2019-05-10:
I'm using i7-4790K for Fedora 30.   Until Fedora 29, beignet comes with Fedora, but it's no longer with Fedora package.  So now I'm using beignet package for Fedora 29 on Fedora 30 system.   Fortunately it works so far, but since beignet is no longer active, Linux OpenCL for Intel GPU is moving to OpenCL™ Runtimes for Intel® Processors.

There are still many users using 4th gen or earlier core processors.   So I want this project to support such processors...

Thanks.

-Tetsuji

PiotrRozenfeld, 2019-05-21:
Michael provided an answer in a copy of this question on [Intel Software Forums](https://software.intel.com/en-us/forums/opencl/topic/809496). If you have additional questions, feel free to ask them here or there.

86423355844265459587182778, 2019-05-22:
Will there be support for CHT/BSW platforms (Graphics microarchitecture Gen8LP, according to [this document](https://01.org/sites/default/files/documentation/intel-gfx-bspec-osrc-chv-bsw-vol04-configurations.pdf))?

PiotrRozenfeld, 2019-05-28:
@86423355844265459587182778 -  Cherry Trail / Braswell support is technically possible in NEO. We had no requests to provide it so far – everyone seemed to be using Beignet. Are you personally interested or just curious? If we decide to implement experimental support for Gen8LP, would you be willing to work with us on enabling and debug?

maverick6664, 2019-05-28:
> Will there be support for CHT/BSW platforms (Graphics microarchitecture Gen8LP, according to [this document](https://01.org/sites/default/files/documentation/intel-gfx-bspec-osrc-chv-bsw-vol04-configurations.pdf))?
> 
> @maverick6664
> If you are using clang 8 and llvm 8, please try [Arch Linux's patches](https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/beignet), i.e. first revert the problematic git commits as specified in the PKGBUILD, then apply [this patch](https://git.archlinux.org/svntogit/community.git/plain/trunk/llvm8.patch?h=packages/beignet).

@86423355844265459587182778 - Sorry for belated reply.   I'll try this one.   It looks like it'll work.   Thanks!

86423355844265459587182778, 2019-05-29:
@PiotrRozenfeld 
Yes, I would be willing to help test and debug the NEO runtime if experimental support for Gen8LP is implemented.

closed 2019-05-29 by maverick6664

gilbahat, 2020-03-19:
Hi,

I am likewise interested in making this run on my Gen8 braswell machine. Beigenet is no longer being developed and is limited to opencl 1.2. This Gen8 machine was bought not long ago, they are still on the market and can be used for productive stuff like folding@home etc.  please reopen this issue.