intel/compute-runtime #169: [FEATURE REQUEST] ISA Assembler

TApplencourt, 2019-05-14:
Hi, 

As a performance engineer, I would like to make some low-level experiments (instructions reordering, measuring possible overheads of some non-naive registers regioning, etc...)

In order to do, I need to disassemble a binary and then re-assemble it. 
Disassemble a binary is possible (using the ahead of time compiler for example), but I was not able to assemble the modify ISA back to binary.

Have you envisaged to support this feature in future releases? 
Thanks!

jchodor, 2019-06-05:
Hi Thomas,
we're currently actively working on this. We're expecting it to be available via ocloc tool within next 2 weeks.
Best regards,
Jarek

jchodor, 2019-06-20:
After d3ed744c1be089d87bb7bdc8faeede0bc4031713 and 2e5ef30009dfda3cce413a0ec4e74c9f721b5cbd, there's still one change pending (will additionally require IGC devel-package that includes : https://github.com/intel/intel-graphics-compiler/commit/9c9ada682d5e9636cd38630359fb7387e0bc3c01).
Last change is currently being tested and should land sometime next week. 

The flow will look like this (skipping optional commnad-line parameters) :
1. create OCL binary either by using regular ocloc command line (to build from .cl files) or by using OCL API : clGetProgramInfo+CL_PROGRAM_BINARIES
2. **disassemble** using ocloc :  "ocloc.exe disasm -file some_ocl_binary.bin" - will create a dump directory with decoded OCL binary and disassembled (human-readable) kernels' ISA
3. hand-modify the disassembled ISA
4. **assemble** using ocloc : "ocloc.exe asm -out new_ocl_binary.bin"
5. Feed newly assembled binary to OCL API using **clCreateProgramWithBinary**+clBuildProgram

jchodor, 2019-06-28:
Available as of https://github.com/intel/compute-runtime/commit/2f42f332d8d73ab2a9431afced0eb64e339d2ec7 . 
We'll be cleaning-up/improving the command line help shortly. Please use instructions posted previously (8 days ago) for currently supported command line options.
Please remember that when building ocloc with this new feature, IGC devel package newer than intel/intel-graphics-compiler@9c9ada6 is required. Otherwise, disassembly feature will be disabled.
Comments/suggestions are welcome.

TApplencourt, 2019-06-28:
Thanks a lot. I will try to install everything on our CentOs and let you know.
Thanks again!

jchodor, 2019-07-05:
We've improved ocloc's command-line help. Please use ocloc not older than https://github.com/intel/compute-runtime/commit/369982995d549d2ee297e7fd76d30bd2d972f033 for best experience.

jchodor, 2019-08-02:
Closing feature request as complete. Please open new issue in case of problems.

closed 2019-08-02 by jchodor