intel/compute-runtime #172: Prerequisits for successful compilation

risingape, 2019-05-19:
Hello,
I try to build the compute runtime in my Opensuse Tumbleweed system. However a unit test fails:

Scanning dependencies of target run_skl_aub_tests
re-creating working directory for SKL 1x3x8 AUBs generation...
Running AUB generation for SKL 1x3x8 in /home/guido/Documents/intel_opencl/compute-runtime/build/bin/skl_aub
product family: skl (18)
set timeout to: 180
Iteration: 1
Iteration: 1. random_seed: 0

unknown file: Failure
C++ exception with description "std::exception" thrown in auxiliary test code (environments or event listeners).
make[2]: *** [unit_tests/aub_tests/CMakeFiles/run_skl_aub_tests.dir/build.make:67: run_skl_aub_tests] Error 1
make[1]: *** [CMakeFiles/Makefile2:4685: unit_tests/aub_tests/CMakeFiles/run_skl_aub_tests.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

Reading other open issues this seems to be known and it was hinted to use the "release" of the IGC sadly without specifying which one.

So far, I tried the master branch of the IGC as well as the 1.0.3 release.

Any advice how to build the compute runtime would be appreciated.

Thanks,

Guido

JacekDanecki, 2019-05-23:
The latest Neo release [19.19.12968](https://github.com/intel/compute-runtime/releases/tag/19.19.12968) was based on [IGC 1.0.4](https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.4). IGC was compiled with llvm/clang 7.
Can you provide steps you used to configure, build, install:
-  SPIRV-LLVM-Translator
- opencl-clang
- IGC
- gmmlib

Have you used llvm/clang 7 binaries from the system, or compiled as part of opencl-clang/IGC builds?

If you want to build only Neo library without unit tests you can use -DSKIP_UNIT_TESTS=1 flag for cmake. See [this command](https://github.com/intel/compute-runtime/blob/master/scripts/docker/Dockerfile-ubuntu-16.04-gcc-5#L9) as example.

JacekDanecki, 2019-06-28:
Closing because of lack of the feedback. Feel free to reopen it if you still observe this issue.

closed 2019-06-28 by JacekDanecki