intel/compute-runtime #173: ocloc tool doesn't support CL C++ kernels..

oscarbg, 2019-05-25:
Hi,
just wanted to finally test the new Neo SPIR-V 1.2 support..
seeing IWOCL keynote slides, in slide 20 "C++ for OpenCL in Clang project" 
it has a simple OpenCL C++ sample:
https://godbolt.org/z/nGvxAC
so I tried to generate SPIR-V from that but I'm having some issue presently:
https://github.com/KhronosGroup/SPIRV-LLVM-Translator/issues/201
so finally remembered that Neo ships with intel-ocloc tool so perhaps I will try with LLVM BC generated by ocloc &/ pass to llvm-spirv tool..
anyway I failed miserably also:

```
// Need to declare the prototype for get_global_id().
int get_global_id( int dim );

template<class T>
T add( T x, T y )
{
    return x + y;
}

__kernel void test( __global float* a, __global float* b)
{
    // Need to use unsigned; uint doesn't work.
    auto index = get_global_id(0);
    a[ index ] = add( b[ index ], b[ index + 1 ] );
}
```
using:
> ocloc -file clpp.cl -device skl
I get:
```
1:4:1: error: unknown type name 'template'
template<class T>
^
1:4:9: error: expected identifier or '('
template<class T>
        ^
1:13:5: error: OpenCL C version 1.2 does not support the 'auto' storage class specifier
    auto index = get_global_id(0);
    ^
1:13:10: warning: type specifier missing, defaults to 'int'
    auto index = get_global_id(0);
    ~~~~ ^
1:15:18: warning: implicit declaration of function 'add' is invalid in OpenCL
    a[ index ] = add( b[ index ], b[ index + 1 ] );
                 ^

Build failed with error code: -11
```
I see ocloc accepts things like -options -cl-std=CL2.0 but of course 
ocloc -file clpp.cl -device skl -options -cl-std=CL2.0
isn't enough..
sadly using -cl-std=cl++ which is which would be needed for OCL C++ kernels & is supported by clang (at least since 8.0) it fails also:

> ocloc -file clpp.cl -device skl -options -cl-std=cl++

```
Invalid format of -cl-std option, expected -cl-std=CLMAJOR.MINOR
Unrecognized build options: -cl-std=cl++
Build failed with error code: -11
```

so question is anyway to compile currently C++ OCL simple kernels like these with ocloc?
if not any recommended way seeing issues also with clang+llvm-spirv?
thanks..

jchodor, 2019-06-06:
Hi,
unfortunately ocloc is currently using under the hood opencl-clang (https://github.com/intel/opencl-clang) which does not support OCL C++ kernel language.
However, you can use a different clang - e.g.  https://github.com/KhronosGroup/SPIR/tree/spirv-1.1 to do the OCLC++ to spirV translation. Once you have spirV, you can feed that into ocl using clCreateProgramWithIl.
Best regards,
Jarek

jchodor, 2019-07-05:
Please reopen if further support is needed.

closed 2019-07-05 by jchodor