intel/compute-runtime #177: OpenCL 2.x on APL/BXT?

eero-t, 2019-06-05:
According to clinfo, Beignet provides OpenCL v2.0 also on BXT/APL, whereas NEO provides only OpenCL 1.2.   Will NEO have feature parity / support OpenCL 2.x also on Atoms at some point?

AdamCetnerowski, 2019-06-10:
For consistency across Atom platforms, NEO supports OCL1.2 at production quality. Experimentally, you can enable a higher version of OpenCL using the ForceOCLVersion debug flag. 

Could you share your interest in having OCL2.x support on Atom platforms? Are you looking for specific features?

eero-t, 2019-06-10:
> Are you looking for specific features?

I don't need OpenCL 2.x yet. I was just wondering why NEO didn't support it when Beignet did (triggered by noticing that ClearLinux just added NEO and removed Beignet).  Good to know that there's at least a flag with which one can test it.

MichalMrozek, 2019-07-08:
Closing this as problem was clarified.
In case of any further inquiries feel free to re-open.

closed 2019-07-08 by MichalMrozek