intel/compute-runtime #178: Abort was called at 136 line in file: ../neo/runtime/os_interface/linux/drm_buffer_object.cpp

KhazAkar, 2019-06-10:
I wanted to run test OpenCL render in Blender. Launch options for blender:
`CYCLES_OPENCL_SPLIT_KERNEL_TEST=1 blender`
It crashes every time I want to render using GPU.
`Read prefs: /home/khazakar/.config/blender/2.79/config/userpref.blend
found bundled python: /home/khazakar/bin/blender-2.79b/2.79/python
Device init success
Compiling OpenCL program split
Kernel compilation of split finished in 4.04s.

Compiling OpenCL program base
Kernel compilation of base finished in 1.16s.

Abort was called at 136 line in file:
../neo/runtime/os_interface/linux/drm_buffer_object.cpp`

JacekDanecki, 2019-06-10:
This abort is being called when DRM_IOCTL_I915_GEM_EXECBUFFER2 ioctl returns error from i915 driver. I've reproduced this issue using blender 2.79b-17 on Fedora 29. I've observed gpu hangs reported by kernel driver
```
[drm] GPU HANG: ecode 9:0:0x85ddfffb, in blender [7534], reason: hang on rcs0, action: reset
i915 0000:00:02.0: Resetting rcs0 for hang on rcs0
```
Let me prioritize this issue in our debug queue.

KhazAkar, 2019-06-16:
Sure, why not, prioritize if you want :)

JacekDanecki, 2020-01-29:
GPU hang is related to barrier function in opencl kernels. When I comment out some part of code containing barriers, blender renders without gpu hang.
Looking at source code of opencl kernels, I've found comment at https://developer.blender.org/diffusion/B/browse/master/intern/cycles/kernel/split/kernel_holdout_emission_blurring_pathtermination_ao.h$76
```
#ifdef __COMPUTE_DEVICE_GPU__
  /* If we are executing on a GPU device, we exit all threads that are not
   * required.
   *
   * If we are executing on a CPU device, then we need to keep all threads
   * active since we have barrier() calls later in the kernel. CPU devices,
   * expect all threads to execute barrier statement.
   */
  if (ray_index == QUEUE_EMPTY_SLOT) {
    return;
  }
#endif /* __COMPUTE_DEVICE_GPU__ */
```
This is not correct according to OpenCL spec https://www.khronos.org/registry/OpenCL/sdk/1.0/docs/man/xhtml/barrier.html. In Blender OpenCL kernels code barriers are used, and if some threads return earlier and don't execute barrier function, gpu hang will occur.

JacekDanecki, 2020-01-29:
As it's issue in Blender code I've filled new issue here: https://developer.blender.org/T73489.
Closing this report as it's not Neo problem.

closed 2020-01-29 by JacekDanecki

JacekDanecki, 2020-10-19:
Problem with OpenCL kernel was fixed in https://developer.blender.org/D9039, and I was able to render scene using Cycles Render Engine with Neo as rendering Device.
I've compiled Blender from commit:
5edba9b42f6 (HEAD -> master, origin/master, origin/HEAD) Fix T81688: BPY_thread_save crashes with Python 3.9