intel/compute-runtime #180: Selection of cl_cache default directory should be changed

alalek, 2019-06-16:
According to [FAQ](https://github.com/intel/compute-runtime/blame/19.23.13131/documentation/FAQ.md#L116) kernels caching implementation is not thread/process safe.

On Linux systems, default kernels cache path is `/tmp/cl_cache`.
Reusing such global directories like '/tmp' for binaries cache is not a good idea.
At least such common place can't guarantee process safety of caching mechanism, so this would lead to weird results.

Suggestion:
- disable current default (don't use `/tmp` contents implicitly).
- activate via `cl_cache_dir` environment variable only.

---

relates #87

HoppeMateusz, 2019-06-17:
Hi,
By default cache is turned off - cl_cache directory needs to be manually created to start using caching mechanism so it shouldn't affect app execution in default configuration.

According to FAQ default location of cl_cache is current working directory - it shouldn't be /tmp/cl_cache in general - only if app is started from this directory - do you see that driver uses /tmp/cl_cache directory always no matter what current directory or app location is?

MichalMrozek, 2019-07-08:
Closing this as problem was clarified.
In case of any further inquiries feel free to re-open.

closed 2019-07-08 by MichalMrozek