intel/compute-runtime #182: Support for MDAPI

TApplencourt, 2019-06-20:
Hi,

Does the MDAPI support is currently enabled on the NEO open-source Linux driver? 

We were not able to install MDAPI (https://github.com/intel/metrics-discovery), but not able to use it using (https://github.com/intel/opencl-intercept-layer/). 

Looking at https://github.com/intel/opencl-intercept-layer/issues/70#issuecomment-481544731, it seems that NEO doesn't yet enable MDAPI.  Is this still the case?

Regards,
Thomas

kevin-harms, 2019-06-20:
More specifically, the opencl-intercept-layer calls the interface clCreatePerfCountersCommandQueueINTEL() however in the driver, it is hard-coded to disable instrumentation.

https://github.com/intel/compute-runtime/blob/0758661c520e712c4b9be20de578ce44614481a3/runtime/os_interface/linux/hw_info_config.cpp#L148

Why is this hard-coded to false? Can we remove this line?

PiotrRozenfeld, 2019-08-12:
We are working on opensourcing the remaining functionality, which will allow us to provide this support. 
Feel free to follow up on https://github.com/intel/metrics-discovery for details.

JacekDanecki, 2020-04-10:
@TApplencourt Support for MDAPI was enabled in Neo in release [20.14.16441](https://github.com/intel/compute-runtime/releases/tag/20.14.16441).

bashbaug, 2020-04-10:
I've verified that MDAPI is working with the latest drivers after adding a very small workaround to the OpenCL Intercept Layer (https://github.com/intel/opencl-intercept-layer/pull/133).  Awesome!  Thanks!

JacekDanecki, 2020-04-14:
Closing, as it was confirmed to work.

closed 2020-04-14 by JacekDanecki