intel/compute-runtime #183: Build issue: 'opencl-c-base.h' file not found

jinz2014, 2019-06-21:
I am building the compute runtime, and got the error on CentOS7.  Thanks for your instructions!

[ 72%] Generating ../../../bin/built_ins/x64/gen9/aux_translation_Gen9lp.spv, ../../../bin/built_ins/x64/gen9/aux_translation_Gen9lp.bin, ../../../bin/built_ins/x64/gen9/aux_translation_Gen9lp.cpp, ../../../bin/built_ins/x64/gen9/aux_translation_Gen9lp.gen
In file included from <built-in>:1:
./opencl-c.h:12:10: fatal error: 'opencl-c-base.h' file not found
#include "opencl-c-base.h"
         ^~~~~~~~~~~~~~~~~

AlexeySachkov, 2019-06-22:
Hi @jinz2014,

Probably this is a bug in [intel/intel-graphics-compiler](https://github.com/intel/intel-graphics-compiler) rather than in intel/compute-runtime, but I'm not sure.

As I can see in [this commit](https://github.com/llvm/llvm-project/commit/af1c230e70aeb72ec9d6363f8d91e3c7db3ef9f2), `opencl-c.h` header file has been split into `opencl-c-base.h` and `opencl-c.h`.

However, as you can see [here](https://github.com/intel/intel-graphics-compiler/blob/a02d2da17c85295d4d6ff34a55feeb79ce03caa6/IGC/BiFModule/CMakeLists.txt#L155), only `opencl-c.h` is being installed.

You could try to update [IGC/BiFModule/CMakeLists.txt](https://github.com/intel/intel-graphics-compiler/blob/a02d2da17c85295d4d6ff34a55feeb79ce03caa6/IGC/BiFModule/CMakeLists.txt) to copy new header too to workaround the problem.

One more note: according to [IGC build instructions](https://github.com/intel/intel-graphics-compiler/blob/master/documentation/build_ubuntu.md) you should use `release_80` version of LLVM - I don't know if it is supported to build compiler/runtime using LLVM `master` branch

jinz2014, 2019-06-22:
Hi Alexey,

Thanks for explaining the issue !   I am not very familiar with CMake. Please let me know how I can copy the new header to  IGC/BiFModule/CMakeLists.txt.  I am using LLVM master branch and building the runtime on CentOS 7.  Is it right that I will need to build IGC again ? 

Thanks

lwesiers, 2019-06-24:
Hello there,

@jinz2014

> Is it right that I will need to build IGC again ?

No, you don't need to rebuild the IGC. You should manually install the missing header.
If you was building IGC with opencl-clang, then please follow this commands:

```
cd <workspace-of-igc>
sudo cp llvm_source/tools/clang/lib/Headers/opencl-c-base.h /usr/local/include
```

I made a fix for this and soon it will be available on https://github.com/intel/intel-graphics-compiler

jinz2014, 2019-06-24:
Thanks for the instruction!

The build is successful and it generates two .rpm files: 
intel-opencl-19.25.0-1.el7.x86_64.rpm
intel-opencl-ocloc-19.25.0-1.el7.x86_64.rpm

Does a user need to install both rpms ?

When running the command:
sudo rpm -i intel-opencl-19.25.0-1.el7.x86_64.rpm

I got the error: 
Failed dependencies:
        intel-gmmlib is needed by intel-opencl-19.25.0-1.el7.x86_64
        intel-igc-opencl is needed by intel-opencl-19.25.0-1.el7.x86_64

However, I did build the gmmlib as described in https://github.com/intel/gmmlib, and igc 

Something is missing, so how can I generate intel-gmmlib and intel-gmmlib after building gmmlibs and igc for the rpm to find the dependencies ? 

Thanks again

JacekDanecki, 2019-06-24:
You can see [this wiki page](https://github.com/JacekDanecki/compute-runtime/wiki/How-to-build-Neo-components) as reference.
Below command will create package
```
make package
```

lwesiers, 2019-06-24:
The fix for igc is available [ac50dc601f4035ed7daafa1cdf32713e4b0c4795](https://github.com/intel/intel-graphics-compiler/commit/ac50dc601f4035ed7daafa1cdf32713e4b0c4795).

jinz2014, 2019-06-24:
Thanks for the wiki page and fix for igc!

Typing the command 'make package' generates the RPM files. However, there are conflicts when installing intel-gmmlib-9.0.0-1.el7.x86_64.rpm:

$ sudo rpm -i  intel-gmmlib-9.0.0-1.el7.x86_64.rpm
     file /usr/local from install of intel-gmmlib-9.0.0-1.el7.x86_64.rpm conflicts with file from package 
     filesystem-3.2.25.el7.x86_64.rpm
     file /usr/local/lib64 from install of intel-gmmlib-9.0.0-1.el7.x86_64.rpm conflicts with file from package 
     filesystem-3.2.25.el7.x86_64.rpm

I am not sure if you encountered such issue when building on CentOS7. 

Thanks

JacekDanecki, 2019-06-25:
We've observed [similar issue](https://github.com/intel/compute-runtime/issues/45) last year in Neo, but it was fixed with [this commit](https://github.com/intel/compute-runtime/commit/b1d020485a7d1f7105a8c9db84c0fc12ff0a0e68).
I cannot reproduce this issue on my side, but I've prepared [this change](https://github.com/JacekDanecki/gmmlib/commit/bac6558846510126fef7439e5ec09de44ba23fa7) on my fork. Can you confirm whether it fixes problem?

jinz2014, 2019-06-25:
Thanks for your chang on your fork. It fixes the problem of installing intel-gmmlib. Then, there are other conflicts when installing intel-gmmlib-devel and intel-opencl-ocloc.  I am not clear if there is an order regarding which RPMs need to be installed first.  Thanks for your fixes. 

$ sudo rpm -i intel-gmmlib-devel-9.0.0-1.el7.x86_64.rpm
        file /usr/local/include from install of intel-gmmlib-devel-9.0.0-1.x86_64 conflicts with file from package filesystem-3.2-25.el7.x86_64
        file /usr/local/include from install of intel-gmmlib-devel-9.0.0-1.x86_64 conflicts with file from package intel-igc-opencl-devel-1.0.9-1.x86_64
        file /usr/local/lib64/pkgconfig from install of intel-gmmlib-devel-9.0.0-1.x86_64 conflicts with file from package intel-igc-opencl-devel-1.0.9-1.x86_64
======================================================================
$ sudo rpm -i intel-opencl-ocloc-19.25.0-1.el7.x86_64.rpm
        file /usr/local/bin from install of intel-ocloc-19.25.0-1.el7.x86_64 conflicts with file from package filesystem-3.2-25.el7.x86_64
        file /usr/local/bin from install of intel-ocloc-19.25.0-1.el7.x86_64 conflicts with file from package intel-igc-opencl-devel-1.0.9-1.x86_64
        file /usr/local/bin from install of intel-ocloc-19.25.0-1.el7.x86_64 conflicts with file from package intel-igc-media-1.0.9-1.x86_64

=====================================================================
I think I can ignore the following error when installing  intel-igc-opencl because I can use export LD_LIBRARY_PATH=/path/to/libopencl-clang.so.9svn(). Is that right ? 

error: Failed dependencies:
        libopencl-clang.so.9svn()(64bit) is needed by intel-igc-opencl-1.0.9-1.x86_64

Thanks again

JacekDanecki, 2019-06-25:
I've updated change for gmmlib [here](https://github.com/JacekDanecki/gmmlib/commit/37a79cfc16197ce42c05f7aa053c2339326ba824)

For intel-opencl-ocloc change is [here](https://github.com/JacekDanecki/compute-runtime/commit/b5c0194a13eaf2a18e5b8c4b1f918efc66eb81e1)

Can you verify them and confirm it works for you?

I suppose libopencl-clang.so.9 will be installed into /usr/local/lib64, so you can set:
```
export LD_LIBRARY_PATH=/usr/local/lib64
```
or add file igc.conf containing
```
/usr/local/lib64
```
to directory /etc/ld.so.conf.d and run:
```
sudo ldconfig
```
and install igc using:
```
rpm -i --nodeps intel-igc-opencl-1.0.9-1.el7.x86_64.rpm
```

jinz2014, 2019-06-25:
Yes, they fixes the conflicts.  Currently, I am not able to see the GPU driver information using clinfo.

I am not sure how users can debug the issue. Should I uninstall some OpenCL runtime installed using yum ? Do I miss some components ?  Should I reboot the machine ?   Thanks for your instruction!

These are the package installed:
$ yum search intel-

intel-cmt-cat.x86_64 : Provides command line interface to CMT, MBM, CAT and CDP
                     : technologies
intel-gmmlib.x86_64 : Intel(R) Graphics Memory Management Library Package
intel-gmmlib-devel.x86_64 : Intel(R) Graphics Memory Management Library Package
intel-gpu-tools.x86_64 : Debugging tools for Intel graphics chips
intel-igc-core.x86_64 : IGC built using CMake
intel-igc-media.x86_64 : IGC built using CMake
intel-igc-opencl.x86_64 : IGC built using CMake
intel-igc-opencl-devel.x86_64 : IGC built using CMake
intel-ocloc.x86_64 : Intel OpenCL GPU driver
intel-opencl.x86_64 : Intel OpenCL GPU driver
libva-intel-hybrid-driver.x86_64 : VA driver for Intel G45 & HD Graphics family
nasm.x86_64 : A portable x86 assembler which uses Intel-like syntax
xorg-x11-drv-intel-devel.i686 : Xorg X11 Intel video driver development package
xorg-x11-drv-intel-devel.x86_64 : Xorg X11 Intel video driver development
                                : package

However, I can see the CPU driver info below. 

$ clinfo
Number of platforms                               1
  Platform Name                                   Intel(R) CPU Runtime for OpenCL(TM) Applications
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 2.1 LINUX
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_depth_images cl_khr_3d_image_writes cl_intel_exec_by_local_thread cl_khr_spir cl_khr_fp64 cl_khr_image2d_from_buffer cl_intel_vec_len_hint
  Platform Host timer resolution                  1ns
  Platform Extensions function suffix             INTEL

  Platform Name                                   Intel(R) CPU Runtime for OpenCL(TM) Applications
Number of devices                                 1
  Device Name                                     Intel(R) Xeon(R) CPU E3-1284L v4 @ 2.90GHz
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 (Build 0)
  Driver Version                                  18.1.0.0920
  Device OpenCL C Version                         OpenCL C 2.0
  Device Type                                     CPU
  Device Available                                Yes

JacekDanecki, 2019-06-26:
I saw you'd issue with clinfo. Have you added your user to video group, as described [here](https://github.com/intel/compute-runtime/blob/master/documentation/Neo_in_distributions.md)?
You can use strace command to check basic issues with configuration:
```
strace -s 255 -f -o log.txt clinfo
```

jinz2014, 2019-06-26:
Yes, I previously install Intel system studio 19. After uninstalling it, the number of platforms is zero using clinfo.
  
However, I don't see any directory called "dri" under /dev. Do I need to be in video group before I can see them ?   

Thanks

-----------------------------------------------------

$ lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                8
On-line CPU(s) list:   0-7
Thread(s) per core:    2
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 71
Model name:            Intel(R) Xeon(R) CPU E3-1284L v4 @ 2.90GHz

$ ls /dev/
autofs           crash       input         net                 random    stderr  tty13  tty22  tty31  tty40  tty5   tty59  ttyS1    vcs6         zero
block            disk        ipmi0         network_latency     raw       stdin   tty14  tty23  tty32  tty41  tty50  tty6   ttyS2    vcsa
bsg              fd          kmsg          network_throughput  rtc       stdout  tty15  tty24  tty33  tty42  tty51  tty60  ttyS3    vcsa1
btrfs-control    full        kvm           null                rtc0      tgt     tty16  tty25  tty34  tty43  tty52  tty61  uhid     vcsa6
bus              fuse        log           nvram               sda       tty     tty17  tty26  tty35  tty44  tty53  tty62  uinput   vfio
char             hpet        loop-control  oldmem              sda1      tty0    tty18  tty27  tty36  tty45  tty54  tty63  urandom  vga_arbiter
console          hugepages   mapper        port                sg0       tty1    tty19  tty28  tty37  tty46  tty55  tty7   usbmon0  vhci
core             hwrng       mcelog        ppp                 shm       tty10   tty2   tty29  tty38  tty47  tty56  tty8   usbmon1  vhost-net
cpu              infiniband  mem           ptmx                snapshot  tty11   tty20  tty3   tty39  tty48  tty57  tty9   vcs      watchdog
cpu_dma_latency  initctl     mqueue        pts                 snd       tty12   tty21  tty30  tty4   tty49  tty58  ttyS0  vcs1     watchdog0

closed 2019-06-27 by jinz2014

mprogram, 2019-07-30:
I have just hit this bug, when followed the instructions from https://github.com/intel/intel-graphics-compiler/issues/54 against system installed dependencies, which I managed to work around with two very helpful suggestions from the above. Thank you, Łukasz, aka @lwesiers: 
> No, you don't need to rebuild the IGC. You should manually install the missing header.
> If you was building IGC with opencl-clang, then please follow this commands:
>
> ```
> cd <workspace-of-igc>
> sudo cp [...]

I believe somebody may have a second look at why it could not have found `opencl-c-base.h` under the referenced scenario even after IGC was patched.