intel/compute-runtime #185: GPU device of i9-9900k can't be identified by clinfo on a ubuntu 16.04

juliusshufan, 2019-06-27:
Hi,

I am using i9-9900k on a ubuntu 16.04.

After install the latest release of opencl runtime, and add the corresponding user to video group, the GEN9 GPU device still can't be detected, only CPU device can be detected.
May I know the 9900k is supported by the latest opencl runtime?

Thanks.

AdamCetnerowski, 2019-06-28:
Yes, this GPU is supported. Which kernel version are you using? For CFL platform i915.alpha_support=1 kernel parameter is required under Ubuntu 16.04.4 with default kernel 4.13. 

If this doesn't help, please run
strace -s 255 -f -o log.txt clinfo
and attach the log here. 
 
To check support for any device, you can follow these steps:
1.	Got to [Ark]( https://ark.intel.com) and find your Device ID. In this case it’s 0x3E98. 
2.	Find the corresponding device ID label in [GMM]( https://github.com/intel/gmmlib/blob/master/Source/inc/common/igfxfmid.h). In this case it’s ICFL_GT2_S8_S2_DT_DEVICE_F0_ID.
3.	Check if this device ID label is enumerated in the [supported device list] (https://github.com/intel/compute-runtime/blob/master/runtime/dll/linux/devices/devices_base.inl)

juliusshufan, 2019-07-01:
@AdamCetnerowski Thanks, I am working with @airMeng on your steps.

airMeng, 2019-07-01:
[log.txt](https://github.com/intel/compute-runtime/files/3344395/log.txt)
Here is the log. Unfortunately I still got non-gou device

juliusshufan, 2019-07-01:
@AdamCetnerowski Following the step, but GPU device still not enabled.
Could you kindly check the log from myside, and you may ignore the one from @airMeng as that one is not valid. Thanks.
[log_clinfo.txt](https://github.com/intel/compute-runtime/files/3345880/log_clinfo.txt)

alalek, 2019-07-01:
Looks like there are missing entries in `/dev/dri`.

Please check this:
```
lspci | grep VGA
```

if iGPU is missing ("VGA compatible controller: Intel Corporation ...") then probably it was disabled by BIOS (you need to change BIOS settings and enable Integrated GPU).

If it still doesn't work, please provide this information:
```
dmesg | grep -E 'drm|i915'
```

juliusshufan, 2019-07-01:
@alalek Thanks for the prompt response.

**00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 630 (Desktop 9 Series**)

test@mlt-i9-gen9:~$ dmesg | grep -E 'drm|i915'
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-29-generic root=UUID=7cbc6a1a-bd7b-4434-adee-67dbd7133aad ro i915.alpha_support=1 quiet splash vt.handoff=7
[    0.000000] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-29-generic root=UUID=7cbc6a1a-bd7b-4434-adee-67dbd7133aad ro i915.alpha_support=1 quiet splash vt.handoff=7
[    2.526713] snd_hda_intel 0000:00:1f.3: failed to add i915 component master (-19)
[    2.555096] snd_hda_codec_hdmi hdaudioC0D2: No i915 binding for Intel HDMI/DP codec
[    2.555194] snd_hda_codec_hdmi hdaudioC0D2: No i915 binding for Intel HDMI/DP codec

alalek, 2019-07-01:
Thank you for update!

dmesg output seems very short.

<details>

<summary>dmesg for i5-6600 with Ubuntu 16.04 +  kernel 4.15</summary>

```
# dmesg | grep -E 'drm|i915'
[    1.688084] [drm] Memory usable by graphics device = 4096M
[    1.688085] fb: switching to inteldrmfb from EFI VGA
[    1.691910] [drm] Replacing VGA console driver
[    1.698002] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    1.698006] [drm] Driver supports precise vblank timestamp query.
[    1.700105] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    1.700492] [drm] Finished loading DMC firmware i915/skl_dmc_ver1_26.bin (v1.26)
[    1.702485] [drm] Initialized i915 1.6.0 20171023 for 0000:00:02.0 on minor 0
[    1.839613] fbcon: inteldrmfb (fb0) is primary device
[    1.884121] i915 0000:00:02.0: fb0: inteldrmfb frame buffer device
[    2.812657] [drm] RC6 on
# uname -a
Linux nnlicv207 4.15.0-54-generic #58~16.04.1-Ubuntu SMP Mon Jun 24 13:21:41 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```
</details>

Could you grab this information?

```
# lspci -s 00:02.0 -nnn -vvv
lsmod | grep i915
```
(run first command under 'root' user)

juliusshufan, 2019-07-01:
Hi Alex, @alalek 

Please kindly see below, hopes which helps. 

**sudo lspci -s 00:02.0 -nnn -vvv**
00:02.0 VGA compatible controller [0300]: Intel Corporation UHD Graphics 630 (Desktop 9 Series) [8086:3e98] (prog-if 00 [VGA controller])
        DeviceName: Onboard - Video
        Subsystem: Gigabyte Technology Co., Ltd Device [1458:d000]
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 11
        Region 0: Memory at 50000000 (64-bit, non-prefetchable) [size=16M]
        Region 2: Memory at 40000000 (64-bit, prefetchable) [size=256M]
        Region 4: I/O ports at 3000 [size=64]
        [virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express (v2) Root Complex Integrated Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0
                        ExtTag- RBE+
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis-, LTR-, OBFF Not Supported
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
        Capabilities: [ac] MSI: Enable- Count=1/1 Maskable- 64bit-
                Address: 00000000  Data: 0000
        Capabilities: [d0] Power Management version 2
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [100 v1] #1b
        Capabilities: [200 v1] Address Translation Service (ATS)
                ATSCap: Invalidate Queue Depth: 00
                ATSCtl: Enable-, Smallest Translation Unit: 00
        Capabilities: [300 v1] #13

**lsmod | grep i915**
i915                 1630208  0
drm_kms_helper        172032  1 i915
drm                   401408  2 i915,drm_kms_helper
i2c_algo_bit           16384  1 i915
video                  45056  1 i915

alalek, 2019-07-01:
Looks like, i915 driver is not bound to the PCI device.
Usually this means that used version of Linux kernel doesn't know anything about the device (or binding is blocked via configuration/kernel boot parameters - boot parameters is fine, blacklisting is not enabled).

<details>

<summary>Expected lines:</summary>

```
    Capabilities: [300 v1] #13
    Kernel driver in use: i915
    Kernel modules: i915
```

</details>

---

> 4.15.0-29-generic

This kernel version is not fresh: https://ubuntu.pkgs.org/16.04/ubuntu-updates-main-amd64/linux-image-4.15.0-29-generic_4.15.0-29.31~16.04.1_amd64.deb.html
Changelog date is about 2018-07-17.

Compare these sources (search for '0x3e98' - your pci device id)
- Ubuntu's [4.15.0-29](https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial/tree/include/drm/i915_pciids.h?id=Ubuntu-hwe-4.15.0-29.31_16.04.1) (missing `3e98`)
- fresh Ubuntu's [4.15.0-54](https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial/tree/include/drm/i915_pciids.h?id=Ubuntu-hwe-4.15.0-54.58_16.04.1) (available `3e98`)

So, try to upgrade kernel (it is useful to have backup if something goes wrong)
```
apt-get update
apt-get upgrade      # upgrades all packages on system (it is useful to have backup of root fs)
# or apt-get install --upgrade \
#        linux-image-generic-hwe-16.04-edge  \
#        ...other_kernel_packages_like_headers_extra_modules_etc...
# ... reboot
# ... check again starting from 'lspci -s 00:02.0 -nnn -vvv'
```

juliusshufan, 2019-07-02:
@alalek  Your solution work indeed, appreciate for your prompt help. I close this issue accordingly.

closed 2019-07-02 by juliusshufan

laygond, 2019-09-09:
@juliusshufan  hello, I see you are running ubuntu16.04 with that CPU. What motherboard are you using for that configuration?  Thanks in advance :)