intel/compute-runtime #186: [BUG] Shared Local Memory Overflow

HE-Wenjian, 2019-07-02:
When the code allocates more local memory than the hardware can offer, no exception is raised. Instead, the local memory wraps around on itself, silently.

I guess programmers would expect at least a warning.

PoC:
```c
// My machine is i7-6700 (skylake) with Intel HD 530 (Gen 9.5), 
// whose SLM is 64 KB per subslice
#define Subslice_SLM_Size (64*1024)  // 64 KB

__kernel void check_slm_wrap_around(){
    const size_t N=Subslice_SLM_Size/sizeof(size_t);
    __local size_t slm[2*N];   // Double the size of the SLM
    for(size_t i=0;i<2*N;++i)
        slm[i]=i;
   
    if(slm[0]==slm[N]){
        printf("They should not be equal! slm[0]=slm[N]=%lu, N=%lu \n", slm[0], N);
    }
}

```
The output of the PoC above:
```
They should not be equal! slm[0]=slm[N]=8192, N=8192
```
Testbed: Ubuntu 16.04 with `compute-runtime` release 19.16.12873

MichalMrozek, 2019-07-04:
Thanks for report, this has been added to our queue with ID NEO-3411.
We will most likely fail kernel creation or build program here.

PiotrZdunowski, 2019-07-23:
This issue should be resolved by https://github.com/intel/compute-runtime/commit/1460713d69066721575c400ae7068bcc4d2c39fb#diff-e947a53eaeb1e0f981fdcf820b5bc306

closed 2019-07-23 by HE-Wenjian