intel/compute-runtime #188: LLVM ERROR: inconsistency in registered CommandLine options

josemonsalve2, 2019-07-02:
Hi, 

I'm trying to compile this library in my system. My system description is: 
```
CPU:          Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz
Kernel:       5.0.0-17-generic
OS:            Ubuntu 19.04
LLVM:        8.0
```
My compilation fails with the following error: 
```
[ 19%] Generating ../../../bin/built_ins/x64/gen9/aux_translation_Gen9core.spv, ../../../bin/built_ins/x64/gen9/aux_translation_Gen9core.bin, ../../../bin/built_ins/x64/gen9/aux_translation_Gen9core.cpp, ../../../bin/built_ins/x64/gen9/aux_translation_Gen9core.gen
: CommandLine Error: Option 'help-list' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
```

With verbose, this is the command that fails.
```
cd ${workdir}/compute-runtime/runtime/built_ins/kernels && LD_LIBRARY_PATH=${workdir}/compute-runtime/build/bin ${workdir}/compute-runtime/build/bin/ocloc -q -file aux_translation.igdrcl_built_in -device skl -64 -out_dir ${workdir}/compute-runtime/build/bin/built_ins/x64/gen9 -cpp_file -options -cl-kernel-arg-info
: CommandLine Error: Option 'help-list' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
```
I have compiled all the dependencies without errors, following these steps: 

1. intel graphics compiler:

```
$ sudo apt-get install flex bison libz-dev cmake
$ git clone -b release_80 https://github.com/llvm-mirror/llvm llvm_source
$ git clone -b release_80 https://github.com/llvm-mirror/clang llvm_source/tools/clang
$ git clone -b ocl-open-80 https://github.com/intel/opencl-clang llvm_source/projects/opencl-clang
$ git clone -b llvm_release_80 https://github.com/KhronosGroup/SPIRV-LLVM-Translator llvm_source/projects/llvm-spirv
$ git clone https://github.com/intel/llvm-patches llvm_patches
$ git clone https://github.com/intel/intel-graphics-compiler igc
# Using master branch, no tag
$ mkdir build
$ cd build
$ make
$ sudo make install
```

2. GmmLib:

```
$ git clone https://github.com/intel/gmmlib.git
$ cd gmmlib/
$ mkdir build
$ cd build/
$ cmake ../ -DCMAKE_BUILD_TYPE=RELEASE
$ make
$ sudo make install
```

3. Libdrm:

```
$ git clone https://anongit.freedesktop.org/git/mesa/drm.git
$ cd drm
$ ./autogen.sh
$ make
$ sudo make install
```

4. Libva

```
$ git clone https://github.com/intel/libva.git
$ cd libva
$ ./autogen.sh
$ make
$ sudo make install
```

What do you think I am missing? 

Thanks for your help. 

Jose

josemonsalve2, 2019-07-02:
I changed compilers to llvm-7 and followed these instructions:
https://github.com/JacekDanecki/compute-runtime/wiki/How-to-build-Neo-components-(system-llvm-clang-7)

I was successful in compiling it. I have not tested it already.

lwesiers, 2019-07-09:
Hello there,

I was not able to reproduce the issue using the latest source code from today. Could you share full-logs (cmake, make) for the setup of llvm8/clang8 for components igc and neo?

josemonsalve2, 2019-07-10:
I could not reproduce it anymore myself. I think it had to do with different versions of LLVM in the system that were conflicting. Using the .deb packages I was able to run NEO in my system, but I was not able to compile the whole Software stack from scratch. It was not clear what versions of what I was supposed to be using, and I would always end up with a different error. Unfortunately I did not document this well. I will close this issue for now. But I really appreciate that you responded to my issue.

closed 2019-07-10 by josemonsalve2