intel/compute-runtime #193: Please provide RPMs

tomdeakin, 2019-07-31:
The [CentOS install instructions](https://github.com/intel/compute-runtime/blob/master/documentation/BUILD_Centos.md)  were quite hard to find. Also they require dependancies which also don't have good CentOS instructions. Are you able to provide RPMs as well as DEBs in the release?

JacekDanecki, 2019-07-31:
I'm preparing spec files to build Neo, gmmlib, and IGC components on [copr](https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl) for CentOS, similar to Fedora builds I'd enabled last week. I'll update these builds weekly after release on github. I'll nofity you in this issue, when first build is ready for tests.

akien-mga, 2019-08-01:
BTW @JacekDanecki, I've rebuilt your Fedora Rawhide SRPMs on Mageia 7, and they seem to work well. Mageia 7 and Cauldron are available as targets in Copr so you might want to enable them there.

I plan to import the packages in Mageia's official repositories eventually.

To do so I'm cleaning up the specs to better match current packaging guidelines, if you're interested I can provide patches, e.g. https://paste.fedoraproject.org/paste/6ObiK1svmuRlTebxB0ZAYg (the `%cmake` macro I used automatically creates a `build` folder on Mageia, but I'm not sure it does on Fedora, so it might still need some tweaks for good cross-distro compatibility).

I would also suggest to rename the spec files to match the `Name` tag, and drop the version string from their filename - it's included in the spec and SRPM already.

JacekDanecki, 2019-08-01:
@akien-mga This sounds great. I've seen on copr I could enable more distributions during builds, but before I enable them I wanted to test builds locally. If you provide me patches for Mageia and Cauldron, it will speed up these distributions enabling on copr. When I'm ready with Centos enabling, I'll include spec files for Fedora and Centos in Neo repository, so it'll be easier to work on them, especially if we want to share them between different distributions.
If you add Neo packages to Mageia's official repositories, please let me know, or create pull request with changes in [documentation/Neo_in_distribution](https://github.com/intel/compute-runtime/blob/master/documentation/Neo_in_distributions.md).

JacekDanecki, 2019-08-01:
@tomdeakin First packages for Centos 7 are available in copr. To use them run below commands:
```
yum install yum-plugin-copr
yum copr enable jdanecki/intel-opencl
yum install intel-opencl
```
You can check them with clinfo:
```
yum install epel-release
yum install clinfo
clinfo
```

tomdeakin, 2019-08-08:
Thanks @JacekDanecki. I managed to install this, but I see no platforms listed with `clinfo`. It might be worth catching up with Larry who's also pointing us in the right direction with getting this working.

JacekDanecki, 2019-08-08:
@tomdeakin Can you provide file log.txt created with command:
```
strace -o log.txt -f -s 255 clinfo
```

tomdeakin, 2019-08-08:
[log.txt](https://github.com/intel/compute-runtime/files/3482729/log.txt)

JacekDanecki, 2019-08-08:
17374 open("/dev/dri/renderD128", O_RDWR) = -1 EACCES (Permission denied)
Here is issue. Have you added your user to video group, as was mentioned [here](https://github.com/intel/compute-runtime/blob/master/documentation/Neo_in_distributions.md#additional-configuration)

JacekDanecki, 2019-08-09:
@akien-mga Spec files have been added to the repo [here](https://github.com/intel/compute-runtime/tree/master/scripts/specs)

akien-mga, 2019-08-09:
@JacekDanecki Thanks, I'll PR some enhancement proposals sometime next week.

tomdeakin, 2019-08-16:
@JacekDanecki Adding the users to the `video` group makes the device visible. Is there a way to tell the installer which group to enable the driver for?

JacekDanecki, 2019-09-02:
This is administrator responsibility to configure users and groups on specific machine.

tomdeakin, 2019-09-08:
I only ask because many of the AMD graphics drivers can be installed in "compute only" mode and allow you to specify the group of user to make the driver available for. It does default to `video`, but you can pick another group on install if you want.

akien-mga, 2019-10-16:
Awesome, thanks! I still intend to contribute some improvements, but time is lacking... Hopefully I'll do it in coming weeks :)

JacekDanecki, 2019-10-17:
@akien-mga See this repo: https://github.com/JacekDanecki/neo-specs