intel/compute-runtime #195: Custom Blender build with IGC "support" segfaults when compiling OpenCL kernels

akien-mga, 2019-08-01:
Moving discussion about this specific segfault from #194 to its own issue.

**System Information**
Operating system: Linux-5.1.20-desktop-2.mga7-x86_64-with-mageia-7-Official 64 Bits
Graphics card 1 (integrated): Mesa DRI Intel(R) HD Graphics 630 (Kaby Lake GT2) Intel Open Source Technology Center 4.5 (Core Profile) Mesa 19.1.3
Graphics card 2 (discrete): AMD VEGAM (DRM 3.30.0, 5.1.20-desktop-2.mga7, LLVM 8.0.0) X.Org 4.5 (Core Profile) Mesa 19.1.3

**Intel Compute Runtime Version**

```
-rw-r--r-- 1 akien akien    86852 Jul 31 15:05 intel-gmmlib-19.2.3-1.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien   108884 Jul 31 15:05 intel-gmmlib-devel-19.2.3-1.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien  5754780 Aug  1 10:56 intel-igc-core-1.0.10-2.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien 60140276 Aug  1 10:58 intel-igc-core-debuginfo-1.0.10-2.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien    35052 Aug  1 10:57 intel-igc-debuginfo-1.0.10-2.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien  8262740 Aug  1 10:57 intel-igc-debugsource-1.0.10-2.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien    64508 Aug  1 10:56 intel-igc-opencl-1.0.10-2.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien   552148 Aug  1 10:58 intel-igc-opencl-debuginfo-1.0.10-2.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien    72332 Aug  1 10:56 intel-igc-opencl-devel-1.0.10-2.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien   698812 Aug  1 11:05 intel-opencl-19.28.13502-1.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien   400436 Jul 31 15:03 intel-opencl-clang-8.0.72-1.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien     9736 Jul 31 15:03 intel-opencl-clang-devel-8.0.72-1.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien  8778080 Aug  1 11:05 intel-opencl-debuginfo-19.28.13502-1.mga7.x86_64.rpm
-rw-r--r-- 1 akien akien   679452 Aug  1 11:05 intel-opencl-debugsource-19.28.13502-1.mga7.x86_64.rpm
```

Custom built on Mageia 7 from @JacekDanecki's Fedora Rawhide SRPMs at https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl/ (builds from 2019-07-23)

---

As shown in #194, we run into blocking issues fast trying to use IGC with Blender 2.80's official builds (`blender-2.80-linux-glibc217-x86_64`), as it segfaults or hangs when accessing Cycles OpenCL settings.

Yet by building Blender from source (default settings, debug build), I managed to past this first issue to dig into actual compatibility issues when using IGC as OpenCL platform for Cycles renders.

**Blender version:** 036312ecff459f7600361d5aba9a0dba896849a1

A patch was needed in my case for IGC to be picked up as a valid OpenCL platform. It basically bypasses Blender's current whitelist of support GPUs:
```diff
diff --git a/intern/cycles/device/opencl/opencl.h b/intern/cycles/device/opencl/opencl.h
index 82b961b8de7..04f07f30365 100644
--- a/intern/cycles/device/opencl/opencl.h
+++ b/intern/cycles/device/opencl/opencl.h
@@ -90,7 +90,7 @@ class OpenCLInfo {
   static bool device_version_check(cl_device_id device, string *error = NULL);
   static string get_hardware_id(const string &platform_name, cl_device_id device_id);
   static void get_usable_devices(vector<OpenCLPlatformDevice> *usable_devices,
-                                 bool force_all = false);
+                                 bool force_all = true);
 
   /* ** Some handy shortcuts to low level cl*GetInfo() functions. ** */
 
```

I can then enable NEO in Edit > Preferences > System:
[image]

But when I try to use it for an actual Cycles render of the default cube with default settings:
[image]

I get a segfault during OpenCL compilation:
```
blender: /home/akien/Projects/mageia/Sandbox/_rpm/BUILD/spirv-llvm/lib/SPIRV/SPIRVWriter.cpp:1174: SPIRV::SPIRVValue* SPIRV::LLVMToSPIRV::transIntrinsicInst(llvm::IntrinsicInst*, SPIRV::SPIRVBasicBlock*): Assertion `cast<MemCpyInst>(II)->getSourceAlignment() == cast<MemCpyInst>(II)->getDestAlignment() && "Alignment mismatch!"' failed.
sh: line 1: 22318 Aborted                 (core dumped) "/home/akien/tmp/blender/build/bin/blender" "--background" "--factory-startup" "--python-expr" "import _cycles; _cycles.opencl_compile(r'1', r'Intel(R) Gen9 HD Graphics NEO', r'Intel(R) OpenCL HD Graphics', r'-cl-no-signed-zeros -cl-mad-enable -D__KERNEL_CL_KHR_FP16__ -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ -D__NO_SUBSURFACE__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__', r'kernel_background.cl', r'/home/akien/.cache/cycles/kernels/cycles_kernel_background_3A063A327B9939998F9E7BFA0349B693_13AC40ED2FE033D81EB1496FB3C61F4D.clbin')" > /dev/null
Cycles: compiling OpenCL program background...
blender: /home/akien/Projects/mageia/Sandbox/_rpm/BUILD/spirv-llvm/lib/SPIRV/SPIRVWriter.cpp:1174: SPIRV::SPIRVValue* SPIRV::LLVMToSPIRV::transIntrinsicInst(llvm::IntrinsicInst*, SPIRV::SPIRVBasicBlock*): Assertion `cast<MemCpyInst>(II)->getSourceAlignment() == cast<MemCpyInst>(II)->getDestAlignment() && "Alignment mismatch!"' failed.
Aborted (core dumped)
```

More details in next comment.

akien-mga, 2019-08-01:
As discussed in #194, I rebuilt LLVM and Clang with `-DCMAKE_BUILD_TYPE=Debug` and rebuilt my IGC packages in Debug mode too.

Here's the backtrace I get when it crashes, preceded by Blender's own debug information from running with `./blender --debug-cycles`:

```
[New Thread 0x7fffae9a4700 (LWP 2970)]
I0801 20:13:36.382897  2859 opencl_util.cpp:899] Enumerating devices for platform Clover.
I0801 20:13:36.382930  2859 opencl_util.cpp:937] Adding new device AMD VEGAM (DRM 3.30.0, 5.1.20-desktop-2.mga7, LLVM 8.0.0).
I0801 20:13:36.382937  2859 opencl_util.cpp:899] Enumerating devices for platform Intel(R) OpenCL HD Graphics.
I0801 20:13:36.382943  2859 opencl_util.cpp:937] Adding new device Intel(R) Gen9 HD Graphics NEO.
[New Thread 0x7fffaf1a5700 (LWP 4291)]
I0801 20:13:47.954680  4291 util_task.cpp:329] Creating pool of 8 threads.
I0801 20:13:47.954725  4291 util_task.cpp:241] Detected 8 processors in active group.
I0801 20:13:47.954730  4291 util_task.cpp:251] Not setting thread group affinity.
[New Thread 0x7fffaf9a6700 (LWP 4292)]
[New Thread 0x7fffb01a7700 (LWP 4293)]
[New Thread 0x7fffa39ff700 (LWP 4294)]
[New Thread 0x7fffa31fe700 (LWP 4295)]
[New Thread 0x7fffa29fd700 (LWP 4296)]
[New Thread 0x7fffa21fc700 (LWP 4297)]
[New Thread 0x7fffa19fb700 (LWP 4298)]
[New Thread 0x7fffa11fa700 (LWP 4299)]
[New Thread 0x7fffa09f9700 (LWP 4300)]
I0801 20:13:47.958765  4291 opencl_split.cpp:632] Creating new Cycles device for OpenCL platform Intel(R) OpenCL HD Graphics, device Intel(R) Gen9 HD Graphics NEO.
I0801 20:13:47.959003  4291 util_task.cpp:347] De-initializing thread pool of task scheduler.
[Thread 0x7fffa09f9700 (LWP 4300) exited]
[Thread 0x7fffa11fa700 (LWP 4299) exited]
[Thread 0x7fffa19fb700 (LWP 4298) exited]
[Thread 0x7fffa21fc700 (LWP 4297) exited]
[Thread 0x7fffa29fd700 (LWP 4296) exited]
[Thread 0x7fffa31fe700 (LWP 4295) exited]
[Thread 0x7fffa39ff700 (LWP 4294) exited]
[Thread 0x7fffb01a7700 (LWP 4293) exited]
I0801 20:13:47.959223  4291 util_task.cpp:329] Creating pool of 8 threads.
[Thread 0x7fffaf9a6700 (LWP 4292) exited]
I0801 20:13:47.959240  4291 util_task.cpp:241] Detected 8 processors in active group.
I0801 20:13:47.959245  4291 util_task.cpp:251] Not setting thread group affinity.
[New Thread 0x7fffa11fa700 (LWP 4301)]
[New Thread 0x7fffa19fb700 (LWP 4302)]
[New Thread 0x7fffa21fc700 (LWP 4303)]
[New Thread 0x7fffa29fd700 (LWP 4304)]
[New Thread 0x7fffa377f700 (LWP 4305)]
[New Thread 0x7fffa09f9700 (LWP 4306)]
[New Thread 0x7fffa01f8700 (LWP 4307)]
[New Thread 0x7fff9f9f7700 (LWP 4308)]
[New Thread 0x7fff9f1f6700 (LWP 4309)]
I0801 20:13:47.989670  4291 opencl_split.cpp:632] Creating new Cycles device for OpenCL platform Intel(R) OpenCL HD Graphics, device Intel(R) Gen9 HD Graphics NEO.
[New Thread 0x7fff9e9f5700 (LWP 4310)]
I0801 20:13:47.990469  4310 session.cpp:753] Requested features:
Experimental features: Off
Max nodes group: 0
Nodes features: 0
Use Hair: False
Use Object Motion: False
Use Camera Motion: False
Use Baking: False
Use Subsurface: False
Use Volume: False
Use Branched Integrator: False
Use Patch Evaluation: False
Use Transparent Shadows: False
Use Principled BSDF: True
Use Denoising: False
Use Displacement: False
Use Background Light: True
I0801 20:13:47.990504  4310 opencl_split.cpp:761] Loading kernels for platform Intel(R) OpenCL HD Graphics, device Intel(R) Gen9 HD Graphics NEO.
I0801 20:13:47.990561  4310 opencl_util.cpp:297] OpenCL program base not found in cache.
I0801 20:13:48.008085  4310 opencl_util.cpp:324] Build options passed to clBuildProgram: '-cl-no-signed-zeros -cl-mad-enable -D__KERNEL_CL_KHR_FP16__ '.
I0801 20:13:48.008338  4310 opencl_util.cpp:297] Loaded program from /home/akien/.cache/cycles/kernels/cycles_kernel_base_46A2F28B047F7077207C6B26AD34C51F_66F686122DEAFAABB0CCA9DBEB37273C.clbin.
I0801 20:13:48.008409  4310 opencl_util.cpp:297] OpenCL program background not found in cache.
[... 44 lines cut ...]
    at /usr/src/debug/intel-igc-1.0.10-2.mga7.x86_64/IGC/OCLFE/igd_fcl_mcl/source/clang_tb.cpp:1691
#16 0x00007fffb06ce16d in IGC::FclOclTranslationCtx<0ul>::Impl::Translate(unsigned long, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, unsigned int) (this=0x7fff97605090, outVersion=1, src=0x7fff97610000, options=0x7fff97610040, internalOptions=0x7fff976100c0, tracingOptions=0x0, tracingOptionsCount=0)
    at /usr/src/debug/intel-igc-1.0.10-2.mga7.x86_64/IGC/AdaptorOCL/ocl_igc_interface/impl/fcl_ocl_translation_ctx_impl.h:117
#17 0x00007fffb06ce419 in IGC::FclOclTranslationCtx<1ul>::TranslateImpl(unsigned long, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, unsigned int) (this=0x7fff976100e0, outVersion=1, src=0x7fff97610000, options=0x7fff97610040, internalOptions=0x7fff976100c0, tracingOptions=0x0, tracingOptionsCount=0)
--Type <RET> for more, q to quit, c to continue without paging--c
    at /usr/src/debug/intel-igc-1.0.10-2.mga7.x86_64/IGC/AdaptorOCL/ocl_igc_interface/impl/fcl_ocl_translation_ctx_impl.cpp:41
#18 0x00007fffb1ef7d0a in IGC::FclOclTranslationCtx<1ul>::Translate<IGC::OclTranslationOutput<1ul> >(CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, unsigned int) (tracingOptionsCount=0, tracingOptions=0x0, internalOptions=<optimized out>, options=<optimized out>, src=<optimized out>, this=<optimized out>) at /usr/include/igc/o [... cut ...]
#29 0x00007ffff61de07f in std::execute_native_thread_routine(void*) (__p=0x7fffa743ca40) at ../../../../../libstdc++-v3/src/c++11/thread.cc:80
#30 0x00007ffff6cec04c in start_thread () at /lib64/libpthread.so.0
#31 0x00007ffff64edbcf in clone () at /lib64/libc.so.6
```

I'm not sure yet why RPM didn't give me the `-debuginfo` packages for `libopencl-clang`, I'll attempt to fix that.

As advised by @lwesiers, I ran the above with `export IGC_ShaderDumpEnableAll=1` and I'll send him the contents of `/tmp/IntelIGC/` by email.

akien-mga, 2019-08-01:
> As advised by @lwesiers, I ran the above with `export IGC_ShaderDumpEnableAll=1` and I'll send him the contents of `/tmp/IntelIGC/` by email.

Actually it's not so heavy so here it is as a ZIP: [IntelIGC-blender-segfault.zip](https://github.com/intel/compute-runtime/files/3458164/IntelIGC-blender-segfault.zip)

akien-mga, 2019-08-01:
Fixed the `opencl-clang` packaging and I now have relevant debug symbols in the backtrace:
```
I0801 20:45:13.923807 17873 opencl_split.cpp:761] Loading kernels for platform Intel(R) OpenCL HD Graphics, device Intel(R) Gen9 HD Graphics NEO.
I0801 20:45:13.923863 17873 opencl_util.cpp:297] OpenCL program base not found in cache.
I0801 20:45:13.938622 17873 opencl_util.cpp:324] Build options passed to clBuildProgram: '-cl-no-signed-zeros -cl-mad-enable -D__KERNEL_CL_KHR_FP16__ '.
I0801 20:45:13.938835 17873 opencl_util.cpp:297] Loaded program from /home/akien/.cache/cycles/kernels/cycles_kernel_base_46A2F28B047F7077207C6B26AD34C51F_66F686122DEAFAABB0CCA9DBEB37273C.clbin.
I0801 20:45:13.938882 17873 opencl_util.cpp:297] OpenCL program background not found in cache.
I0801 20:45:13.995116 17873 opencl_util.cpp:297] OpenCL program background not found on disk.
I0801 20:45:13.995455 17865 opencl_util.cpp:297] OpenCL program background not found in cache.
Cycles: compiling OpenCL program background...
I0801 20:45:14.057822 17865 opencl_util.cpp:297] Build flags: -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ -D__NO_SUBSURFACE__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__
[Detaching after vfork from child process 17874]
AL lib: (WW) GetSymbol: Failed to load jack_error_callback: /lib64/libjack.so.0: undefined symbol: jack_error_callback
AL lib: (WW) jack_msg_handler: Cannot connect to server socket err = No such file or directory
AL lib: (WW) jack_msg_handler: Cannot connect to server request channel
AL lib: (WW) jack_msg_handler: jack server is not running or cannot be started
AL lib: (WW) jack_msg_handler: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
AL lib: (WW) jack_msg_handler: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
AL lib: (WW) ALCjackBackendFactory_init: jack_client_open() failed, 0x11
AL lib: (WW) alc_initconfig: Failed to initialize backend "jack"
blender: /home/akien/Projects/mageia/Sandbox/_rpm/BUILD/intel-opencl-clang-8.0.72/spirv-llvm/lib/SPIRV/SPIRVWriter.cpp:1174: SPIRV::SPIRVValue* SPIRV::LLVMToSPIRV::transIntrinsicInst(llvm::IntrinsicInst*, SPIRV::SPIRVBasicBlock*): Assertion `cast<MemCpyInst>(II)->getSourceAlignment() == cast<MemCpyInst>(II)->getDestAlignment() && "Alignment mismatch!"' failed.
sh: line 1: 17875 Aborted                 (core dumped) "/home/akien/tmp/blender/build/bin/blender" "--background" "--factory-startup" "--python-expr" "import _cycles; _cycles.opencl_compile(r'1', r'Intel(R) Gen9 HD Graphics NEO', r'Intel(R) OpenCL HD Graphics', r'-cl-no-signed-zeros -cl-mad-enable -D__KERNEL_CL_KHR_FP16__ -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ -D__NO_SUBSURFACE__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__', r'kernel_background.cl', r'/home/akien/.cache/cycles/kernels/cycles_kernel_background_3A063A327B9939998F9E7BFA0349B693_13AC40ED2FE033D81EB1496FB3C61F4D.clbin')" > /dev/null
I0801 20:45:17.329959 17865 opencl_util.cpp:297] Separate-process building of /home/akien/.cache/cycles/kernels/cycles_kernel_background_3A063A327B9939998F9E7BFA0349B693_13AC40ED2FE033D81EB1496FB3C61F4D.clbin failed, will fall back to regular building.
Cycles: compiling OpenCL program background...
I0801 20:45:17.365284 17865 opencl_util.cpp:297] Build flags: -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ -D__NO_SUBSURFACE__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__
I0801 20:45:17.365296 17865 opencl_util.cpp:324] Build options passed to clBuildProgram: '-cl-no-signed-zeros -cl-mad-enable -D__KERNEL_CL_KHR_FP16__ -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ -D__NO_SUBSURFACE__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__'.
blender: /home/akien/Projects/mageia/Sandbox/_rpm/BUILD/intel-opencl-clang-8.0.72/spirv-llvm/lib/SPIRV/SPIRVWriter.cpp:1174: SPIRV::SPIRVValue* SPIRV::LLVMToSPIRV::transIntrinsicInst(llvm::IntrinsicInst*, SPIRV::SPIRVBasicBlock*): Assertion `cast<MemCpyInst>(II)->getSourceAlignment() == cast<MemCpyInst>(II)->getDestAlignment() && "Alignment mismatch!"' failed.

Thread 60 "blender" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffa19fb700 (LWP 17865)]
0x00007ffff6425a7a in raise () from /lib64/libc.so.6

(gdb) bt
#0  0x00007ffff6425a7a in raise () at /lib64/libc.so.6
#1  0x00007ffff640e524 in abort () at /lib64/libc.so.6
#2  0x00007ffff640e40f in _nl_load_domain.cold.0 () at /lib64/libc.so.6
#3  0x00007ffff641a9a2 in  () at /lib64/libc.so.6
#4  0x00007fffaff14ff7 in SPIRV::LLVMToSPIRV::transIntrinsicInst(llvm::IntrinsicInst*, SPIRV::SPIRVBasicBlock*) (this=0x7fff96b69240, II=0x7fff94b115b8, BB=0x7fff92c19da0)
    at /usr/src/debug/intel-opencl-clang-8.0.72-1.mga7.x86_64/spirv-llvm/lib/SPIRV/SPIRVWriter.cpp:1144
[... cut ...]
#26 0x000000000158dabf in ccl::OpenCLDevice::OpenCLProgram::compile_kernel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const*) ()
#27 0x000000000158efd1 in ccl::OpenCLDevice::OpenCLProgram::compile() ()
#28 0x00000000020c0603 in ccl::TaskScheduler::thread_run(int) ()
#29 0x00000000020c200e in ccl::thread::run(void*) ()
#30 0x00007ffff61de07f in std::execute_native_thread_routine(void*) (__p=0x7fffa6c3c7c0) at ../../../../../libstdc++-v3/src/c++11/thread.cc:80
#31 0x00007ffff6cec04c in start_thread () at /lib64/libpthread.so.0
#32 0x00007ffff64edbcf in clone () at /lib64/libc.so.6
```

dvrogozh, 2019-08-01:
As I see you get assertion failure:
```
blender: /home/akien/Projects/mageia/Sandbox/_rpm/BUILD/spirv-llvm/lib/SPIRV/SPIRVWriter.cpp:1174: SPIRV::SPIRVValue* SPIRV::LLVMToSPIRV::transIntrinsicInst(llvm::IntrinsicInst*, SPIRV::SPIRVBasicBlock*): Assertion cast<MemCpyInst>(II)->getSourceAlignment() == cast<MemCpyInst>(II)->getDestAlignment() && "Alignment mismatch!"' failed.
```

I wonder which version of LLVM you use? 7/8/9? In case of LLVM 7/8 it needs to be patched with certain patches listed in opencl-clang. And SPIRV translator should be patched as well. Was it and if yes with which patches?

I am trying to see the code in SPIRVWriter.cpp which thrown assertion, but I don't see it in 70, 80, 90 branches:
https://github.com/KhronosGroup/SPIRV-LLVM-Translator/blob/llvm_release_70/lib/SPIRV/SPIRVWriter.cpp#L1174
https://github.com/KhronosGroup/SPIRV-LLVM-Translator/blob/llvm_release_80/lib/SPIRV/SPIRVWriter.cpp#L1174
https://github.com/KhronosGroup/SPIRV-LLVM-Translator/blob/llvm_release_90/lib/SPIRV/SPIRVWriter.cpp#L1174

@AlexeySotkin, @AlexeySachkov who might be able to give some comments.

akien-mga, 2019-08-01:
I use LLVM 8.0.0 provided by my distro (though I rebuilt it in Debug mode to be able to use `IGC_ShaderDumpEnableAll`. As such I likely don't have the relevant patches you mentioned, but I guess it means Fedora and CentOS packages provided at https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl/packages/ don't either, since LLVM is not custom packaged? (CC @JacekDanecki)

I assume I should make a llvm build with those patches then? https://github.com/intel/opencl-clang/tree/ocl-open-80/patches/clang

For SPRIV translator, I do have patches applied from Jacek's package (see `%prep` section): https://copr-be.cloud.fedoraproject.org/results/jdanecki/intel-opencl/fedora-rawhide-x86_64/00976285-intel-opencl-clang/opencl-clang-8.0.72-1.spec

dvrogozh, 2019-08-01:
If I remember correctly, this spirv patch https://github.com/intel/opencl-clang/blob/ocl-open-80/patches/spirv/0001-Update-LowerOpenCL-pass-to-handle-new-blocks-represn.patch has its clang counterpart. So, yes, I would suggest to try apply patches from https://github.com/intel/opencl-clang/tree/ocl-open-80/patches/clang, rebuild LLVM and try again. Mind that these patches are from upstream and should appear in LLVM-9.

akien-mga, 2019-08-02:
I rebuild Clang with the patches from https://github.com/intel/opencl-clang/tree/ocl-open-80/patches/clang, and then rebuilt intel-opencl-clang, but I still get the same assert.

For SPIRV translator, I'm using https://github.com/KhronosGroup/SPIRV-LLVM-Translator/commit/83298e3c9b124486c16d0fde54c764a6c5a2b554 which is the one @JacekDanecki packaged. It seems to be for the LLVM 8.0 branch as expected, but it's a bit old.

I'm not familiar with the stack's versioning though so I don't know what are the "known good" versions to use for each component, I just reused the existing RPM packaging.

I'll give a try to the current https://github.com/KhronosGroup/SPIRV-LLVM-Translator/tree/llvm_release_80 and see what happens.

akien-mga, 2019-08-02:
> I'll give a try to the current https://github.com/KhronosGroup/SPIRV-LLVM-Translator/tree/llvm_release_80 and see what happens.

The patch in https://github.com/intel/opencl-clang/tree/ocl-open-80/patches/spirv didn't apply on `llvm_release_80/HEAD`, so I packaged https://github.com/KhronosGroup/SPIRV-LLVM-Translator/commit/6b3920fed310f223b1a69b0273eb7c243d9f6257 instead, which is the base revision tracked in https://github.com/intel/opencl-clang/commit/a09003852db739002e94c6e8d02b0a888ee23e98 (which is also the opencl-clang version that is currently packaged on https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl/ and that I use too).

I now get a different assert:
```
blender: /usr/include/llvm/ADT/ilist_iterator.h:140: llvm::ilist_iterator<OptionsT, IsReverse, IsConst>::reference llvm::ilist_iterator<OptionsT, IsReverse, IsConst>::operator*() const [with OptionsT = llvm::ilist_detail::node_options<llvm::Instruction, true, false, void>; bool IsReverse = false; bool IsConst = false; llvm::ilist_iterator<OptionsT, IsReverse, IsConst>::reference = llvm::Instruction&]: Assertion `!NodePtr->isKnownSentinel()' failed.
sh: line 1: 25964 Aborted                 (core dumped) "/home/akien/tmp/blender/build/bin/blender" "--background" "--factory-startup" "--python-expr" "import _cycles; _cycles.opencl_compile(r'1', r'Intel(R) Gen9 HD Graphics NEO', r'Intel(R) OpenCL HD Graphics', r'-cl-no-signed-zeros -cl-mad-enable -D__KERNEL_CL_KHR_FP16__ -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ -D__NO_SUBSURFACE__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__', r'kernel_background.cl', r'/home/akien/.cache/cycles/kernels/cycles_kernel_background_3A063A327B9939998F9E7BFA0349B693_13AC40ED2FE033D81EB1496FB3C61F4D.clbin')" > /dev/null
Cycles: compiling OpenCL program background...
blender: /usr/include/llvm/ADT/ilist_iterator.h:140: llvm::ilist_iterator<OptionsT, IsReverse, IsConst>::reference llvm::ilist_iterator<OptionsT, IsReverse, IsConst>::operator*() const [with OptionsT = llvm::ilist_detail::node_options<llvm::Instruction, true, false, void>; bool IsReverse = false; bool IsConst = false; llvm::ilist_iterator<OptionsT, IsReverse, IsConst>::reference = llvm::Instruction&]: Assertion `!NodePtr->isKnownSentinel()' failed.
```

dvrogozh, 2019-08-02:
@akien-mga : Indeed, spirv translator recently got a number of new commits in llvm_release_80 branch and custom patch noted in opencl-clang now fails to apply. Thanks for pointing this out. I've submitted intel/opencl-clang#83 to track resolution. @AlexeySotkin : we need your help here, also, please, check whether spirv translator plans to provide release update for 80 branch.

@akien-mga : Basically, you can expect that custom opencl-clang patch(es) should apply clean on top of the matching release. For example, in case of spirv translator:
```
# git clone https://github.com/intel/opencl-clang.git
# git checkout -b my-v8.0.1 v8.0.1  # v8.0.1 is a release tag
# 
# git clone https://github.com/KhronosGroup/SPIRV-LLVM-Translator.git
# git checkout -b my-v8.0.1-1 v8.0.1-1  # v8.0.1-1 is a release tag
# git am ../opencl-clang/patches/spirv/0001-Update-LowerOpenCL-pass-to-handle-new-blocks-represn.patch
Applying: Update LowerOpenCL pass to handle new blocks represntation in LLVM IR
# 
```

dvrogozh, 2019-08-02:
>  I packaged KhronosGroup/SPIRV-LLVM-Translator@6b3920f instead
This will likely be ok, but I would suggest to work either with the HEAD of branches (which was not possible in this particular case - don't hesitate to file issues in this case) or with the latest releases (here you should have succeeded). For example, these opencl-clang and spriv translator releases should correspond to each other:
* https://github.com/intel/opencl-clang/releases/tag/v8.0.1
* https://github.com/KhronosGroup/SPIRV-LLVM-Translator/releases/tag/v8.0.1-1

Mind that patch list might change from release to release...

> but maybe these 8.0 branches actually expect the current tip of llvm's 8.0 branch / recent 8.0.1 release?

Good question. I don't know:(. They might. Anyway, considering that we are experiencing and debugging some issue it makes sense to replicate an issue on the latest code (or fail to replicate and we will know it is fixed). The best I can suggest right now is try out latest tagged releases of each component. But optimally we need a patch update from opencl-clang team. @AlexeySotkin: please, help.

JacekDanecki, 2019-08-05:
> For SPIRV translator, I'm using [KhronosGroup/SPIRV-LLVM-Translator@83298e3](https://github.com/KhronosGroup/SPIRV-LLVM-Translator/commit/83298e3c9b124486c16d0fde54c764a6c5a2b554) which is the one @JacekDanecki packaged. It seems to be for the LLVM 8.0 branch as expected, but it's a bit old.
> I'm not familiar with the stack's versioning though so I don't know what are the "known good" versions to use for each component, I just reused the existing RPM packaging.

For copr builds I've reused commits from [19.28.13502 release](https://github.com/intel/compute-runtime/releases/tag/19.28.13502) and then updated Neo to [19.29.13530 release](https://github.com/intel/compute-runtime/releases/tag/19.29.13530).
Fedora builds use original llvm/clang from Fedora, so I don't apply patches on llvm nor clang. Only I apply patches on spirv-llvm-translator from opencl-clang.
Centos builds are different, as there is not llvm/clang 8 for Centos yet, so I'm building IGC, opencl-clang with llvm/clang sources, and all patches are applied during IGC build. In these builds llvm/clang is linked statically. 
The same build method as for Centos I'm using for deb packages for Ubuntu 16.04 on [intel-opencl ppa](https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl).
On [intel-opencl ppa](https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl) for Ubuntu 18.04 I'm using patched clang from Ubuntu 18.04, and apply patches for spirv-llvm-translator.
I've reproduced hang on Ubuntu 16.04 (the same as on Fedora 29 and 30) with blender 2.80-rc3, but did not reproduce aborts, as I've used released versions with assert disabled in spirv-llvm-translator.

AlexeySotkin, 2019-08-05:
> But optimally we need a patch update from opencl-clang team. @AlexeySotkin: please, help.

I have updated the patch in opencl-clang. But I don't think it will help to resolve the issue. Maybe full call stack of the latest assert `!NodePtr->isKnownSentinel()` could help.

JacekDanecki, 2019-10-29:
With [Neo 19.43.14583](https://github.com/intel/compute-runtime/releases/tag/19.43.14583) and Debug version of IGC 1.0.2714.1 there is assert in IGC instead of spirv-llvm-translator.

JacekDanecki, 2019-11-18:
As a workaound for IGC crash you can apply below patch, or set debug variable EnableStatelessToStatefulBufferOffsetOpt to 0. Unfortunately, although compilation passes there is gpu hang during kernel execution, as reported in #178
```
diff --git a/runtime/program/program.cpp b/runtime/program/program.cpp
index c1706d58..09fc438d 100644
--- a/runtime/program/program.cpp
+++ b/runtime/program/program.cpp
@@ -85,10 +85,6 @@ Program::Program(ExecutionEnvironment &executionEnvironment, Context *context, b
         if (DebugManager.flags.EnableStatelessToStatefulBufferOffsetOpt.get() != -1) {
             enableStatelessToStatefullWithOffset = DebugManager.flags.EnableStatelessToStatefulBufferOffsetOpt.get() != 0;
         }
-
-        if (enableStatelessToStatefullWithOffset) {
-            internalOptions += "-cl-intel-has-buffer-offset-arg ";
-        }
     }
 
     internalOptions += "-fpreserve-vec3-type ";
```

JacekDanecki, 2020-01-29:
Gpu hang reported in #178 was root-caused as Blender issue, where OpenCL kernels were developed with wrong assumptions about barrier usage on GPU device.

JacekDanecki, 2020-01-29:
@lwesiers any update on fixing segfault in IGC?

JacekDanecki, 2020-01-29:
As it's IGC problem I've created new issue on IGC side: https://github.com/intel/intel-graphics-compiler/issues/123
Closing this issue here.

closed 2020-01-29 by JacekDanecki