intel/compute-runtime #196: NEO in Fedora repositories

bkmgit, 2019-08-08:
It would be helpful to have NEO in the main Fedora repositories, since at present Beignet see [here](https://koji.fedoraproject.org/koji/packageinfo?packageID=beignet) is provided, but may not be appropriate for all future processors.

JacekDanecki, 2019-08-08:
Before Neo is in the main Fedora repositories, you can use packages from [copr](https://github.com/intel/compute-runtime/blob/master/documentation/Neo_in_distributions.md#fedora-30-and-rawhide)

kwizart, 2019-08-17:
+1 to have them in official Fedora repositories.
As a side note, intel-gmmlib is already in (I maintain that as part of the intel-media-driver VAAPI stack).
https://koji.fedoraproject.org/koji/buildinfo?buildID=1310353

Hopefully we can coordinate on a same version for the fedora repositories.

JacekDanecki, 2019-09-02:
@kwizart Sure, as intel-gmmlib version in Fedora is what Neo requires I'll not update intel-gmmlib package in copr. If Neo needs any newer version for specific release, and this version is not available in Fedora I'll notify you. Adding Neo to Fedora repositories is on my to-do list.

JacekDanecki, 2019-09-20:
@kwizart I can see you've updated intel-gmmlib for Fedora 31 and 32 to version 19.3.2.  As Neo has just [switched](https://github.com/intel/compute-runtime/commit/4de1d8e589b5c2addd20c48b42965f2768aa75e0) to this version and we use Fedora 30, 31 in external CI, I've updated intel-gmmlib on [copr](https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl/build/1037803/) for Fedora 30, 31, rawhide.

kwizart, 2019-09-20:
OK, then that's going to be an issue as the Fedora updates policy will forbid any ABI breakage for stables releases.
In others words if using your updated copr repos and the intel-media-driver (provided by rpmfusion-nonfree)  up to f30, end-users will see the breakage.

I think there are few ways to avoid the issue (in no particular order):
- Have a Fedora exception granted  to allow any intel-gmmlib updates. But then there is a need to verify that the intel-media-driver will work with the given libva releases and the newer intel-gmmlib
- When using a newer gmmlib source than provided by fedora you uses a different package name in your copr repo so the packages can co-exists.
- your copr repo doesn't replace the fedora intel-gmmlib one and can accommodate the version there (at least on older/stable releases.)
- Maybe we can set a directive for package to allow a given package to be parallel installable. But it's unlikely to work if NEO is added into fedora...

I hope for the long term that a stable ABI can be made for gmmlib

kwizart, 2019-09-20:
* And another way is probably to build the NEO stack as a module.
So it will use the system intel-gmmlib when possible or a modular version (that can be parallele installable since the SONAME will be different).

JacekDanecki, 2019-09-23:
As Neo has weekly releases we could more often update gmmlib than media-driver.
Even if we have different gmmlib installed in the system, we could find issues when using [media sharing](https://github.com/intel/compute-runtime/blob/master/documentation/cl_intel_va_api_media_sharing.md) extension.

JacekDanecki, 2019-10-11:
I'll rename gmmlib provided on copr (soon), so it will not conflict with package from Fedora.

truboxl, 2019-10-20:
What will happen to OpenCL support for older Intel on Fedora?

What should people with older Intel do on Fedora to get back OpenCL support? Beignet seems unavailable on F30...

JacekDanecki, 2019-10-28:
Beignet was deprecated and is no longer maintained by Intel. You can apply patches  from [Arch Linux](https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/beignet) to make Beignet compatible with recent versions of llvm.

kwizart, 2019-10-28:
@JacekDanecki anything missing to have a version in the official fedora repo ?
It would be easier to keep a version that works with the intel-gmmlib package in fedora for each fedora version, then update  intel-gmmlib as needed for your copr repo ?

JacekDanecki, 2019-10-31:
@kwizart If you asking  about gmmlib or Neo?

kwizart, 2019-10-31:
I'm asking on Neo, gmmlib is already in fedora.
Someone need to volunteer to have it (even if it's not the latest version) in a given fedora release.

JacekDanecki, 2019-11-05:
Adding Neo and it's dependencies to Fedora is on my todo list.
I've just updated gmmlib on copr to 19.3.4 as it's required by Neo. As I can see in koji the latest gmmlib is 19.3.2. In what Fedora release can we expect gmmlib 19.3.4?

JacekDanecki, 2020-01-24:
IGC stopped compiling on Fedora rawhide (32). New issue in IGC repository was created [here](https://github.com/intel/intel-graphics-compiler/issues/122)

JacekDanecki, 2020-01-28:
Fix for igc compilation was provided, and new package 1.0.3151-2.fc31 was prepared on copr.

JacekDanecki, 2020-02-12:
Fedora rawhide (32) switched to llvm 10 (llvm-10.0.0-0.1.rc1.fc32.x86_64), and [copr](https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl-ci/build/1237257) build failed. Released IGC was tested on llvm 9. I'll prepare IGC packages based on llvm 10.

JacekDanecki, 2020-02-12:
IGC packages compiled with llvm 10 for Fedora 32 were prepared on [copr:jdanecki/intel-opencl-ci](https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl-ci/monitor/detailed)

kwizart, 2020-02-12:
> Adding Neo and it's dependencies to Fedora is on my todo list.
> I've just updated gmmlib on copr to 19.3.4 as it's required by Neo. As I can see in koji the latest gmmlib is 19.3.2. In what Fedora release can we expect gmmlib 19.3.4?

intel-gmmlib can always be updated to latest in rawhide, (eventually branch if not too close to release).
https://koji.fedoraproject.org/koji/packageinfo?packageID=27672

But for older branches it would be nice to have a version of neo that would be compatible with the given intel-gmmlib.

JacekDanecki, 2020-02-17:
intel-opencl-clang package stopped to link on Fedora rawhide. New issue created in [opencl-clang repository](https://github.com/intel/opencl-clang/issues/108)

JacekDanecki, 2020-02-17:
As I can see in Fedora 31 there is gmmlib [19.3.2-1](https://koji.fedoraproject.org/koji/buildinfo?buildID=1381935), so Neo [19.44.14658](https://github.com/intel/compute-runtime/releases/tag/19.44.14658) is compatible with such version. Packages for Fedora 31 are available on [sourceforge mirror](https://sourceforge.net/projects/intel-compute-runtime/files/19.44.14658/fedora-31)

For Fedora 30 gmmlib is version [19.2.3-1](https://koji.fedoraproject.org/koji/buildinfo?buildID=1310353), so Neo [19.37.14191](https://github.com/intel/compute-runtime/releases/tag/19.37.14191) can be used.

JacekDanecki, 2020-03-02:
Workaround [patch](https://github.com/JacekDanecki/neo-specs/blob/master/specs/fedora/32/clang.patch) has been prepared for Fedora 32 and rawhide.
Builds on copr: https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl/build/1279706

JacekDanecki, 2020-04-08:
IGC stopped to compile on fedora:
* 32: https://download.copr.fedorainfracloud.org/results/jdanecki/intel-opencl-ci/fedora-32-x86_64/01330622-intel-igc/builder-live.log.gz
* rawhide: https://download.copr.fedorainfracloud.org/results/jdanecki/intel-opencl-ci/fedora-rawhide-x86_64/01330622-intel-igc/builder-live.log.gz

IGC can be compiled with:
```
gcc (GCC) 10.0.1 20200216 (Red Hat 10.0.1-0.8)
```
Fails with:
```
gcc (GCC) 10.0.1 20200328 (Red Hat 10.0.1-0.11)
```

JacekDanecki, 2020-04-16:
With patch https://github.com/intel/intel-graphics-compiler/commit/a58dd6de4c29595a0f93cff167b487d777e4559e.patch I was able to build IGC packages on Fedora 32 and rawhide on copr.

JacekDanecki, 2020-04-24:
New IGC compilation issue on Fedora rawhide: https://github.com/intel/intel-graphics-compiler/issues/132

JacekDanecki, 2020-05-06:
Another IGC issue: https://github.com/intel/intel-graphics-compiler/pull/133
Currently only builds under Fedora 30 are building on copr.

JacekDanecki, 2020-07-08:
@kwizart As Fedora 32 and 33 use gmmlib 20.2.2, I use these versions to build Neo on copr for Fedora 32 and 33. For Fedora 31 I build gmmlib on copr.

kwizart, 2020-07-08:
If there is no ABI break with gmmlib we can go forward that way.
But it will be nice to have a dedicated version of  Neo imported in fedora that will match with the fedora/(and 3rd part) gmmlib version.

JacekDanecki, 2020-07-08:
Gmmlib 20.2.2 provides libigdgmm.so.11, so it should not break ABI, I've not verified it.
Currently Neo uses IGFX_DG1 symbol from gmmlib 20.2.2, so to compile Neo under Fedora 31 with original gmmlib, we should disable support for DG1 or cherry-pick changes for DG1 support from gmmlib 20.2.2 to Fedora 31 gmmlib.

michal-kopania, 2020-10-29:
In Fedora 33 gmmlib version is 20.3.2 Installing opencl causes error:
Problem: conflicting requests
  - nothing provides intel-gmmlib = 20.2.2 needed by intel-opencl-20.28.17293-2.fc33.x86_64

JacekDanecki, 2020-10-29:
See issue https://github.com/intel/compute-runtime/issues/356
I've prepared Neo spec update and locally Neo 20.28.17293-3 was built correctly.
Unfortunately on copr build failed: https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl/build/1727203/

michal-kopania, 2020-11-02:
I have checked issue #356 and downgraded gmmlib to 20.2.2.
I was able to install intel-opencl but running clinfo caused core dump. 
```
clinfo
Abort was called at 42 line in file:
/builddir/build/BUILD/compute-runtime-20.28.17293/shared/source/built_ins/built_ins.cpp
Aborted (core dumped)

```
When I removed intel-opencl clinfo started to work again.
Platform #0: NVIDIA CUDA
 -- Device #0: GeForce 940MX
Platform #1: Portable Computing Language
 -- Device #0: pthread-Intel(R) Core(TM) i7-7500U CPU @ 2.70GHz
Platform #2: Clover

JacekDanecki, 2020-11-02:
What IGC version have you installed?

michal-kopania, 2020-11-02:
1.0.4241

JacekDanecki, 2020-11-02:
Can you provide output from:
```
LD_DEBUG=libs clinfo
```
and file log.txt created by command:
```
strace -f -s 255 -o log.txt clinfo
```

JacekDanecki, 2020-11-02:
After dnf update on Fedora 33 I've reproduced abort:
```
clinfo
Abort was called at 42 line in file:
/builddir/build/BUILD/compute-runtime-20.28.17293/shared/source/built_ins/built_ins.cpp
Aborted (core dumped)
```
In LD_DEBUG=libs clinfo I can see:
```
      5146:     /lib64/libigc.so.1: error: symbol lookup error: undefined symbol: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEmc, version LLVM_10 (fatal)
Abort was called at 42 line in file:
/builddir/build/BUILD/compute-runtime-20.28.17293/shared/source/built_ins/built_ins.cpp

$ nm -D /lib64/libigc.so.1 | grep  _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEmc
                 U _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEmc@@LLVM_10
$ c++filt _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEmc
std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::resize(unsigned long, char)
```
So for some reasons this symbol is missing in newer Fedora packages, so I need to rebuild IGC too. It could explain problems with Neo build on copr.

michal-kopania, 2020-11-02:
I do not know if you still need it, but I attach a link to the requested output generated before and after installing intel-opencl from: 
```
==============================================================================================================================================================================================
 Package                               Architecture                Version                                    Repository                                                                 Size
==============================================================================================================================================================================================
Installing:
 intel-opencl                          x86_64                      20.28.17293-2.fc33                         copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                      1.2 M
Installing dependencies:
 intel-igc-core                        x86_64                      1.0.4241-1.fc33                            copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                      6.6 M
 intel-igc-opencl                      x86_64                      1.0.4241-1.fc33                            copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                       73 k

Transaction Summary
==============================================================================================================================================================================================
Install  3 Packages

```
https://drive.google.com/file/d/1DihK3vjJcpeDf6dCRbR-1mQ4Mzi_Agz9/view?usp=sharing
https://drive.google.com/file/d/1q2wpL-8wRP2XXI2A98RKReYvnW1X9pYP/view?usp=sharing

JacekDanecki, 2020-11-09:
Packages for:
*    Neo 20.44.18297
*    IGC 1.0.5353
*    L0 1.0.16
 were built on copr for Fedora 33.
 Neo is using gmmlib from Fedora (updates-testing):
```
dnf install intel-opencl
Last metadata expiration check: 0:00:17 ago on Mon Nov  9 22:40:25 2020.
Dependencies resolved.
==============================================================================================================================================================================================================================================================
 Package                                               Architecture                                Version                                                    Repository                                                                                 Size
==============================================================================================================================================================================================================================================================
Installing:
 intel-opencl                                          x86_64                                      20.44.18297-1.fc33                                         copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                                      1.4 M
Installing dependencies:
 intel-gmmlib                                          x86_64                                      20.3.2-1.fc33                                              updates-testing                                                                           129 k
 intel-igc-core                                        x86_64                                      1.0.5353-1.fc33                                            copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                                      142 M
 intel-igc-opencl                                      x86_64                                      1.0.5353-1.fc33                                            copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                                       33 M

Transaction Summary
==============================================================================================================================================================================================================================================================
Install  4 Packages

Total download size: 177 M
Installed size: 680 M
```

ozeszty, 2020-11-17:
Thanks for the copr builds! Unfortunately currently I can't update them due to conflict with `intel-opencl-clang-10.0.12-1.fc33.x86_64`:
```
--> Rozpoczynanie rozwiązywania zależności
---> Pakiet intel-igc-core.x86_64 1.0.4241-1.fc33 zostanie zaktualizowany
---> Pakiet intel-igc-core.x86_64 1.0.5435-1.fc33 będzie aktualizacją
---> Pakiet intel-igc-opencl.x86_64 1.0.4241-1.fc33 zostanie zaktualizowany
---> Pakiet intel-igc-opencl.x86_64 1.0.5435-1.fc33 będzie aktualizacją
---> Pakiet intel-opencl.x86_64 20.28.17293-2.fc33 zostanie zaktualizowany
---> Pakiet intel-opencl.x86_64 20.45.18403-1.fc33 będzie aktualizacją
---> Pakiet intel-gmmlib.x86_64 20.2.2-1.fc33 zostanie zaktualizowany
---> Pakiet intel-gmmlib.x86_64 20.3.2-1.fc33 będzie aktualizacją
--> Ukończono rozwiązywanie zależności
Rozwiązano zależności.

Problem: package intel-opencl-clang-10.0.12-1.fc33.x86_64 requires libLLVMSPIRVLib.so.10()(64bit), but none of the providers can be installed
  - cannot install both spirv-llvm-translator-11.0.0-0.1.fc33.x86_64 and spirv-llvm-translator-10.0.12-1.fc33.x86_64
  - cannot install the best update candidate for package spirv-llvm-translator-10.0.12-1.fc33.x86_64
  - cannot install the best update candidate for package intel-opencl-clang-10.0.12-1.fc33.x86_64
=======================================================================================================================================================================================================================================
 Pakiet                                              Architektura                         Wersja                                              Repozytorium                                                                       Rozm.
=======================================================================================================================================================================================================================================
Aktualizowanie:
 intel-gmmlib                                        x86_64                               20.3.2-1.fc33                                       updates-testing                                                                    129 k
 intel-igc-core                                      x86_64                               1.0.5435-1.fc33                                     copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                               142 M
 intel-igc-opencl                                    x86_64                               1.0.5435-1.fc33                                     copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                                33 M
 intel-opencl                                        x86_64                               20.45.18403-1.fc33                                  copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                               1.4 M
Pomijanie sprzecznych pakietów:
(dodanie „--best --allowerasing” do wiersza poleceń wymusi ich aktualizację):
 spirv-llvm-translator                               x86_64                               11.0.0-0.1.fc33                                     updates-testing                                                                    666 k
...
Błąd: Błąd testu transakcji:
  plik /usr/lib64/libopencl-clang.so.10 z instalacji intel-igc-opencl-1.0.5435-1.fc33.x86_64 jest w konflikcie z plikiem z pakietu intel-opencl-clang-10.0.12-1.fc33.x86_64
```

JacekDanecki, 2020-11-17:
I've changed how IGC is being built for Fedora, currently all IGC components are built with llvm/sources during the same build, so package intel-opencl-clang is not created anymore, and libopencl-clang.so.10 library is provided in intel-igc-opencl package.
Please remove old intel-opencl-clang package, and then install new IGC packages.

ozeszty, 2020-11-17:
I understand, I was trying to remove `intel-opencl-clang` before, but it was wiping out all packages from your copr.
In that case, I think, `intel-igc-opencl` should obsolete `intel-opencl-clang` (I haven't checked what else it containes) - then this issue would be resolved by dnf without any manual intervention.
If you update `intel-igc-opencl` spec and build it, I can test whether this works. If not, I'll just reinstall.

JacekDanecki, 2020-11-18:
I'll fix dependencies in next IGC update.

ozeszty, 2020-12-10:
Thanks, I'll test it then.

JacekDanecki, 2021-01-21:
@ozeszty IGC specs were updated for Fedora 32 and 33 to add conflict with intel-opencl-clang:
* https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl/build/1891301/
* https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl/build/1891302/

JacekDanecki, 2021-02-11:
IGC compilation under Fedora rawhide failed: https://github.com/intel/intel-graphics-compiler/issues/165

JacekDanecki, 2021-02-24:
Neo 21.07.19042 built on Fedora 34 and rawhide (35): https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl/build/2016696/

masterzorag, 2021-05-18:
Operative on:
```sh
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           92
Model name:                      Intel(R) Celeron(R) CPU N3350 @ 1.10GHz
```
Attached [clinfo output](https://github.com/intel/compute-runtime/files/6504777/clinfo.txt)

Builted [sample](https://github.com/cartermc24/RSA-OpenCL) linking the ocl-icd library and preloaded it, executes as expected:
```sh
[user@localhost RSA-OpenCL]$ ldd a.out
	linux-vdso.so.1 (0x00007ffd69be4000)
	libigdrcl.so => not found
	libOpenCL.so.1 => /lib64/libOpenCL.so.1 (0x00007fbb2d4dc000)
	libc.so.6 => /lib64/libc.so.6 (0x00007fbb2d311000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007fbb2d30a000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fbb2d524000)
[user@localhost RSA-OpenCL]$ LD_PRELOAD=/usr/lib64/intel-opencl/libigdrcl.so ./a.out example_conf.txt out
CasRSA_CL OpenCL 1.2 implementation of RSA
Carter McCardwell, Northeastern University NUCAR
http://coe.neu.edu/~cmccardw - mccardwell.net
--------------------------------------
INPUT:
	M:35

	p:23

	q:41

This GPU supports 12 compute units
Building CL Kernel...	[Done]
Run Parameters: Local: 1 - Global: 1
Encrypted Result: 545

Writing result to outfile...	[Done]
Done - Time taken: 32082.179688 ms
[user@localhost RSA-OpenCL]$
```
flawless, thanks :+1:

JacekDanecki, 2021-06-02:
Closing this issue, as we don't plan maintain Neo in Fedora repositories. Neo packages for Fedora will be provided by copr: https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl

closed 2021-06-02 by JacekDanecki