intel/compute-runtime #198: Driver stopped working

isgursoy, 2019-08-09:
After #132 I needed to make OpenCL by iGPU. It was good in the fresh system for some time. This morning I noticed freezes while running some OpenGL programs. OpenCL was also broken in Intel. I just enabled auto-updates, nothing special to kill Intel.

All drivers are latest.
cat /sys/class/drm/card0/error:
```
GPU HANG: ecode 9:0:0x85ddfffb, in luxmark.bin [4310], reason: Hang on rcs0, action: reset
Kernel: 4.15.0-58-generic
Time: 1565370386 s 992781 us
Boottime: 112 s 766352 us
Uptime: 108 s 773733 us
Active process (on ring render): luxmark.bin [4310], score 0
Reset count: 0
Suspend count: 0
Platform: SKYLAKE
PCI ID: 0x191b
PCI Revision: 0x06
PCI Subsystem: 1991:5594
IOMMU enabled?: 0
DMC loaded: yes
DMC fw version: 1.26
GT awake: yes
RPM wakelock: yes
PM suspended: no
EIR: 0x00000000
IER: 0x08000000
GTIER[0]: 0x01010101
GTIER[1]: 0x01010101
GTIER[2]: 0x00000070
GTIER[3]: 0x00000101
PGTBL_ER: 0x00000000
FORCEWAKE: 0xffff0001
DERRMR: 0x2077efef
CCID: 0x00000000
Missed interrupts: 0x00000000
  fence[0] = 00000000
  fence[1] = 00000000
  fence[2] = 00000000
  fence[3] = 00000000
  fence[4] = 00000000
  fence[5] = 00000000
  fence[6] = 00000000
  fence[7] = 00000000
  fence[8] = 00000000
  fence[9] = 00000000
  fence[10] = 00000000
  fence[11] = 00000000
  fence[12] = 00000000
  fence[13] = 00000000
  fence[14] = 00000000
  fence[15] = 00000000
  fence[16] = 00000000
  fence[17] = 00000000
  fence[18] = 00000000
  fence[19] = 00000000
  fence[20] = 00000000
  fence[21] = 00000000
  fence[22] = 00000000
  fence[23] = 00000000
  fence[24] = 00000000
  fence[25] = 00000000
  fence[26] = 00000000
  fence[27] = 00000000
  fence[28] = 00000000
  fence[29] = 00000000
  fence[30] = 00000000
  fence[31] = 00000000
ERROR: 0x00000000
FAULT_TLB_DATA: 0x00000009 0xa1f408a9
DONE_REG: 0x07ffffff
render command stream:
[... 786 lines cut ...]
  printf() buffer size                            4194304 (4MiB)
  Built-in kernels                                block_motion_estimate_intel;block_advanced_motion_estimate_check_intel;block_advanced_motion_estimate_bidirectional_check_intel;
  Motion Estimation accelerator version (Intel)   2
    Device-side AVC Motion Estimation version     1
      Supports texture sampler use                Yes
      Supports preemption                         No
  Device Extensions                               cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_depth_images cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_image2d_from_buffer cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_media_block_io cl_intel_driver_diagnostics cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_khr_fp64 cl_khr_subgroups cl_khr_il_program cl_intel_spirv_device_side_avc_motion_estimation cl_intel_spirv_media_block_io cl_intel_spirv_subgroups cl_khr_spirv_no_integer_wrap_decoration cl_khr_mipmap_image cl_khr_mipmap_image_writes cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_motion_estimation cl_intel_device_side_avc_motion_estimation cl_intel_advanced_motion_estimation cl_intel_va_api_media_sharing 

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  No platform
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   No platform
  clCreateContext(NULL, ...) [default]            No platform
  clCreateContext(NULL, ...) [other]              Success [NV]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  No platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  Invalid device type for platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  No platform
	NOTE:	your OpenCL library only supports OpenCL 2.0,
		but some installed platforms support OpenCL 2.1.
		Programs using 2.1 features may crash
		or behave unexepectedly

```

JacekDanecki, 2019-08-13:
Do you know what kernel version have you used before update?

isgursoy, 2019-08-13:
Hi,
As I see, OpenCL is not only trigger for driver reset and it begins* after nvidia driver installation. 
Kernel was 5.0.0.23. I tried kernels since 4.15 to 5.3. 
*I see rare xorg hangs in Intel only session. I also tried ubuntu live usb versions since 16.04.0 to 19.04. Live session also suffers from driver reset.

JacekDanecki, 2019-08-26:
For Xorg what video driver are you using: intel or modesetting?
What is the output of below command executed as root:
```
 cat /proc/`pidof Xorg`/maps | grep drivers | awk '{print $6}' | sort | uniq
```

JacekDanecki, 2019-09-27:
Closing after a month of inactivity.

closed 2019-09-27 by JacekDanecki