intel/compute-runtime #199: Upgrade fails due to dependency name on Ubuntu Eoan

dino99, 2019-08-10:


dino99, 2019-08-10:
Trying to upgade from  19.13.12717 to 19.29.13530
but get error:
The following packages have unmet dependencies:
 intel-opencl-icd : Depends: libigdgmm5 but it is not installable

Even if dependency is pointing to libigdgmm9, which is the only Eoan version available.
Looks like the SONAME bump is disturbing the upgrade script.

> intel-gmmlib (19.2.3+ds1-2) unstable; urgency=medium
> 
>   * Upload to unstable.
>   * debian/control: Bump Standards-Version.
> 
>  -- Sebastian Ramacher <sramacher@debian.org>  Sun, 07 Jul 2019 18:43:00 +0200
> 
> intel-gmmlib (19.2.3+ds1-1) experimental; urgency=medium
> 
>   * New upstream release.
>   * SONAME bump: libigdgmm5 -> libigdgmm9.
>   * debian/patches: Refresh patches.
>   * debian/: Bump debhelper compat to 12.
>   * debian/control:
>     - Bump Standards-Version.
>     - Set RRR: no.
>   * debian/libigdgmm-dev.install: Remove static library, no longer built
>     upstream.
> 
>  -- Sebastian Ramacher <sramacher@debian.org>  Sun, 07 Jul 2019 11:05:49 +0200

tjaalton, 2019-08-10:
the hardcoded dependency on gmmlib was wrong, bumped it now in -0ubuntu3 which should be available soon

not an upstream bug anyway

dino99, 2019-08-10:
Upgrade ok with that last ubuntu3 version; thanks for the fix

closed 2019-08-10 by dino99