intel/compute-runtime #200: Half of compute units are detected on CentOS

jinz2014, 2019-08-11:
I am not sure if others have seen the issue before. After installing the runtime using yum or from the source, the GPU runtime can only detect half of the total number of compute units of an integrated GPU on CentOS7.  

Thanks for your investigation! 

  Max compute units                               24  (It should be 48)

JacekDanecki, 2019-08-13:
Can you provide output from command:
```
lspci -nn
```

jinz2014, 2019-08-13:
00:00.0 Host bridge [0600]: Intel Corporation Broadwell-U Host Bridge - DMI [8086:1618] (rev 0a)
00:01.0 PCI bridge [0604]: Intel Corporation Broadwell-U PCI Express x16 Controller [8086:1601] (rev 0a)
00:02.0 VGA compatible controller [0300]: Intel Corporation Iris Pro Graphics P6300 [8086:162a] (rev 0a)
00:1d.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 [8086:8c26] (rev 05)
00:1f.0 ISA bridge [0601]: Intel Corporation QM87 Express LPC Controller [8086:8c4f] (rev 05)
00:1f.2 SATA controller [0106]: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] [8086:8c03] (rev 05)
01:00.0 Ethernet controller [0200]: Mellanox Technologies MT27520 Family [ConnectX-3 Pro] [15b3:1007]

jinz2014, 2019-08-23:
Could you please give some updates ?

JacekDanecki, 2019-08-26:
What is the output of the below command executed as root:
```
cat /sys/kernel/debug/dri/0/i915_capabilities | grep "EU total"
```
What kernel version are you using?

JacekDanecki, 2019-08-26:
Can you also provide content of this file: /sys/kernel/debug/dri/0/i915_sseu_status

JacekDanecki, 2019-08-27:
Neo is using number of compute units from [kernel](https://github.com/intel/compute-runtime/blob/master/runtime/os_interface/linux/hw_info_config.cpp#L94), and it looks like Centos kernel reports incorrect value of EU's.
On kernel 4.15.0-46-generic from Ubuntu 18.04.1 content of i915_sseu_status is different, with correct value of EU:
```
SSEU Device Info
  Available Slice Mask: 0003
  Available Slice Total: 2
  Available Subslice Total: 6
  Available Subslice Mask: 0007
  Available Subslice Per Slice: 3
  Available EU Total: 48
  Available EU Per Subslice: 8
  Has Pooled EU: no
  Has Slice Power Gating: yes
  Has Subslice Power Gating: no
  Has EU Power Gating: no
SSEU Device Status
  Enabled Slice Mask: 0000
  Enabled Slice Total: 0
  Enabled Subslice Total: 0
  Enabled Subslice Mask: 0000
  Enabled Subslice Per Slice: 0
  Enabled EU Total: 0
  Enabled EU Per Subslice: 0
```
Can you check newer kernel from https://www.kernel.org?

JacekDanecki, 2019-08-27:
Similar  issue was fixed in Ubuntu by patch provided in [this bugzilla](https://bugs.freedesktop.org/show_bug.cgi?id=108712). I've verified that Centos kernel has not this patch, so you can apply patch from [comment 8](https://bugs.freedesktop.org/show_bug.cgi?id=108712#c8) and rebuild kernel package.

JacekDanecki, 2019-08-27:
On [this wiki page](https://wiki.centos.org/HowTos/Custom_Kernel) you can find how to build kernel with custom patches.

dvrogozh, 2019-08-28:
Patch got sent to Linux stable: https://www.spinics.net/lists/stable/msg325511.html

dvrogozh, 2019-09-07:
kernel.org 4.19 LTS got patched: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v4.19.71&id=a3eb2eba2f79950b788342562604d36c5a260da7

I hope distros will pick up that at some point.