intel/compute-runtime #202: questions about relations between Neo driver and intel SDK for Opencl

huangzhiyuan, 2019-08-14:
Hi, My system is Ubuntu16.04, and I used to install intel [SDK ](http://registrationcenter-download.intel.com/akdlm/irc_nas/vcp/15625/intel_sdk_for_opencl_applications_2019.4.314.tar.gz)for opencl, then use **clinfo** command detect the device on my machine, like cpu or GEN8/GEN9. So I want know the relations between Neo driver and Intel SDK for opencl. Is only Neo driver  enough? or need both? After install the release [deb-file ](https://github.com/intel/compute-runtime/releases) can I use **clinfo** to verify the hardware device? Look forward to your reply， thanks~~

JacekDanecki, 2019-08-14:
Neo is OpenCL runtime library, and is enough to execute OpenCL applications. See [this guide](https://github.com/bashbaug/OpenCLPapers/blob/markdown/OpenCLOnLinux.md) or [this article](https://software.intel.com/en-us/articles/opencl-drivers).
SDK contains additional tools to develop OpenCL applications, see [this article](https://software.intel.com/en-us/openclsdk-devguide).
Starting with Neo 19.30.13641 release we provide binary packages for Ubuntu 18.04.
For Ubuntu 16.04 you can use packages from [ppa repository](https://github.com/intel/compute-runtime/blob/master/documentation/Neo_in_distributions.md#ubuntu-1604-and-1804-ppa).

MichalMrozek, 2019-08-26:
Looks like question is answered.
Feel free to re-open.

closed 2019-08-26 by MichalMrozek