intel/compute-runtime #205: extGetSurfaceHandle segfaults with no way to avoid

hedgepigdaniel, 2019-08-27:
When using ffmpeg to decode video with VA-API and then using openCL filters directly on the returned hardware frames, I experienced a segmentation fault in compute-runtime. For example:

```
$ ffmpeg -init_hw_device vaapi=intel:/dev/dri/renderD128 -init_hw_device opencl=ocl@intel -hwaccel vaapi -hwaccel_device intel -filter_hw_device ocl -hwaccel_output_format vaapi -i 1382.mkv -t 10 -filter:v 'hwmap=derive_device=opencl,unsharp_opencl' -f null -
```

compute-runtime version: 19.32.13826

Stacktrace:

```
#0  0x0000000000000000 in  ()
#1  0x00007f00c8596f98 in NEO::VASharingFunctions::extGetSurfaceHandle(unsigned int*, unsigned int*) (this=0x55cfb2011fe0, this=0x55cfb2011fe0, handleId=0x7ffe4338d330, vaSurface=0x7ffe4338d550)
    at /usr/src/debug/compute-runtime-19.32.13826/runtime/sharings/va/va_sharing_functions.h:40
#2  0x00007f00c8596f98 in NEO::VASurface::createSharedVaSurface(NEO::Context*, NEO::VASharingFunctions*, unsigned long, unsigned int*, unsigned int, int*)
    (context=0x55cfb1669480, sharingFunctions=0x55cfb2011fe0, flags=1, surface=0x7ffe4338d550, plane=0, errcodeRet=<optimized out>) at /usr/src/debug/compute-runtime-19.32.13826/runtime/sharings/va/va_surface.cpp:64
#3  0x00007f00c859600e in clCreateFromVA_APIMediaSurfaceINTEL(cl_context, cl_mem_flags, VASurfaceID*, cl_uint, cl_int*) (context=<optimized out>, flags=1, surface=0x7ffe4338d550, plane=0, errcodeRet=0x7ffe4338d554)
    at /usr/src/debug/compute-runtime-19.32.13826/runtime/sharings/va/cl_va_api.cpp:51
#4  0x00007f00e9f7fcb0 in opencl_map_from_qsv (flags=<optimized out>, src=0x55cfb178c340, dst=0x55cfb178c980, dst_fc=0x55cfb178bb40) at libavutil/hwcontext_opencl.c:2277
#5  0x00007f00e9f7fcb0 in opencl_map_to (hwfc=0x55cfb178bb40, dst=0x55cfb178c980, src=0x55cfb178c340, flags=<optimized out>) at libavutil/hwcontext_opencl.c:2830
#6  0x00007f00ebb17ff1 in hwmap_filter_frame (link=link@entry=0x55cfb1852640, input=<optimized out>) at libavfilter/vf_hwmap.c:339
#7  0x00007f00eba6aae0 in ff_filter_frame_framed (frame=<optimized out>, link=0x55cfb1852640) at libavfilter/avfilter.c:1071
#8  0x00007f00eba6aae0 in ff_filter_frame_to_filter (link=0x55cfb1852640) at libavfilter/avfilter.c:1219
#9  0x00007f00eba6aae0 in ff_filter_activate_default (filter=<optimized out>) at libavfilter/avfilter.c:1268
#10 0x00007f00eba6aae0 in ff_filter_activate (filter=<optimized out>) at libavfilter/avfilter.c:1430
#11 0x00007f00eba6db68 in ff_filter_graph_run_once (graph=graph@entry=0x55cfb2005e40) at libavfilter/avfiltergraph.c:1454
#12 0x00007f00eba6f419 in push_frame (graph=0x55cfb2005e40) at libavfilter/buffersrc.c:187
#13 0x00007f00eba6f419 in av_buffersrc_add_frame_internal (ctx=ctx@entry=0x55cfb202a800, frame=frame@entry=0x55cfb1d862c0, flags=flags@entry=4) at libavfilter/buffersrc.c:261
#14 0x00007f00eba6f876 in av_buffersrc_add_frame_flags (ctx=0x55cfb202a800, frame=0x55cfb1d862c0, flags=4) at libavfilter/buffersrc.c:170
#15 0x000055cfaf2d0984 in ifilter_send_frame (frame=0x55cfb1d862c0, ifilter=0x55cfb1649d40) at fftools/ffmpeg.c:2196
#16 0x000055cfaf2d0984 in send_frame_to_filters (ist=0x55cfb1dd6040, decoded_frame=0x55cfb1d862c0) at fftools/ffmpeg.c:2270
#17 0x000055cfaf2d119e in decode_video (decode_failed=<optimized out>, eof=<optimized out>, duration_pts=<optimized out>, got_output=<optimized out>, pkt=<optimized out>, ist=<optimized out>) at fftools/ffmpeg.c:2469
#18 0x000055cfaf2d119e in process_input_packet (ist=<optimized out>, pkt=0x7ffe4338dad0, no_eof=0) at fftools/ffmpeg.c:2623
#19 0x000055cfaf2d4369 in process_input (file_index=<optimized out>) at fftools/ffmpeg.c:4518
#20 0x000055cfaf2d4369 in transcode_step () at fftools/ffmpeg.c:4638
#21 0x000055cfaf2d4369 in transcode () at fftools/ffmpeg.c:4692
#22 0x000055cfaf2b2176 in main (argc=24, argv=0x7ffe4338e258) at fftools/ffmpeg.c:4894
```

I stepped through `VASharingFunctions::initFunctions()` and found that `reinterpret_cast<VAExtGetSurfaceHandlePFN>(getLibFunc("DdiMedia_ExtGetSurfaceHandle"));` returned null, and hence the segfault occurred when calling `vaExtGetSurfaceHandlePFN` which was null.

I found that whether or not the `DdiMedia_ExtGetSurfaceHandle` symbol was available depends on which VA-API runtime is being used. Specifically,
- It works fine when https://github.com/intel/media-driver/ is used, which on Arch requires it to be installed and the environment variable `LIBVA_DRIVER_NAME=iHD`
- It segfaults when using https://github.com/intel/intel-vaapi-driver (which unfortunately is the default in the absence of environment variables)

I'm inclined to think that this means that the `cl_intel_va_api_media_sharing` extension is not supported in the case where a libva driver other than https://github.com/intel/media-driver/ (or another driver that implements `DdiMedia_ExtGetSurfaceHandle`) is in use, since there is no way `clCreateFromVA_APIMediaSurfaceINTEL` could succeed. Therefore in that case:
- it should not appear in the list of supported extensions reported by compute-runtime
- The specific dependency on https://github.com/intel/media-driver/ should be documented

What do you think?

JacekDanecki, 2019-08-27:
cl_intel_va_api_media_sharing extension in Neo was implemented and tested with [iHD media driver](https://github.com/intel/media-driver). As intel-vaapi-driver will not be actively maintained this extension will be supported by [iHD media driver](https://github.com/intel/media-driver). For more details about media support see [this page](https://01.org/linuxmedia)

hedgepigdaniel, 2019-08-29:
Sure, that would help.

I understand that you don't plan to implement the extension for intel-vaapi-driver, but it would be nice if intel-compute-runtime would accurately report this fact rather than allowing a segfault to happen.

Am I wrong to think that since intel-compute-runtime doesn't support the `cl_intel_va_api_media_sharing` extension when intel-vaapi-driver is in use, it is inaccurate to report the extension in the list of supported device extensions in that environment?

```shell
$ LIBVA_INTEL_DRIVER=i965 clinfo
Number of platforms                               1
  Platform Name                                   Intel(R) OpenCL HD Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 2.1 
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_depth_images cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_image2d_from_buffer cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_media_block_io cl_intel_driver_diagnostics cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_khr_fp64 cl_khr_subgroups cl_khr_il_program cl_intel_spirv_device_side_avc_motion_estimation cl_intel_spirv_media_block_io cl_intel_spirv_subgroups cl_khr_spirv_no_integer_wrap_decoration cl_khr_mipmap_image cl_khr_mipmap_image_writes cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_motion_estimation cl_intel_device_side_avc_motion_estimation cl_intel_advanced_motion_estimation cl_intel_va_api_media_sharing 
  Platform Host timer resolution                  1ns
  Platform Extensions function suffix             INTEL

  Platform Name                                   Intel(R) OpenCL HD Graphics
Number of devices                                 1
  Device Name                                     Intel(R) Gen9 HD Graphics NEO
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 NEO 
  Driver Version                                  19.32.13826
  Device OpenCL C Version                         OpenCL C 2.0 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               24
  Max clock frequency                             1100MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
    Supported affinity domains                    (n/a)
  Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
  Preferred work group size multiple              32
  Max sub-groups per work group                   32
  Sub-group sizes (Intel)                         8, 16, 32
  Preferred / native vector sizes                 
    char                                                16 / 16      
    short                                                8 / 8       
    int                                                  4 / 4       
    long                                                 1 / 1       
    half                                                 8 / 8        (cl_khr_fp16)
    float                                                1 / 1       
    double                                               1 / 1        (cl_khr_fp64)
  Half-precision Floating-point support           (cl_khr_fp16)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
  Single-precision Floating-point support         (core)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  Yes
  Double-precision Floating-point support         (cl_khr_fp64)
    Denormals                                     Yes
    Infinity and NANs                             Yes
[... 72 lines cut ...]
NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Intel(R) OpenCL HD Graphics
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [INTEL]
  clCreateContext(NULL, ...) [default]            Success [INTEL]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 Intel(R) OpenCL HD Graphics
[... cut ...]
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.12
  ICD loader Profile                              OpenCL 2.2
```

I think if `cl_intel_va_api_media_sharing` was not in the list of supported extensions, ffmpeg would fail with a [useful error message](https://github.com/FFmpeg/FFmpeg/blob/a80fdbcf13919d1280278e71c4ae29e760fb82ab/libavutil/hwcontext_opencl.c#L975-L989), which would be much more useful to the user than a segfault.

JacekDanecki, 2019-09-02:
Current sharing implementation verifies dependency on libva during cmake and later in runtime. Checking whether DdiMedia_ExtGetSurfaceHandle function is available using libva interface in media driver would require code similar to below:
```
bool isiHDLibraryAvailable(int lib) {
    VAExtGetSurfaceHandlePFN testVaExtGetSurfaceHandlePFN;
    VAGetLibFuncPFN testVaGetLibFuncPFN;

    testVaGetLibFuncPFN = reinterpret_cast<VAGetLibFuncPFN>(fdlsym(lib, "vaGetLibFunc"));
    int fd = open("/dev/dri/renderD128", O_RDWR);

    VADisplay va_dpy = vaGetDisplayDRM(fd);
    int major_version = 0, minor_version = 0;
    vaInitialize(va_dpy, &major_version, &minor_version);

    testVaExtGetSurfaceHandlePFN = reinterpret_cast<VAExtGetSurfaceHandlePFN>(testVaGetLibFuncPFN(va_dpy, "DdiMedia_ExtGetSurfaceHandle"));
     
    vaTerminate(va_dpy);
    close(fd);

    if (testVaExtGetSurfaceHandlePFN) {
        return true;
    }
    return false;
}
```
and modification in VASharingFunctions::isVaLibraryAvailable()
```
bool VASharingFunctions::isVaLibraryAvailable() {
    auto lib = fdlopen(Os::libvaDllName, RTLD_LAZY);
    if (lib) {
        bool ret = isiHDLibraryAvailable(lib);
        fdlclose(lib);
        return ret;
    }
    return false;
}
```
To enable above code Neo or your application need to be linked with libva-drm.so library.

JacekDanecki, 2019-11-18:
I think default driver in libva has been changed with [this commit](https://github.com/intel/libva/commit/09c8c2e20b7763fc44299172dbb16061b57f9c71)

JacekDanecki, 2020-01-31:
I've checked with ffmpeg n4.2.2 compiled to enable OpenCL support on Arch, default is now iHD, and provided command works.
It's segfaults when we force LIBVA_DRIVER_NAME=i965 environment variable, which is not supported configuration. Closing issue.

closed 2020-01-31 by JacekDanecki