intel/compute-runtime #214: Out of order queue is ordered

TApplencourt, 2019-09-24:
Hi,

I have a [simple program](https://github.com/alcf-perfengr/alcl/blob/master/C/event_unorder.c), where I submit my kernels in an out of order queue.

Every even kernel is independents, every odd kernel depends on the same user-even. 
I submit all my kernel, sleep for a while, and them trigger my event.

For example I submit 4 kernel (A, B , C , D), where A, and C need to wait for a user-event, but C and D have no dependency.

## Expected behavior

I except all my even kernel to execute (the one with no depdency), and then my odd kernels.
(B,D then A,B). 

## Actual Behavior

My kernels are executed in the order of the enqueuing. (A,B,C,D). See logfile in next section.

## How to reproduce

OpenCL version used: ` 19.37.14191`

We will use  OpenCL Intercept Loader to see the time of e
1- Compile the file
2- Run it `./event_unorder 0 0 1 1 10` with OpenCL intercept Layer to print information 
about enqueuing timing (`export CLI_DevicePerformanceTimelineLogging=1`)

Output:
```
 ./event_unorder 0 0 1 1 10
[...]
Device Timeline for hello_world (enqueue 1) = 686595186105936 ns (queued), 686595186149722 ns (submit), 686596188876103 ns (start), 686596188880353 ns (end)
Device Timeline for hello_world (enqueue 2) = 686595186182619 ns (queued), 686596187521783 ns (submit), 686596188889202 ns (start), 686596188893118 ns (end)
Device Timeline for hello_world (enqueue 3) = 686595186256393 ns (queued), 686596188817252 ns (submit), 686596188935560 ns (start), 686596188946726 ns (end)
Device Timeline for hello_world (enqueue 4) = 686595186283534 ns (queued), 686596188867511 ns (submit), 686596188984534 ns (start), 686596188988784 ns (end)
Device Timeline for hello_world (enqueue 5) = 686595186362292 ns (queued), 686596188892244 ns (submit), 686596189011125 ns (start), 686596189015041 ns (end)
Device Timeline for hello_world (enqueue 6) = 686595186392017 ns (queued), 686596188930668 ns (submit), 686596189038684 ns (start), 686596189042517 ns (end)
Device Timeline for hello_world (enqueue 7) = 686595186493428 ns (queued), 686596188971883 ns (submit), 686596189077928 ns (start), 686596189079428 ns (end)
Device Timeline for hello_world (enqueue 8) = 686595186541115 ns (queued), 686596188987994 ns (submit), 686596189087032 ns (start), 686596189088448 ns (end)
Device Timeline for hello_world (enqueue 9) = 686595186566353 ns (queued), 686596189005904 ns (submit), 686596189096103 ns (start), 686596189097603 ns (end)
Device Timeline for hello_world (enqueue 10) = 686595186594375 ns (queued), 686596189023214 ns (submit), 686596189105209 ns (start), 686596189106625 ns (end)
```

As you can see the kernel are queue, submited, started and ended in order.

MichalMrozek, 2019-09-25:
It is expected with the current implementation.
Because one queue is used those will get serialized.
If you would use 2 queues here I think you would see different output.
In general I would advocate to not use user events, multiple optimization are not triggered in those code paths.

TApplencourt, 2019-09-25:
Thanks for your quick answers!

>It is expected with the current implementation.
> Because one queue is used those will get serialized.
>  If you would use 2 queues here I think you would see different output.

The goal is to show that "out of order" queue can process kernel in an "out of order" fashion.
During our experimentation, we never saw an enqueued commands finishing execution in a different order they were queued.   

Perhapts a more general question is, what is the behavior of the "out of order" queue in the current implementation?

MichalMrozek, 2019-09-25:
Our out of order queue implementation does not actually change the order of submissions.
We are not inserting any synchronization between kernels , so they would be able to run concurrently and that is the place where we gain performance in scenarios where machine is underutilized.

i.e. consider following:
kernel A -> short
kernel B -> long
kernel C -> short
kernel D -> short

and imagine that we fit all the kernels to run concurrently ( gws is low ), I would expect that A C and D will complete prior to B also one thing, do not use profiling to observe that as that serializes as well.

For user events our implementation also serializes the submission that are blocked on user events, that's why you see what you see.

I don't think you will be able to see reverse ordering with current implementation in one queue as the easiest way to achieve that is by user events and those will serialize.

TApplencourt, 2019-09-25:
```
i.e. consider following:
kernel A -> short
kernel B -> long
kernel C -> short
kernel D -> short

and imagine that we fit all the kernels to run concurrently ( gws is low ), I would expect that A C and D will complete prior to B
```
I confirm that it's the case.  Thanks a lot for the insight! 

> also one thing, do not use profiling to observe that as that serializes as well.

Yes, using profiling events to time my kernels execution does indeed serialize them. 
Does that mean that when using profiling tools  (ie VTune) I will not be able to see any out-of-order execution of my kernels?

MichalMrozek, 2019-09-25:
"Does that mean that when using profiling tools (ie VTune) I will not be able to see any out-of-order execution of my kernels?"

That is correct, those tools will serialize results as well.

TApplencourt, 2019-09-26:
Thanks a lot!  It's clearer now.  Do you think that it's worth-it do add a comment about this behavior is the Documentation?  Other users may have the same question as us.

Anyway, I will close this Issue. Thanks again.

closed 2019-09-26 by TApplencourt

MichalMrozek, 2019-10-01:
I have added those details in our FAQ 6061ccc6fd6c66318702467d86268f97425e58f8