intel/compute-runtime #217: After installing 19.39, I have segfault running "clinfo"

dbabokin, 2019-10-04:
```
~/env/neo/neo-19.39> clinfo
Abort was called at 99 line in file:
../neo/runtime/built_ins/built_ins.cpp
Aborted (core dumped)
```

19.32-19.38 environments worked fine.
I'm running Ubuntu 18.04

jdanecki, 2019-10-05:
Can you provide steps you used to install this release? When I use steps provided on [release page](https://github.com/intel/compute-runtime/releases/tag/19.39.14278) clinfo works correctly.
Can you share file log.txt created by command:
```
strace -f -s 255 -o log.txt clinfo
```
Output provided by clinfo after setting below environment variable will be helpful also
```
export LD_DEBUG=files
clinfo
```
Can you also share output from below command:
```
dpkg -l intel*
```

dbabokin, 2019-10-06:
Looks like I found what causes the problem - I have fresh installation of 19.39 in the system, but LD_LIBRARY_PATH points to files with 19.34. So, the crash. I'm not sure if there a more graceful way to handle this problem, i.e. reporting properly some library version mismatches. I'll send you the dumps that you requested by email.

dbabokin, 2019-10-16:
The problem was not a broken / wrong system package, but LD_LIBRARY_PATH pointing to the wrong lib. Would be good to have more informative error message in this case (mismatching libraries version, etc), instead of mistery abort.