intel/compute-runtime #221: Possible issue regarding barriers

jjfumero, 2019-10-08:
Since the last driver update from CentOS 7, I detected a problem regarding reductions in OpenCL. I suspect the problem is related to the barrier implementation. 

I will try to describe the problem, but I could not create a simple test-case to show this since I generate parallel reductions from an SSA-IR (GraalIR) to OpenCL [1]. Just for completeness, this is the OpenCL kernel I am running: [link](https://gist.github.com/jjfumero/fd5fa1e9112fad7e674fa27df97b3968).

My goal here is to ask if you have detected/changed something related to the barriers implementation? It is a bug in the OpenCL driver? is there any work-around? 

#### Some details:
By using the following Intel Graphics Driver:

```bash
  Name:						 Intel(R) Gen9 HD Graphics NEO
  Vendor:					 Intel(R) Corporation
  Device OpenCL C version:			 OpenCL C 2.0 
  Driver version:				 19.34.13959
  Profile:					 FULL_PROFILE
  Version:					 OpenCL 2.1 NEO 
```

give me wrong results (not all the time though) when computing reductions in OpenCL. Since the result is not deterministic, I think the problem is related to the barriers. **However, this kernel used to work properly using an older driver version:**

```bash
  Name:						 Intel(R) Gen9 HD Graphics NEO
  Vendor:					 Intel(R) Corporation
  Device OpenCL C version:			 OpenCL C 2.0 
  Driver version:				 19.23.13131
  Profile:					 FULL_PROFILE
  Version:					 OpenCL 2.1 NEO 
```

#### Links

[1] https://github.com/beehive-lab/TornadoVM

MichalMrozek, 2019-10-08:
Looks like suspect is barrier implementation, can you move this to Compiler GitHub?
https://github.com/intel/intel-graphics-compiler

paigeale, 2019-10-10:
This does not appear to be a compiler issue. I can confirm there is no difference in the barrier implementation between the two drivers in fact the dumps show identical assembly.

jjfumero, 2019-10-17:
Hi all, 
   I just updated the driver on CentOS (regular updates) to this version:

```bash
  Name:				       Intel(R) Gen9 HD Graphics NEO
  Vendor:					 Intel(R) Corporation
  Device OpenCL C version:			 OpenCL C 2.0 
  Driver version:				 19.40.14409
  Profile:					 FULL_PROFILE
  Version:					 OpenCL 2.1 NEO 
```

Now my reductions work as used to be! Thank you for the updates! Out of curiosity, what was the problem? barriers? 

Thank you, 
Juan

JacekDanecki, 2019-10-17:
Since Neo 19.34.13959 to 19.40.14409 IGC was updated 2 times, so probably it was fixed with IGC update. I'm closing this issue, as it works for you now. If you find new problem, please reopen this issue or create new one.

closed 2019-10-17 by JacekDanecki