intel/compute-runtime #222: Unable to use clCreateProgramWithIL

rahulghangas, 2019-10-09:
Hi,

I was trying to use SPIR-V with OpenCL using clCreateProgramWithIL but it throws as an error
CL_INVALID_VALUE.

I'm using 
`ioc64 -cmd=build -input=my_kernel.cl -spirv64=my_kernel.spirv`
to generate the SPIR-V binary and SPIRV-TOOLS by Khronos to validate it.

Here's the revlevant output from clinfo as well
`Platform Name                                   Intel(R) CPU Runtime for OpenCL(TM) Applications
Number of devices                                 1
  Device Name                                     Intel(R) Xeon(R) Gold 6134 CPU @ 3.20GHz
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 (Build 0)
  Driver Version                                  18.1.0.0920
  Device OpenCL C Version                         OpenCL C 2.0 
  Device Type                                     CPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               32
  Max clock frequency                             3200MHz
  Device Partition                                (core)
    Max number of sub-devices                     32
    Supported partition types                     by counts, equally, by names (Intel)
  Max work item dimensions                        3
  Max work item sizes                             8192x8192x8192
  Max work group size                             8192
  Preferred work group size multiple              128
  Max sub-groups per work group                   1`

I was hoping someone could help me out with this.

jchodor, 2019-10-09:
Hi,
what's the implementation/signature of readBinaryFile?
Are you sure that you need (const void *)&binary   instead of  (const void *)binary? 

Please note that signature of clCreateProgramWithIL is :
cl_program clCreateProgramWithIL (cl_context context, const void *il, size_t length, cl_int *errcode_ret)  - i.e. single indirection to il

and **not** :

cl_program clCreateProgramWithIL (cl_context context, const void **il, size_t length, cl_int *errcode_ret) - i.e. double indirection to il

rahulghangas, 2019-10-11:
Thanks, this works. I didn't realise I was doing something that silly.

closed 2019-10-11 by rahulghangas

milthorpe, 2019-10-11:
Out of interest, can anyone explain why [`clCreateProgramWithIL`](https://www.khronos.org/registry/OpenCL/sdk/2.1/docs/man/xhtml/clCreateProgramWithIL.html) accepts a binary pointer as `void*`, whereas [`clCreateProgramWithBinary`](https://www.khronos.org/registry/OpenCL/sdk/2.1/docs/man/xhtml/clCreateProgramWithBinary.html) the program binaries are `unsigned char*`? This seems inconsistent.

bashbaug, 2019-10-16:
Hello,

> Out of interest, can anyone explain why [`clCreateProgramWithIL`](https://www.khronos.org/registry/OpenCL/sdk/2.1/docs/man/xhtml/clCreateProgramWithIL.html) accepts a binary pointer as `void*`, [...]

I agree this is inconsistent, but the inconsistency appears to be more the use of `unsigned char**` for program binaries for `clCreateProgramWithBinary` and less the use of `void*` for IL for `clCreateProgramWithIL` - almost all other APIs use `void*` to pass or return arbitrary data, or `char*` for those which clearly pass or return text.

I looked back through the (non-public, unfortunately) Khronos bugzilla and found that `clCreateProgramWithBinary` used to accept `void**` for program binaries, but this was changed (in 2009!) to accept `unsigned char*` to be consistent with the query for `clGetProgramInfo(CL_PROGRAM_BINARIES)`.  Unfortunately this was before my time in Khronos, so I can't offer more insight than that.

With benefit of hindsight, I wonder if it would have made more sense for `clGetProgramInfo(CL_PROGRAM_BINARIES)` to return `void*[]` instead?  I initially thought this might require an extra explicit cast, but given that the array of program binaries to `clGetProgramInfo` as a `void*`, this doesn't appear to be the case.

Hopefully this is helpful, even though I don't think I've answered your question completely.  Thanks!