intel/compute-runtime #225: 19.40 compiling issue

gogazago, 2019-10-14:
Hi I am getting compiling error 

error log
make[2]: Leaving directory '/builddir/build/BUILD/compute-runtime-19.40.14409/clr-build'
make[2]: *** [igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen12LPlp.dir/build.make:73: bin/built_ins/x64/gen12lp/aux_translation_Gen12LPlp.spv] Aborted (core dumped)
make[1]: *** [CMakeFiles/Makefile2:1317: igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen12LPlp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 31%] Generating ../../../bin/built_ins/x64/gen9/copy_buffer_rect_Gen9core.spv, ../../../bin/built_ins/x64/gen9/copy_buffer_rect_Gen9core.bin, ../../../bin/built_ins/x64/gen9/copy_buffer_rect_Gen9core.cpp, ../../../bin/built_ins/x64/gen9/copy_buffer_rect_Gen9core.gen

gmmlib version : 19.3.2.1
igc version: 1.0.2652

anyone know the reason I am getting this error?

JacekDanecki, 2019-10-15:
Yes, it's known issue. Please use [this IGC commit](https://github.com/intel/intel-graphics-compiler/commit/0efaaf958f644e454c84f86ac83109e59eb0580a).

gogazago, 2019-10-17:
I used that commit but still failing with same error message :(

JacekDanecki, 2019-10-18:
Can you provide output from cmake command running for IGC and Neo? How did you install IGC?

gogazago, 2019-11-11:
[build.log](https://github.com/intel/compute-runtime/files/3833016/build.log)
here is build log for IGC, 
for clearlinux

gogazago, 2019-11-15:
commit 710e6273dd1698e8b11763e9ff74c0dd57dddb83 (HEAD -> master, tag: igc-1.0.2714.1, origin/releases/igc-1.0.2714)
Author: Łukasz Węsierski <lukasz.wesierski@intel.com>
Date:   Thu Oct 17 15:59:06 2019 +0200

    Fix for gen12 compilation in NEO
seems like I need use this commit one right?

JacekDanecki, 2019-11-15:
Neo [release 19.40.14409](https://github.com/intel/compute-runtime/releases/tag/19.40.14409) was compiled with [igc 1.0.2597](https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.2597).

[igc 1.0.2714.1](https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.2714.1) was used by Neo releases: 
- [19.43.14583](https://github.com/intel/compute-runtime/releases/tag/19.43.14583)
- [19.44.14658](https://github.com/intel/compute-runtime/releases/tag/19.44.14658)

Can you verify whether components with revisions provided on [release page](https://github.com/intel/compute-runtime/releases) build under clearlinux?

gogazago, 2019-11-15:
Name        : intel-compute-runtime
Version     : 19.43.14583
yes I am using latest release build and keep failing 
make[2]: Leaving directory '/builddir/build/BUILD/compute-runtime-50f9674d79a031f1de56d0ba372318d4a0a1133e/clr-build'
make[2]: Leaving directory '/builddir/build/BUILD/compute-runtime-50f9674d79a031f1de56d0ba372318d4a0a1133e/clr-build'
make[2]: *** [igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen12LPlp.dir/build.make:62: bin/scheduler/x64/gen12lp/scheduler_Gen12LPlp.bin] Aborted (core dumped)
make[1]: *** [CMakeFiles/Makefile2:2111: igdrcl_lib_release/scheduler/CMakeFiles/scheduler_Gen12LPlp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 20%] Built target scheduler_Gen9core
make[2]: *** [igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen12LPlp.dir/build.

gogazago, 2019-11-15:
let me check one by one

gogazago, 2019-11-20:
we have temporary disable GEN12 build,

need to fix it anyway

JacekDanecki, 2019-11-25:
@gogazago Can you provide spec file you use to build Neo under clealinux?

gogazago, 2019-11-25:
there is no neo build, we build it one by one
if you go clearlinux/package/intel-compute-runtime
you will see my patch include 
--DSKIP_UNIT_TESTS=1
+-DSKIP_UNIT_TESTS=1 \
+-DSUPPORT_GEN12LP=0
disabling GEN12LP

JacekDanecki, 2019-11-26:
Can you provide steps to reproduce this issue? I can't reproduce this on my setup.
I've tried with clearlinux
```
swupd info
Distribution:      Clear Linux OS
Installed version: 31700
Version URL:       https://cdn.download.clearlinux.org/update
Content URL:       https://cdn.download.clearlinux.org/update
```
I've enabled Gen12LP build in spec file
```
%cmake .. -DBUILD_TYPE=Release \
-DSKIP_UNIT_TESTS=1 \
-DSUPPORT_GEN12LP=1
```
and cmake reported enabled TGLLP support
```
-- All supported platforms:  TGLLP ICLLP LKF EHL SKL KBL GLK CFL BXT BDW
```
and executed rpmbuild  by
```
rpmbuild -bc intel-compute-runtime.spec --nodeps
```
and build finished correctly
```
make[2]: Leaving directory '/rpmbuild/BUILD/compute-runtime-19.45.14764/clr-build'
[100%] Built target igdrcl_dll
make[1]: Leaving directory '/rpmbuild/BUILD/compute-runtime-19.45.14764/clr-build'
/usr/bin/cmake -E cmake_progress_start //rpmbuild/BUILD/compute-runtime-19.45.14764/clr-build/CMakeFiles 0
+ popd
//rpmbuild/BUILD/compute-runtime-19.45.14764
+ exit 0
```

gogazago, 2019-11-26:
let me retry

Thanks,
Sean.

gogazago, 2019-11-26:
yes it works!! :) Thanks, idk what I have changed on other sources :(

Thanks,
Sean

JacekDanecki, 2019-11-27:
Closing as it works now.

closed 2019-11-27 by JacekDanecki