intel/compute-runtime #226: Out-of-bounds access in shuffle() and shuffle2()

etomzak, 2019-10-15:
OpenCL specifies that the `mask` parameter to `shuffle()` and `shuffle2()` wraps around when it is larger than the (combined) vector size(s) [[source](https://www.khronos.org/registry/OpenCL/sdk/1.1/docs/man/xhtml/shuffle.html)]. The compute-runtime implementation appears not to implement the wrap-around, which creates a potential out-of-bounds access.

The issue is similar to the one fixed here: https://github.com/jrprice/Oclgrind/pull/178

JacekDanecki, 2019-10-25:
This looks like compiler bug, please open new issue in [igc repo](https://github.com/intel/intel-graphics-compiler/issues)

paigeale, 2019-11-01:
Yes this is a bug in the compiler we will be checking in a fix for this shortly.

paigeale, 2019-11-08:
Bug is fixed with f75a1e835cf2375e01dc6083b7988d62444aa62e. Please Close

MichalMrozek, 2019-11-19:
Closing as @paigeale suggested.

closed 2019-11-19 by MichalMrozek